fwupd/fwupd #128: How to add Retrode support

hadess, 2017-06-06:
The Retrode (a cartridge dumper and joystick adapter) has upgradeable firmwares:
http://www.retrode.org/documentation/firmware/

It uses DFU, and the firmware files are availble. How do the firmwares need to be packaged to allow them to be automatically used by fwupd, on the command-line to start? And then, is there documentation about the process for vendors to upload their wrapped firmwares to LVFS.

Both answers to those questions would be useful to have in the README for fwupd, at least as a link in the LVFS case.

hughsie, 2017-06-06:
So, AVR DFU isn't actually proper DFU... `dfu-programmer` is for Atmel devices only, rather than the standards-compliant `dfu-util`. That said, I think we can do something here. Have you read the pages at http://www.fwupd.org/ -- the main problem is usually getting the vendor to agree to distribute the firmware on the LVFS rather than any technical limitation. The libdfu library in fwupd just needs to be taught about the Atmel-specific calls and then the flash should just work. Have you reached out to the vendor?

hadess, 2017-06-07:
I've added more links to fwupd.org to the README, that would have helped me out when filing this bug.

I've also retitled the bug. Not sure when I would get the time to dive into adding support for the AVR variant, but at least it'll make it clearer what this bug is about.

hughsie, 2017-10-28:
I've done this in https://github.com/hughsie/fwupd/pull/292 for a different vendor, although for this issue I've not heard anything back from the Retrode people yet. @hadess do you have any contacts there? Could you try emailing as a valid customer looking for Linux support?

hughsie, 2017-11-08:
I've tried emailing Retrode directly, with no response. If they do want to support fwupd and the LVFS all the support is there and it should Just Work (tm).

closed 2017-11-08 by hughsie

Limero, 2017-11-08:
You should contact skaman directly. Skaman is the person that currently does firmware releases and seems very active. http://forum.retrode.org/index.php/topic,382.0.html

hughsie, 2017-11-08:
@Limero I've sent him a direct message on the forum. If you're a member there perhaps you can explain to him that I'm not trying to sell him anything. :)