fwupd/fwupd #129: 0.9.3 Test suite is  due to missing hwids

superm1, 2017-06-07:
```
        ninja test
[0/1] Running all tests.
1/6 dfu-self-test                           OK       0.03 s
2/6 fwupd-self-test                         OK       0.01 s
3/6 raspberrypi-self-test                   OK       0.15 s
4/6 udev-self-test                          OK       0.02 s
5/6 unifying-self-test                      OK       0.01 s
6/6 fu-self-test                            FAIL     0.25 s

OK:         5
FAIL:       1
SKIP:       0
TIMEOUT:    0

The output from the failed tests:

6/6 fu-self-test                            FAIL     0.25 s

--- command ---
/build/fwupd-0.9.3/obj-x86_64-linux-gnu/src/fu-self-test
--- stdout ---
/fwupd/hwids:                                                                                                                                                                                                  --- stderr ---
**
Fu:ERROR:../src/fu-self-test.c:69:fu_hwids_func: assertion failed: (sysfsdir != NULL)
-------
```

I don't see any `hwids `under `data/tests`, did you forget to commit this?

closed 2017-06-07 by hughsie

hughsie, 2017-06-07:
I'll spend a bit of time enabling Travis CI so this kind of thing doesn't happen anymore....

superm1, 2017-06-07:
I was gonna suggest the same thing :)
The challenge with Travis CI I think is these build dependencies are fairly new and travis runs on at latest ubuntu 14.04 containers IIRC.  Eg you'll hit the polkit issues, get stuck with a really old meson and such.