fwupd/fwupd #134: fwupd.spec isn't build by meson/ninja

vathpela, 2017-06-12:
fwupd.spec should probably get built as part of the normal build process, rather than with external tooling.

hughsie, 2017-06-12:
I think we need to do the substitution in the Travis script, rather than meson, otherwise it makes the end product non-reproducible.

superm1, 2017-06-13:
Meson does have an [RPM module](http://mesonbuild.com/RPM-module.html), but it isn't meant for programmatic use.  It just creates the initial template with the list of the dependencies and requires someone to go fill in the rest.

As @hughsie said, by the CI being run now on both Fedora and Debian packaging any time that either need changing to the point that a build fails to build or fails to install packages it should be caught immediately.

closed 2017-06-13 by superm1