fwupd/fwupd #137: Bug v0.9.3 /usr/lib/fwupd/remotes.d: No such file or directory

PieterDeBruijn, 2017-06-13:
3 users (and me) have this error.
I’m getting an error when running `fwupdmgr refresh` in the v0.9.3 release:
Error opening directory “/usr/lib/fwupd/remotes.d”: No such file or directory

More info:
https://aur.archlinux.org/packages/fwupd/

hughsie, 2017-06-13:
So, I'm guessing /usr/lib/fwupd exists but not /usr/lib/fwupd/remotes.d ? Did the arch build perhaps create the former and not the latter?

closed 2017-06-13 by hughsie

PieterDeBruijn, 2017-06-13:
Correct. Manually adding the dir, fixes the error:
sudo mkdir /usr/lib/fwupd/remotes.d
and refresh runs without issues. This can be fixed (best) in the Arch build script? Then I'll report back downstream.

hughsie, 2017-06-13:
You can either create the directory in the pkgbuild, or backport that last patch, which ever is easiest.

ArchangeGabriel, 2017-06-13:
What is this directory used for? Should it be created by the package even with the patch?

superm1, 2017-06-13:
It's for allowing packages to ship additional remote locations.  I think it makes sense to have it created directly by meson (or meson_post_install.sh).  Patches welcome!