fwupd/fwupd #139: No such method 'UpdateMetadataWithId'

Lubo, 2017-06-14:
When running `$ fwupdmgr refresh`, I get `No such method 'UpdateMetadataWithId'`. I'm using Arch Linux x86_64 with fwupd 0.9.3 from community repository.

hughsie, 2017-06-14:
If you updated fwupd recently, you either need to reboot or restart the fwupd process.

Lubo, 2017-06-14:
Ok, I've tried both and it seems like the fwupd process segfaults shortly after being started. Here's log from journal:

```
Jun 14 10:25:00 <hostname> systemd[1]: Starting Firmware update daemon...
Jun 14 10:25:02 <hostname> fwupd[2019]: ignoring add-delay as device usb:00:01:03 already pending
Jun 14 10:25:02 <hostname> fwupd[2019]: ignoring add-delay as device usb:00:01:01 already pending
Jun 14 10:25:02 <hostname> fwupd[2019]: ignoring add-delay as device usb:00:05 already pending
Jun 14 10:25:02 <hostname> fwupd[2019]: ignoring add-delay as device usb:00:04 already pending
Jun 14 10:25:02 <hostname> fwupd[2019]: disabling plugin because: failed to coldplug raspberrypi: Raspberry PI firmware updating not supported, no /boot/start.elf
Jun 14 10:25:03 <hostname> systemd[1]: Started Firmware update daemon.
Jun 14 10:25:04 <hostname> systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jun 14 10:25:04 <hostname> systemd[1]: fwupd.service: Unit entered failed state.
Jun 14 10:25:04 <hostname> systemd[1]: fwupd.service: Failed with result 'core-dump'.
```

My device is Dell XPS 13 (9360). Core dump is attached below.

[core.fwupd.0.2b98b8842d71470bb75d79b61c1a2971.2019.zip](https://github.com/hughsie/fwupd/files/1074016/core.fwupd.0.2b98b8842d71470bb75d79b61c1a2971.2019.zip)

hughsie, 2017-06-14:
Cab you get a backtrace please? I don't have arch debug symbols installed.

Lubo, 2017-06-14:
Unfortunately, neither do I. This is all I have:

```
#0  0x00007ffff7485b87 in  () at /usr/lib/libglib-2.0.so.0
#1  0x00007ffff7485850 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff7486830 in g_variant_get_va () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff7486ac0 in g_variant_get () at /usr/lib/libglib-2.0.so.0
#4  0x000000000040df40 in  ()
#5  0x00007ffff7128d3c in  () at /usr/lib/libgio-2.0.so.0
#6  0x00007ffff744c8b5 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#7  0x00007ffff744cc78 in  () at /usr/lib/libglib-2.0.so.0
#8  0x00007ffff744cf92 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#9  0x000000000040af64 in main ()
```

hughsie, 2017-06-14:
I think you have hit the problem fixed by b38a4fdfd631a800d4171a4eb3e519075755f430

Lubo, 2017-06-14:
Seems so. I'll report back after the next release, if it's OK.

ArchangeGabriel, 2017-06-14:
@Lubo Can you try to update to `fwupd-0.9.3-3`? It works for me now.

Lubo, 2017-06-14:
`fwupd-0.9.3-3`, which includes b38a4fdfd631a800d4171a4eb3e519075755f430, did it for me. Thank you, guys.

closed 2017-06-14 by Lubo