fwupd/fwupd #141: fwupd.service cannot be enabled

Lubo, 2017-06-14:
When trying to enable the service using `$ systemctl enable fwupd.service`, I get this error message:

```
The unit files have no installation config (WantedBy, RequiredBy, Also, Alias
settings in the [Install] section, and DefaultInstance for template units).
This means they are not meant to be enabled using systemctl.
Possible reasons for having this kind of units are:
1) A unit may be statically enabled by being symlinked from another unit's
   .wants/ or .requires/ directory.
2) A unit's purpose may be to act as a helper for some other unit which has
   a requirement dependency on it.
3) A unit may be started when needed via activation (socket, path, timer,
   D-Bus, udev, scripted systemctl call, ...).
4) In case of template units, the unit is meant to be enabled with some
   instance name specified.
```

I am using Arch Linux x86_64 and systemd 232.

hughsie, 2017-06-14:
fwupd is supposed to be activated with D-Bus activation; why are you trying to enable it on the command line?

Lubo, 2017-06-14:
Wasn't aware of that. I'm unable to find documentation, where can I find out more about fwupd installation and about usage of the service file, please?

hughsie, 2017-06-14:
I don't think we have any docs, it's more of a "install it and it just works" kind of thing. Sorry.

closed 2017-06-14 by hughsie