fwupd/fwupd #145: "FakeDevice" Integrated Webcam shows up on real systems

superm1, 2017-06-19:
I was just running a git build (that I self generated from CI packaging scripts) and noticed that the _FakeDevice_ is showing up on real `fwupdmgr get-devices` output.

I am thinking by default the 'test' plugin shouldn't be installed.

hughsie, 2017-06-19:
We need the test plugin for the CI otherwise there's nothing to test.

superm1, 2017-06-19:
Right, but I mean perhaps it shouldn't be stroed into the debs and rpms, but just copied into the docker filesystem during test.

ArchangeGabriel, 2017-06-19:
Is it this thing:
```
  Guid:                 9bab7d9d-9396-5929-bd36-64ff7bc84804
  Guid:                 31f4c6bf-8c1e-5c88-8153-984cbe1dad0e
  DeviceID:             usb:00:07
  Plugin:               usb
  Flags:                none
  Version:              153.6
  Created:              2017-06-19
```
?

By the way, I seem to remember having a device name above that, but it apparently does not show anymore…

ArchangeGabriel, 2017-06-19:
Hum, wrote too fast, guid does not match https://github.com/hughsie/fwupd/blob/master/plugins/test/fu-plugin-test.c.

superm1, 2017-06-19:
No, that's a real device.  This is the FakeDevice: https://github.com/hughsie/fwupd/blob/81f96797f6a4d5d48124d871ec441e177a7ec597/plugins/test/fu-plugin-test.c#L57

superm1, 2017-06-19:
Another idea rather than not installing into real filesystems, maybe only initialize the test plugin if something is in the environment when fwupd is spawned and set that up in the CI environment?

hughsie, 2017-06-19:
I actually set the CI=true environment var in the travis builds, so I guess we could self-disable if that env var isn't set?

superm1, 2017-06-19:
Might need some hokey way to pass that into the d-bus spawned fwupd (not sure if env variables will carry).  Otherwise that sounds good.

hughsie, 2017-06-19:
Ahh yes, right. That was exactly the reason I shifted away from the env var; it was tricky to "inject" it into the spawned DBus activated process.

superm1, 2017-06-19:
Something like this (untested):

```
diff --git a/contrib/ci/build_and_install_debs.sh b/contrib/ci/build_and_install_debs.sh
index faba5a3..4c98ce5 100755
--- a/contrib/ci/build_and_install_debs.sh
+++ b/contrib/ci/build_and_install_debs.sh
@@ -14,5 +14,8 @@ dpkg-buildpackage
 dpkg -i ../*.deb
 
 # run the installed tests
-/etc/init.d/dbus start
-gnome-desktop-testing-runner fwupd
+if [ "$CI" = "true" ]; then
+       sed "s,Exec=,Exec=CI=$CI ," /usr/share/dbus-1/system-services/org.freedesktop.fwupd.service
+       /etc/init.d/dbus start
+       gnome-desktop-testing-runner fwupd
+fi
```

superm1, 2017-06-20:
I've got a PR that will do this open here now: https://github.com/hughsie/fwupd/pull/148

superm1, 2017-06-20:
Closed with https://github.com/hughsie/fwupd/commit/f3f4390155f4768dd4c5b9664877397f0a60607c

closed 2017-06-20 by superm1