fwupd/fwupd #146: Devices name do not show anymore

ArchangeGabriel, 2017-06-19:
Before, I used to get devices name on `fwupdmgr get-devices`, but not anymore, see below:
```
  Guid:                 9bab7d9d-9396-5929-bd36-64ff7bc84804
  Guid:                 31f4c6bf-8c1e-5c88-8153-984cbe1dad0e
  DeviceID:             usb:00:07
  Plugin:               usb
  Flags:                none
  Version:              153.6
  Created:              2017-06-19

  Guid:                 2b07aa6b-6b41-5e92-8f2f-ff742f711a81
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
  Plugin:               udev
  Flags:                internal|locked
  DeviceVendor:         NVIDIA Corporation
  Created:              2017-06-19
```

Not sure what changed in between.

hughsie, 2017-06-19:
Does this commit fix it? https://github.com/hughsie/fwupd/commit/de5dd8f33dbc643d7e55087eef29f5a59b6dda1c

ArchangeGabriel, 2017-06-19:
It makes `ninja test` fail:
```
ninja: Entering directory `build'
[0/1] Running all tests.
1/8 dfu-self-test                           OK       0.02 s
2/8 fwupd-self-test                         FAIL     0.03 s
3/8 raspberrypi-self-test                   OK       0.01 s
4/8 udev-self-test                          OK       0.01 s
5/8 unifying-self-test                      OK       0.01 s
6/8 dell-self-test                          OK       0.02 s
7/8 synapticsmst-self-test                  OK       0.01 s
8/8 fu-self-test                            OK       2.04 s

OK:         7
FAIL:       1
SKIP:       0
TIMEOUT:    0

The output from the failed tests:

2/8 fwupd-self-test                         FAIL     0.03 s

--- command ---
/build/fwupd/src/build/libfwupd/fwupd-self-test
--- stdout ---
/fwupd/enums: OK
/fwupd/result: 
ColorHug2
  Guid:                 2082b5e0-7a64-478a-b1b2-e3404fab6dad
  Guid:                 00000000-0000-0000-0000-000000000000
  DeviceID:             USB:foo
  DisplayName:          ColorHug2
  Flags:                allow-offline|require-ac
  FirmwareHash:         SHA1(beefdead)
  Created:              1970-01-01
  Modified:             1970-01-02
  AppstreamId:          org.dave.ColorHug.firmware
  UpdateDescription:    <p>Hi there!</p>
  UpdateVersion:        1.2.3
  FilenameCab:          firmware.bin
  UpdateHash:           SHA1(deadbeef)
  Size:                 1.0 kB
  UpdateUri:            http://foo.com
  Trusted:              payload
--- stderr ---
**
Fwupd:ERROR:../fwupd-0.9.3/libfwupd/fwupd-self-test.c:154:fwupd_result_func: assertion failed (error == NULL): --- /tmp/b       2017-06-19 17:37:18.370967086 +0200
+++ /tmp/a      2017-06-19 17:37:18.370967086 +0200
@@ -2,6 +2,7 @@ ColorHug2
   Guid:                 2082b5e0-7a64-478a-b1b2-e3404fab6dad
   Guid:                 00000000-0000-0000-0000-000000000000
   DeviceID:             USB:foo
+  DisplayName:          ColorHug2
   Flags:                allow-offline|require-ac
   FirmwareHash:         SHA1(beefdead)
   Created:              1970-01-01
 (-g-type-private--GTypeFlags, 0)
-------

Full log written to /build/fwupd/src/build/meson-logs/testlog.txt
FAILED: test 
/usr/bin/python -u /usr/bin/mesontest --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
```

But other than that it seems to work:
```
  Guid:                 9bab7d9d-9396-5929-bd36-64ff7bc84804
  Guid:                 31f4c6bf-8c1e-5c88-8153-984cbe1dad0e
  DeviceID:             usb:00:07
  DisplayName:          HP HD Camera
  Plugin:               usb
  Flags:                none
  Version:              153.6
  Created:              2017-06-19

  Guid:                 2b07aa6b-6b41-5e92-8f2f-ff742f711a81
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
  DisplayName:          GM107GLM [Quadro M1000M]
  Plugin:               udev
  Flags:                internal|locked
  DeviceVendor:         NVIDIA Corporation
  Created:              2017-06-19
```
Though I think that before the name was above, not in the middle. And if that wasn’t the case, maybe it should be then at least. ;)

superm1, 2017-06-19:
Yeah the test failure was fixed with https://github.com/hughsie/fwupd/commit/0eaca2da65d376ddd3ec170286feee5e88d68433

ArchangeGabriel, 2017-06-19:
OK, works with both patches.

closed 2017-06-19 by ArchangeGabriel