fwupd/fwupd #147: Failure to build 0.9.4

ArchangeGabriel, 2017-06-19:
Hi there,

I’ve seen there was a new release. When trying to build… erf no, _install_, it fails:
```
Running custom install script '/build/fwupd/src/fwupd-0.9.4/po/make-images.sh'
Failed to run install script '/build/fwupd/src/fwupd-0.9.4/po/make-images.sh'
FAILED: install 
```

Two questions:
1. Why isn’t that happening during build rather than install?
2. Looking at https://github.com/hughsie/fwupd/commit/2297c09f8134d42b221576dff497bc6498eff1a8, there seems to be new dependencies. But the meson file doesn’t list all the one that were added to the docker images, like `dejavu`, `python` ones or even `adobe-source-han-sans-cn-fonts`. Is this because meson cannot detect those? If so `Any missing dependencies will be mentioned while running meson.` is not accurate, and maybe those dependencies should be listed.

Thanks!

hughsie, 2017-06-19:
> Why isn’t that happening during build rather than install?

It is running a python script to generate the images.

> or even adobe-source-han-sans-cn-fonts

I don't think you can sanely check for the existence of a font in a meson file, but if you know how to locate a python module or a font please let me know.

superm1, 2017-06-19:
Conceivably that could be switched over to a custom_target and check that the images were generated properly, but it already exits non-zero on failures so that might be a wasted effort.

ArchangeGabriel, 2017-06-20:
I don’t know how this could be handled by meson. I’ve discovered it because of fwupd actually, so…

My point is, given the README, how are people expected to guess they need those deps?

ArchangeGabriel, 2017-06-20:
For the record, I’ve been able to build by adding `python-pillow`, `python-cairo`, `python-gobject` and `ttf-dejavu`. Seems that `adobe-source-han-sans-cn-fonts` is not mandatory, I guess it means the Chinese images are built with DejaVu then, right? Also it seems they are no French version, can I contribute a translation somewhere?

superm1, 2017-06-20:
@ArchangeGabriel for translation stuff take a look at https://www.transifex.com/freedesktop/fwupd/
You're packaging fwupd for arch right?  Would you consider to submit a Dockerfile and compile script to maintain for CI?  New deps and problems like this can be caught a lot more easily.

Regarding the python dependencies I put this together to test for them:
https://github.com/hughsie/fwupd/commit/14998921730392f035ff997181489053047c80b9

I'm not so sure about testing for fonts though.

ArchangeGabriel, 2017-06-20:
Asked to join on Transifex. Good idea for the python dependencies.

Regarding Dockerfile and CI, we have one in the making I think, but nothing ready yet. I’ll keep you informed.

superm1, 2017-06-20:
I've merged that python checking stuff to master.  If the fonts turn out to be a problem in the future we'll revisit, otherwise i'll close this issue out.

Feel free to open a PR when you've got something ready to go that builds arch stuff you want in CI.

Thanks!

closed 2017-06-20 by superm1

ArchangeGabriel, 2017-06-20:
Apparently our Docker image is ready: https://hub.docker.com/r/archlinux/base/. So I’ll try to work out something. ;)