fwupd/fwupd #149: running verify unlocks all devices

superm1, 2017-06-20:
I've found that running `#fwupdmgr verify` will unlock all devices that were locked and then not relock them when complete.

Furthermore due to the order of events, even devices that don't support checksums will be unlocked.

hughsie, 2017-06-21:
So, Unlock() isn't meant to actually prevent them from being flashed (perhaps it's the wrong name) -- the unlock dance is done to avoid reading the PROM of PCI hardware unless it's really required. We found that unconditionally reading the PCI OptionROM on some hardware causes it to stop working or throw messages into the journal. I figured doing it as needed was better than unconditionally at startup, as the mitigation on broken hardware was "don't do verify-update" rather than "disable fwupd".

superm1, 2017-06-21:
So I think the existing Dell plugin's TPM devices are using it a little differently then (description here: https://github.com/hughsie/fwupd/tree/master/plugins/dell).

hughsie, 2017-06-21:
It sounds like you're using Unlock() more in the spirit of the name. I think we need a new vfunc for the option rom stuff...

closed 2017-06-22 by hughsie