fwupd/fwupd #151: `fwupd: report 0x00 unknown length` logged per keypress

da2x, 2017-06-29:
I’ve got a wired Logitech G810 keyboard (Nordic layout).

The following message is repeated one–two times for every single keypress:

    fwupd: report 0x00 unknown length

fwupd-0.9.4-1.fc26.x86_64

hughsie, 2017-07-04:
Can you try the recently released 0.9.5 please? I've rewritten a chunk of the notification handling code. Thanks!

sentabi, 2017-07-07:
I've similiar issue
```
Jul 07 11:03:19 situal fwupd[7598]: report 0x02 unknown length
Jul 07 11:03:19 situal fwupd[7598]: report 0x20 unknown length
Jul 07 11:03:24 situal fwupd[7598]: report 0x02 unknown length
Jul 07 11:03:24 situal fwupd[7598]: report 0x02 unknown length
Jul 07 11:03:24 situal fwupd[7598]: report 0x20 unknown length
```
it's generate tons of these log 

Other logs
```
Jul 07 11:04:27 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:13] Unknown lvalue 'ProtectControlGroups' in section 'Service'
Jul 07 11:04:27 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:15] Unknown lvalue 'ProtectKernelModules' in section 'Service'
Jul 07 11:04:27 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:13] Unknown lvalue 'ProtectControlGroups' in section 'Service'
Jul 07 11:04:27 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:15] Unknown lvalue 'ProtectKernelModules' in section 'Service'
Jul 07 11:04:31 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:13] Unknown lvalue 'ProtectControlGroups' in section 'Service'
Jul 07 11:04:31 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:15] Unknown lvalue 'ProtectKernelModules' in section 'Service'
Jul 07 11:04:45 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:13] Unknown lvalue 'ProtectControlGroups' in section 'Service'
Jul 07 11:04:45 situal systemd[1]: [/usr/lib/systemd/system/fwupd.service:15] Unknown lvalue 'ProtectKernelModules' in section 'Service'
```
using latest version 
```
fwupd-0.9.5-1.fc25.x86_64
```

hughsie, 2017-07-07:
> Jul 07 11:03:19 situal fwupd[7598]: report 0x02 unknown length

These are not in the HID++ specification... I guess we just need to ignore them.

> Unknown lvalue 'ProtectKernelModules' in section 'Service'

Harmless, if you update systemd the warning will go away.

closed 2017-07-07 by hughsie