fwupd/fwupd #152: Firmware update rendered laptop in bootable

vadi2, 2017-07-01:
Dell latitude E7470 - there were two firmware updates available in the gnome software centre. I've pressed the button to install them, it did something, but then the updates were listed again. I've pressed the button several times to no visible effect.

When I next rebooted the laptop,

vadi2, 2017-07-01:
Pressed the button too early and the mobile website doesn't allow editing. On the next reboot, two firmware updates installed seemingly successfully, however the laptop is now unbootable: it throws up some error about invalid parameter.

How to fix?

This is a work laptop so I hope to resolve this by Monday.

vadi2, 2017-07-01:
Message in question:

[image]

superm1, 2017-07-01:
Can you share more about your OS setup?

* What OS?
* Are you dual booting?
* Was this Dell's factory install image or something you set up yourself?
* Did you do some manual setup of any boot entries ever?
* Did you manually create an /efi/boot/bootx64.efi at some point?

As for resolving this back into a normal state I would recommend you go into BIOS setup and delete the Linux firmware update boot entry for now until we figure out what's going on.

vadi2, 2017-07-01:
Of course!

Ubuntu 17.04, not dualbooting, own setup - laptop previously had Windows 10. I don't believe I've manually setup any bootentries or messed with bootx before - only thing I did was follow a tutorial to get Virtualbox to get with secure boot - that is the signed modules.

vadi2, 2017-07-01:
[image]

Would I press the delete option?

superm1, 2017-07-01:
Yes press delete there.
If it's still not booting you're missing the entry for regular Ubuntu. You can manually add it with add boot option, and then browse to /efi/ubuntu/shimx64.efi
Title it 'ubuntu'

When you get back up and running can you share a tree listing of your /boot/efi and efibootmgr -v output

vadi2, 2017-07-01:
I couldn't press delete - it seems that was the only option available. When I added the Ubuntu boot option, I was able to delete the `fwupx64.efi` one.

The Ubuntu boot option worked splended, appreciate the prompt help. Here are the listings:

vadi2, 2017-07-01:
```
vadi@volga:~$ tree /boot/efi
/boot/efi [error opening dir]

0 directories, 0 files
vadi@volga:~$ efibootmgr -v
BootCurrent: 0003
Timeout: 2 seconds
BootOrder: 0003
Boot0000* Windows Boot Manager	HD(1,GPT,810fffed-b9bf-437b-a9f6-f8bdbe5ee113,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...6................
Boot0002* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,13e8fdcb-3ae3-440d-b562-46776b769bc9,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0003* ubuntu	PciRoot(0x0)/Pci(0x17,0x0)/Sata(4,32768,0)/HD(1,GPT,13e8fdcb-3ae3-440d-b562-46776b769bc9,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
vadi@volga:~$ 
```

Software centre also shows that the TPM 2.0 update is still outstanding, but the other firmware update is gone.

vadi2, 2017-07-03:
The BIOS settings are protected with an admin password, and I was told that firmware updates on Windows require entering the said admin password. I didn't have to on Linux - could that be the issue that it couldn't change the boot settings back?

superm1, 2017-07-05:
@vadi2 
Capsule BIOS updates are not protected by an admin password (even on Windows).

Can you share that tree output when run as root (under `sudo`)? 

It looks to me like you were missing the `ubuntu `boot entry in the first place, so i'm wondering how you had this booting previously. Ubuntu install should have created the boot entry for you.

vadi2, 2017-07-06:
Here it is:

```
/boot/efi
└── EFI
    └── ubuntu
        ├── fw
        │   ├── fwupdate-1DHirQ.cap
        │   └── fwupdate-YEh6w2.cap
        ├── fwupx64.efi
        ├── grub.cfg
        ├── grubx64.efi
        ├── mmx64.efi
        └── shimx64.efi

3 directories, 7 files
```

superm1, 2017-07-06:
Well nothing funky in that tree listing. My _(educated)_ guess on what's going on is that there was some automatic variable cleanup magic that went on.  Since you had the stale Windows boot entry still listed but not in BootOrder, maybe something got clobbered.

So my current thought is we can't do anything about your current problem with the current amount of information available and reproducibility.  
I do think it might make sense to start logging information about the various Boot* variables to journalctl prior to the reboot though when an update is scheduled.  At least in the future it should make debugging easier.

vadi2, 2017-07-06:
Sure, how could I do that?

superm1, 2017-07-06:
I've done that in the above commit.
So i'm going to close this. 
If possible, can you update to current master (at least f043f1f or later) and share `journalctl `output for `fwupd.service` if this happens again?

closed 2017-07-06 by superm1

vadi2, 2017-07-07:
I don't think I'm able to: `Native dependency appstream-glib found: NO found '0.6.9' but need: '>= 0.6.13'` - this is already using latest Ubuntu 17.04.

superm1, 2017-07-07:
I tried to make some modifications to see if it could work with appstream-glib 0.6.9 by disabling the unifying plugin in that situation, but it also will break on the hwids stuff.

I would grab appstream-glib 0.6.13 from 17.10 development for now:
https://launchpad.net/ubuntu/+source/appstream-glib/0.6.13-1ubuntu1

superm1, 2017-07-07:
The the one function call from 0.6.13 is guarded, but it will cause the engine to not load.
https://github.com/hughsie/fwupd/blob/master/src/fu-hwids.c#L45

I made a stab at disabling hwids at https://github.com/hughsie/fwupd/tree/wip/superm1/older-appstream that you can try to compile against.

dragon788, 2018-01-26:
I've had a couple cases on a pretty old BIOS for the Dell XPS 13 9360 where the EFI entries weren't getting correctly added/cleaned up when I would install a different distribution so at one point I had 4 or 5 entries, getting fully up to date appears to have fixed the issue (but I also haven't gone on a reinstall spree like I had been on since then).

superm1, 2018-01-26:
Yeah if there is still an issue, please file it when you encounter it.