fwupd/fwupd #155: Unifying Receiver Fails to Update

byronclark, 2017-07-09:
I have the following unifying receiver device:
```
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  DeviceID:             /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3
  DisplayName:          Unifying Receiver
  Description:          <p>A Unifying receiver allows you to connect multiple compatible keyboards and mice to a laptop or desktop computer with a single USB receiver. Updating the firmware on your Unifying receiver improves performance, adds new features and fixes security issues.</p>
  Plugin:               unifying
  Flags:                allow-online|supported
  DeviceVendor:         Logitech
  DeviceVendorId:       USB:0x046D
  Version:              RQR24.00_B0018
  VersionBootloader:    BOT03.00_B0006
  Created:              2017-07-09
```

I'm using fwupd 0.9.5

I enabled the testing lvfs remote and get this error when attempting to update the device:
```
failed to erase all pages: failed to send data: Device 046d:aaac has not been opened
```

Full verbose output of the update operation:
```
# fwupdmgr --verbose update
Downloading RQR24.05_B0029 for Unifying Receiver...
(fwupdmgr:3827): Fu-DEBUG: downloading https://secure-lvfs.rhcloud.com/downloads/4511b9b0d123bdbe8a2007233318ab215a59dfe6-Logitech-Unifying-RQR24.05_B0029.cab to /root/.cache/fwupdmgr/4511b9b0d123bdbe8a2007233318ab215a59dfe6-Logitech-Unifying-RQR24.05_B0029.cab
Fetching firmware https://secure-lvfs.rhcloud.com/downloads/4511b9b0d123bdbe8a2007233318ab215a59dfe6-Logitech-Unifying-RQR24.05_B0029.cab
(fwupdmgr:3827): Fu-DEBUG: ignoring status code 301 (MOVED PERMANENTLY)
(fwupdmgr:3827): Fu-DEBUG: progress: 25%
Downloading…           [**********                             ](fwupdmgr:3827): Fu-DEBUG: progress: 50%
Downloading…           [********************                   ](fwupdmgr:3827): Fu-DEBUG: progress: 53%
Downloading…           [*********************                  ](fwupdmgr:3827): Fu-DEBUG: progress: 78%
Downloading…           [*******************************        ](fwupdmgr:3827): Fu-DEBUG: progress: 100%
Downloading…           [****************************************]
Updating RQR24.05_B0029 on Unifying Receiver...
(fwupdmgr:3827): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:3827): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               failed to erase all pages: failed to send data: Device 046d:aaac has not been opened
```

hughsie, 2017-07-10:
Can you get a --verbose fwupd daemon log please when you're trying to do the update? I can't see why the device would fail to open after the `doing detach in idle` debug message. Thanks.

byronclark, 2017-07-10:
Here's the verbose log from the daemon:
```
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0250 Fu  Called Install(/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3,8)
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0250 Fu  got option reason
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0250 Fu  got option filename
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0250 Fu  Emitting PropertyChanged('Status'='decompressing')
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0251 As  found file 0        RQR24.05_B0029.hex
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0251 As  found file 1        com.logitech.Unifying.RQR24.metainfo.xml
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0252 As  found file 2        RQR24.05_B0029.hex.asc
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0252 Fu  Using keyring at /var/lib/fwupd/gnupg
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0255 Fu  Adding public key /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0258 Fu  importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0258 Fu  Adding public key /etc/pki/fwupd/GPG-KEY-Hughski-Limited
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0262 Fu  importing key 7E5439F64986F7A9E973809BAD8A528FEC44881E [0] Success
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0266 Fu  returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0266 Fu  marking payload as trusted
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0269 Fu  requirement 0.8.2 ge 0.9.5 on org.freedesktop.fwupd passed
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0269 Fu  no requirement on firmware{(null)}
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0269 Fu  requirement BOT03.0[0-1]_* regex BOT03.00_B0006 on bootloader passed
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0269 Fu  no requirement on firmware{vendor-id}
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0269 Fu  performing update_prepare() on upower
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0269 Fu  FuPending: trying to open database '/var/lib/fwupd/pending.db'
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0270 Fu  FuPending: get res
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0270 Unifying doing detach in idle
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0270 Unifying detaching
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0270 Unifying detaching device
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0270 Unifying [HID] host->device:10 ff 80 f0 49 43 50
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0276 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3 removed
Jul 10 07:39:00 zacapa fwupd[11447]: 13:39:00:0359 Fu  emit removed from usb: usb:00:03
Jul 10 07:39:02 zacapa fwupd[11447]: 13:39:02:0260 Unifying USB add usb:00:03
Jul 10 07:39:02 zacapa fwupd[11447]: 13:39:02:0262 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3 added
Jul 10 07:39:02 zacapa fwupd[11447]: 13:39:02:0262 Unifying waiting for device to settle...
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0263 Unifying opening unifying device using USB
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0263 Unifying claiming interface 0x00
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0264 Unifying host->device:    80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0264 Unifying UDEV add /dev/hidraw1 = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:046D:AAAC.000C/hidraw/hidraw1
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0265 Unifying no HID_ID, skipping
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0267 Unifying device->host:    80 00 00 06 04 00 6b ff 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0267 Unifying host->device:    90 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0268 Unifying device->host:    90 00 00 0e 42 4f 54 30 33 2e 30 30 5f 42 30 30 30 36 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0268 Unifying type:                        bootloader-texas
Jul 10 07:39:03 zacapa fwupd[11447]: flags:                        can-flash,active,is-open,requires-attach
Jul 10 07:39:03 zacapa fwupd[11447]: hidpp-version:                0.00
Jul 10 07:39:03 zacapa fwupd[11447]: hidpp-id:                0xff
Jul 10 07:39:03 zacapa fwupd[11447]: usb-device:                0x15d1f40
Jul 10 07:39:03 zacapa fwupd[11447]: platform-id:                /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3
Jul 10 07:39:03 zacapa fwupd[11447]: vendor:                        Logitech
Jul 10 07:39:03 zacapa fwupd[11447]: product:                Unifying [bootloader-texas]
Jul 10 07:39:03 zacapa fwupd[11447]: version-bootloader:        BOT03.00_B0006
Jul 10 07:39:03 zacapa fwupd[11447]: version-firmware:        RQR24.xx_Bxxxx
Jul 10 07:39:03 zacapa fwupd[11447]: guid:                        USB\VID_046D&PID_AAAC
Jul 10 07:39:03 zacapa fwupd[11447]: flash-addr-high:        0x6bff
Jul 10 07:39:03 zacapa fwupd[11447]: flash-addr-low:        0x0400
Jul 10 07:39:03 zacapa fwupd[11447]: flash-block-size:        0x0080
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0268 Unifying /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3 is in replug, quitting loop
[... cut ...]
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0770 Fu  no requirement on id{org.freedesktop.fwupd}
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0771 Fu  no requirement on firmware{(null)}
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0771 Fu  no paramater given for firmware{bootloader}
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0771 Fu  no requirement on firmware{vendor-id}
Jul 10 07:39:03 zacapa fwupd[11447]: 13:39:03:0771 Fu  ignoring usb:00:03 [USB BootLoader] as not updatable live or offline
```

pkovacs, 2017-07-18:
Getting the very same error here with 0.9.5-1.fc26 on Fedora 26.   I've got an M510 and K350 connected to the Logitech dongle.  

> fwupdmgr update
Downloading RQR12.07_B0029 for Unifying Receiver...
Updating RQR12.07_B0029 on Unifying Receiver...
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened

> lsusb -d 0x046D:
Bus 003 Device 003: ID 046d:aaaa Logitech, Inc.

hkskoglund, 2017-07-19:
I  get device  "046d:aaac"  not opened, but my device is acutally on "046d:c52b" according to lsusb.

[henning@satellite ~]$ sudo fwupdmgr -v install ./.cache/gnome-software/3.24/firmware/4511b9b0d123bdbe8a2007233318ab215a59dfe6-Logitech-Unifying-RQR24.05_B0029.cab
(fwupdmgr:13098): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:13098): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:13098): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:13098): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               failed to erase all pages: failed to send data: Device 046d:aaac has not been opened
[henning@satellite ~]$ lsusb -d 0x046D:
Bus 002 Device 024: ID 046d:c52b Logitech, Inc. Unifying Receiver
[henning@satellite ~]$

hughsie, 2017-07-19:
Can either of you get a --verbose log of fwupd please. Thanks!

hkskoglund, 2017-07-19:
13:45:49:0960 Fu  FuMain: acquired name: org.freedesktop.fwupd
13:45:54:0175 Unifying [HID] host->device:10 01 00 10 00 00 aa 
13:45:55:0725 Unifying [HID] device->host:10 01 8f 00 10 01 00 
13:45:55:0726 Unifying [HID] host->device:10 02 00 10 00 00 aa 
13:45:56:0139 Unifying [HID] device->host:10 02 8f 00 10 01 00 
13:45:56:0139 Unifying [HID] device->host:10 01 8f 00 10 01 00 
13:45:56:0139 Unifying failed to probe /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2: invalid SubID
13:45:59:0178 Unifying [HID] device->host:02 00 00 06 90 fd 00 00 
13:45:59:0178 Unifying report 0x02 unknown length
13:45:59:0178 Unifying [HID] host->device:10 01 00 10 00 00 aa 
13:45:59:0227 Unifying [HID] device->host:02 00 00 00 e0 ff 00 00 
13:45:59:0227 Unifying report 0x02 unknown length
13:45:59:0227 Unifying ignoring message
13:45:59:0233 Unifying [HID] device->host:10 01 8f 00 10 01 00 
13:45:59:0233 Unifying [HID] device->host:01 01 00 00 00 00 00 00 
13:45:59:0233 Unifying [HID] host->device:10 02 00 10 00 00 aa 
13:45:59:0405 Unifying [HID] device->host:10 02 8f 00 10 01 00 
13:45:59:0405 Unifying [HID] device->host:10 02 8f 00 10 01 00 
13:45:59:0405 Unifying failed to probe /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2: invalid SubID
13:46:04:0183 Unifying [HID] device->host:02 00 00 ff 4f ff 00 00 
13:46:04:0183 Unifying report 0x02 unknown length
13:46:04:0183 Unifying [HID] host->device:10 01 00 10 00 00 aa 
13:46:04:0195 Unifying [HID] device->host:10 01 8f 00 10 01 00 
13:46:04:0195 Unifying [HID] device->host:01 01 00 00 00 00 00 00 
13:46:04:0195 Unifying [HID] host->device:10 02 00 10 00 00 aa 
13:46:04:0623 Unifying [HID] device->host:10 02 8f 00 10 01 00 
13:46:04:0623 Unifying [HID] device->host:20 02 01 01 00 00 00 00 00 00 00 00 00 00 00 
13:46:04:0623 Unifying report 0x20 unknown length
13:46:09:0189 Unifying [HID] host->device:10 01 00 10 00 00 aa 
13:46:09:0273 Unifying [HID] device->host:10 01 8f 00 10 01 00 
13:46:09:0275 Unifying [HID] host->device:10 02 00 10 00 00 aa 
13:46:09:0607 Unifying [HID] device->host:10 02 8f 00 10 01 00 
13:46:09:0607 Unifying [HID] device->host:20 02 01 03 00 00 00 00 00 00 00 00 00 00 00 
13:46:09:0607 Unifying report 0x20 unknown length
13:46:14:0194 Unifying [HID] host->device:10 01 00 10 00 00 aa 
13:46:15:0487 Unifying [HID] device->host:10 01 8f 00 10 01 00 
13:46:15:0488 Unifying [HID] host->device:10 02 00 10 00 00 aa 
13:46:15:0837 Unifying [HID] device->host:10 02 8f 00 10 01 00 
13:46:15:0837 Unifying [HID] device->host:20 02 01 03 11 00 00 00 00 00 00 00 00 00 00 
13:46:15:0837 Unifying report 0x20 unknown length
13:46:18:0557 Fu  Called Install(*,7)
13:46:18:0557 Fu  got option reason
13:46:18:0557 Fu  got option filename
13:46:18:0557 Fu  Emitting PropertyChanged('Status'='decompressing')
13:46:18:0558 Fu  Emitting PropertyChanged('Status'='decompressing')
13:46:18:0560 As  found file 0	RQR24.05_B0029.hex
13:46:18:0560 As  found file 1	com.logitech.Unifying.RQR24.metainfo.xml
13:46:18:0561 As  found file 2	RQR24.05_B0029.hex.asc
13:46:18:0586 Fu  Using keyring at /var/lib/fwupd/gnupg
13:46:18:0590 Fu  Adding public key /etc/pki/fwupd/GPG-KEY-Hughski-Limited
13:46:18:0601 Fu  importing key 7E5439F64986F7A9E973809BAD8A528FEC44881E [0] Success
13:46:18:0601 Fu  Adding public key /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
13:46:18:0609 Fu  importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
13:46:18:0613 Fu  returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
13:46:18:0613 Fu  marking payload as trusted
13:46:18:0617 Fu  requirement 0.8.2 ge 0.9.5 on org.freedesktop.fwupd passed
13:46:18:0617 Fu  no requirement on firmware{(null)}
13:46:18:0617 Fu  requirement BOT03.0[0-1]_* regex BOT03.01_B0008 on bootloader passed
13:46:18:0617 Fu  no requirement on firmware{vendor-id}
13:46:18:0617 Fu  performing update_prepare() on upower
13:46:18:0617 Fu  FuPending: trying to open database '/var/lib/fwupd/pending.db'
13:46:18:0619 Fu  FuPending: get res
13:46:18:0619 Unifying doing detach in idle
13:46:18:0619 Unifying detaching
13:46:18:0619 Unifying detaching device
13:46:18:0619 Unifying [HID] host->device:10 ff 80 f0 49 43 50 
13:46:18:0632 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2 removed
13:46:18:0664 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.2/0003:046D:C52B.0049/0003:046D:101B.004A/hidraw/hidraw25 removed
13:46:18:0722 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.2/0003:046D:C52B.0049/0003:046D:2011.004B/hidraw/hidraw26 removed
13:46:18:0725 Fu  emit removed from usb: usb:00:02
[... 196 lines cut ...]
juli 19 15:39:14 satellite kernel: hid-generic 0003:046D:AAAC.0046: hiddev0,hidraw24: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-2/input0
juli 19 15:39:14 satellite mtp-probe[13601]: checking bus 2, device 27: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-2"
juli 19 15:39:15 satellite mtp-probe[13601]: bus: 2, device: 27 was not an MTP device
juli 19 15:39:16 satellite sudo[13570]: pam_unix(sudo:session): session closed for user root
juli 19 15:39:16 satellite audit[13570]: USER_END pid=13570 uid=0 auid=1000 ses=3 msg='op=PAM:session_close grantors=pam_keyinit,pam_limits,pam_keyinit,pam_limits,pam_systemd,pam_unix acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[... cut ...]
juli 19 15:39:22 satellite kernel: logitech-hidpp-device 0003:046D:2011.004B: input,hidraw26: USB HID v1.11 Keyboard [Logitech K520] on usb-0000:00:14.0-2:2
juli 19 15:39:23 satellite fwupd[2450]: report 0x20 unknown length
juli 19 15:39:24 satellite fwupd[2450]: report 0x02 unknown length
juli 19 15:39:24 satellite PackageKit[1471]: get-updates transaction /2528_eadbbdbe from uid 1000 finished with success after 58ms
juli 19 15:39:27 satellite fwupd[2450]: report 0x20 unknown length
juli 19 15:39:27 satellite fwupd[2450]: report 0x02 unknown length

pkovacs, 2017-07-19:
Attaching logs of `fwupdmgr` commands and `journalctl` output from separate terminal while running fwupd in verbose mode.

[fwupd,txt](https://github.com/hughsie/fwupd/files/1160517/fwupd.txt)

intgr, 2017-07-20:
I'm having a problem with Unifying firmware update as well, fwupd 0.9.5 on Arch Linux. In my case the error message says:
```
failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened
```
following which the receiver stops working and needs to be re-plugged manually. This occurs on both of my two Unifying receiver dongles. Both device IDs are also 046d:c52b.

ddreggors, 2017-07-21:
Same issue here in Fedora 26. 
`[root@mobile-tek ~]# fwupdmgr --verbose update
Downloading RQR12.07_B0029 for Unifying [bootloader-nordic]...
(fwupdmgr:6120): Fu-DEBUG: downloading https://secure-lvfs.rhcloud.com/downloads/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab to /root/.cache/fwupdmgr/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab
Fetching firmware https://secure-lvfs.rhcloud.com/downloads/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab
(fwupdmgr:6120): Fu-DEBUG: ignoring status code 301 (MOVED PERMANENTLY)
(fwupdmgr:6120): Fu-DEBUG: progress: 27%
Downloading…           [**********                             ](fwupdmgr:6120): Fu-DEBUG: progress: 54%
Downloading…           [*********************                  ](fwupdmgr:6120): Fu-DEBUG: progress: 56%
Downloading…           [**********************                 ](fwupdmgr:6120): Fu-DEBUG: progress: 84%
Downloading…           [*********************************      ](fwupdmgr:6120): Fu-DEBUG: progress: 100%
Downloading…           [****************************************]
Updating RQR12.07_B0029 on Unifying [bootloader-nordic]...
(fwupdmgr:6120): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:6120): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened
`

hughsie, 2017-07-21:
Just on a hunch, are any of you doing this on 32 bit computers?

ddreggors, 2017-07-21:
Not I, 64 bit here.

pkovacs, 2017-07-21:
x86_64 here

byronclark, 2017-07-21:
Also on x86_64.

Anoian, 2017-07-21:
I have the same problem on gnome wayland on arch, do you want my data, too?
x86_64

pkovacs, 2017-07-21:
I pulled the Logitech dongle out and, looking at the engraved codes on oe side of the metal, I read:

`LZ304BS - DJ`
`M/N:C - U0007`

pkovacs, 2017-07-22:
I was able to perform the firmware upgrade on a different computer -- both running exactly the same setup, Fedora 26 workstation, and both running the same revision of fwupd (0.9.5-1.fc26).  The newer comp where the update was successful is also x86_64, but is a newer Intel i7 system  and uses UEFI.  The older comp, where the update was failing, is a BIOS-based Intel E6850.  I first unplugged the dongle from the old system and switched OFF the keyboard and mouse.  The newer system uses wired accessories.  I then plugged the dongle into a USB port on the new system and ran the update with no problems.  Worked on the first try and `RQR12.07_B0029` was written out with no errors.  I moved the dongle back to the old comp and it's working fine.

> fwupdmgr -v update
Downloading RQR12.07_B0029 for Unifying Receiver...
(fwupdmgr:3186): Fu-DEBUG: downloading https://secure-lvfs.rhcloud.com/downloads/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab to /root/.cache/fwupdmgr/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab
Fetching firmware https://secure-lvfs.rhcloud.com/downloads/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab
(fwupdmgr:3186): Fu-DEBUG: ignoring status code 301 (MOVED PERMANENTLY)
(fwupdmgr:3186): Fu-DEBUG: progress: 27%
Downloading…           [**********                             ](fwupdmgr:3186): Fu-DEBUG: progress: 54%
Downloading…           [*********************                  ](fwupdmgr:3186): Fu-DEBUG: progress: 56%
Downloading…           [**********************                 ](fwupdmgr:3186): Fu-DEBUG: progress: 84%
Downloading…           [*********************************      ](fwupdmgr:3186): Fu-DEBUG: progress: 100%
Downloading…           [****************************************]
Updating RQR12.07_B0029 on Unifying Receiver...
(fwupdmgr:3186): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:3186): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:3186): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:3186): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               [****************************************](fwupdmgr:3186): Fwupd-DEBUG: Emitting ::status-changed() [device-restart]
Restarting device…     [****************************************](fwupdmgr:3186): Fwupd-DEBUG: Emitting ::device-changed((null))
(fwupdmgr:3186): Fwupd-DEBUG: Emitting ::changed()

fnxweb, 2017-07-22:
At the risk of adding a **<me too>** ... erm, me too.  Fedora 26, recently upgraded from 25 (4.11.10-300.fc26.x86_64 / fwupd-0.9.5-1.fc26.x86_64).

```
$ sudo fwupdmgr update --verbose
Downloading RQR12.07_B0029 for Unifying Receiver...
(fwupdmgr:12736): Fu-DEBUG: skpping download as file already exists
Updating RQR12.07_B0029 on Unifying Receiver...
(fwupdmgr:12736): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:12736): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened

```
If you'd like verbose daemon logs, you'll have to explain how(!), all I'm getting in _journalctl_ is a copy of what @pkovacs is seeing:  wads of “**report 0x20 unknown length**” (and some **0x02**s and one “**unknown SubID 81**” which may have come from an update attempt).

I'm not sure it'd be relevant, but the items I have are a K800 keyboard and an MX Master mouse.

jamsla, 2017-07-22:
I'm seeing this today on an up-to-date Arch Linux system (fwupd is version 0.9.5-3).
```
$ fwupdmgr update -v
Downloading RQR12.07_B0029 for Unifying [bootloader-nordic]...
(fwupdmgr:1533): Fu-DEBUG: skpping download as file already exists
Updating RQR12.07_B0029 on Unifying [bootloader-nordic]...
(fwupdmgr:1533): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:1533): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened
```

Here's the output of fwupdmgr get-devices:
```
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  DeviceID:             /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1
  DisplayName:          Unifying [bootloader-nordic]
  Description:          <p>A Unifying receiver allows you to connect multiple compatible keyboards and mice to a laptop or desktop computer with a single USB receiver. Updating the firmware on your Unifying receiver improves performance, adds new features and fixes security issues.</p>
  Plugin:               unifying
  Flags:                allow-online|supported
  DeviceVendor:         Logitech
  DeviceVendorId:       USB:0x046D
  Version:              RQR12.01_B0019
  VersionBootloader:    BOT01.02_B0014
  Created:              2017-07-22

  Guid:                 7039436b-6acf-433b-86a1-368ec2ef7e1f
  DeviceID:             UEFI-7039436b-6acf-433b-86a1-368ec2ef7e1f-dev0
  DisplayName:          MS-7885
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac
  Version:              0.0.2550
  VersionLowest:        0.0.2550
  Created:              2017-07-22

  Guid:                 eb7d0014-915d-548d-b014-23e153b3d6ee
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_03_0_0000_03_00_0
  DisplayName:          GM204 [GeForce GTX 980]
  Plugin:               udev
  Flags:                internal
  DeviceVendor:         NVIDIA Corporation
  DeviceVendorId:       PCI:0x10DE
  Created:              2017-07-22

  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9f1820d3-8b97-5d02-8889-b47b2037247b
  DeviceID:             usb:00:07
  DisplayName:          USB Receiver
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x046D
  Version:              18.1
  Created:              2017-07-22
```
I too have been seeing `fwupd[1420]: report 0x20 unknown length` spammed to the journal every few seconds since.

hughsie, 2017-07-22:
Can any of you guys try with git master? I don't think there's anything Unifying specific in master that's going to fix this, but I'm wondering if the device hash table fix is the true fix for this bug. Thanks.

jamsla, 2017-07-22:
To try that quickly, I just modified the Arch package's PKGBUILD to download master.tar.gz instead of ${pkgver}.tar.gz, and built and installed that (`updpkgsums`/`makepkg -s`/`pacman -U`). After rebooting, I don't see a change:
```
Downloading RQR12.07_B0029 for Unifying Receiver...
(fwupdmgr:1663): Fu-DEBUG: skpping download as file already exists
Updating RQR12.07_B0029 on Unifying Receiver...
(fwupdmgr:1663): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:1663): Fwupd-DEBUG: Emitting ::status-changed() [idle]

(fwupdmgr:1663): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:1663): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:1663): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened
```

bshkrv, 2017-07-22:
At first I had this error
```failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened```
but after trying to update a bunch of times it changed to
```not found /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5.3```

hughsie, 2017-07-22:
So, I spent a few hours trying to debug this just now. I think what's happening is the low-level GUsbDevice is being closed, rather than the LuDevice which wraps it. Basically, the LuDevice is OPEN, but the underlying device is closed. I think it's being closed by another plugin. Debugging continues...

hughsie, 2017-07-22:
Okay, this might be tricky, so well done if you manage to do this for me...

Can you build and install:
 * libgusb, branch `wip/hughsie/GUsbContextFlags`
 * fwupd, branch `wip/hughsie/GUsbContextFlags`

You need to make sure the latter actually links against the former. If this works, please yell and I'll add some more commit text, push them both to master and do a new tarball of the former. Thanks!

jamsla, 2017-07-23:
Sorry for the wait - I'm happy to report that the GUsbContextFlags branches seem to fix the problem:
```
Downloading RQR12.07_B0029 for Unifying Receiver...
(fwupdmgr:1631): Fu-DEBUG: skpping download as file already exists
Updating RQR12.07_B0029 on Unifying Receiver...
(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:1631): Fwupd-DEBUG: Emitting ::status-changed() [idle]

(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::status-changed() [device-write]
Writing…               (fwupdmgr:1631): Fwupd-DEBUG: Emitting ::device-added((null))
(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::changed()
Writing…               [****************************************](fwupdmgr:1631): Fwupd-DEBUG: Emitting ::status-changed() [device-restart]
Restarting device…     [****************************************](fwupdmgr:1631): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::status-changed() [idle]

(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::device-changed((null))
(fwupdmgr:1631): Fwupd-DEBUG: Emitting ::changed()
```
After this, I had no problems using the receiver. When it was failing, I had to extract and re-insert the receiver before I could continue.

Here's the log from the daemon, in case that's of interest: [fwupd.log.gz](https://github.com/hughsie/fwupd/files/1167963/fwupd.log.gz)

Some notes:
   * Firstly, in the daemon log the write process only gets as far as "Unifying written 29568/49696 bytes [59.5%]". I assume that's normal and the remaining space is just unused?
   * Secondly, I'm still seeing a report every five seconds in the journal, although instead of "report 0x20 unknown length" it's now of the form "device->host:20 01 02 00 00 df 0f 00 00 00 00 00 00 00 00". I imagine this is an unrelated matter.

Many thanks for all your effort on this!

hughsie, 2017-07-23:
I'm confused why you're seeing the *debug* log in the journal, that's supposed to be Info/Warning etc.

hughsie, 2017-07-23:
Ohh, and also thanks for the quick testing!

jamsla, 2017-07-23:
Don't worry about the debug log in the journal - I had used "systemctl edit fwupd" to add the --verbose flag. If I revert that, things are quieter :)

hughsie, 2017-07-24:
@pkovacs -- rather a strange question -- is the root file-system different between the BIOS and UEFI machines, perchance?

hughsie, 2017-07-24:
If you're a Fedora user, karma appreciated for the update: https://bodhi.fedoraproject.org/updates/FEDORA-2017-06c7317f1f

pkovacs, 2017-07-24:
@hughsie Both computers are using lvm2 logical volumes for the root filesystem with ext4 as the real file system underneath.  The UEFI system is ssd-based and I have the `noatime` bit set on the filesystem.   The BIOS system is using an ordinary disk drive.   Looks like you have closed the issue in any event.

SinBirb, 2017-07-24:
Can you please release a new version so that other distros can also update? I can't manage to build libgusb and linking it to fwupd. Thank you.

hughsie, 2017-07-24:
I did a new upstream libgusb today. A new fwupd will be a few more days.

hkskoglund, 2017-07-27:
I tried the new libgusb package, and I am unable to install firmware.

[henning@satellite ~]$ sudo dnf info libgusb
Last metadata expiration check: 0:59:00 ago on to. 27. juli 2017 kl. 11.22 +0200.
Pakker som er installert
Name         : libgusb
Versjon      : 0.2.11
Release      : 1.fc26
Arkitektur   : x86_64
Størrelse    : 108 k
Source       : libgusb-0.2.11-1.fc26.src.rpm
Repo         : @System
From repo    : updates-testing
Summary      : GLib wrapper around libusb1
URL          : https://gitorious.org/gusb/
License      : LGPLv2+
Description  : GUsb is a GObject wrapper for libusb1 that makes it easy to do
             : asynchronous control, bulk and interrupt transfers with proper
             : cancellation and integration into a mainloop.

Then I enabled debugging environement LIBUSB_DEBUG 4 on fwupd

sudo env LIBUSB_DEBUG=4 /usr/libexec/fwupd/fwupd -v and ran fwupdmgr -v update

I see three lines right after another, it seems like libusb is closing a device, probably 0x46D:AAAC? and then device write fails.

10:19:14:0652 Fu  using cc4cbfa9-bf9d-540b-b92b-172ce31013c1 for USB\VID_046D&PID_AAAC
10:19:14:0652 Fu  using de81631f-2117-51f8-bcb1-4f874a293b29 for USB\VID_046D&PID_AAAC&REV_0301
[ 8.654633] [00005431] libusb: debug [libusb_close] 
...
10:19:14:0655 Unifying host->device:    d0 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

10:19:09:0534 Fu  FuMain: acquired name: org.freedesktop.fwupd
10:19:12:0149 Fu  Called GetUpdates()
10:19:12:0149 Fu  no requirement on id{org.freedesktop.fwupd}
10:19:12:0149 Fu  no requirement on firmware{(null)}
10:19:12:0149 Fu  no requirement on firmware{bootloader}
10:19:12:0149 Fu  no requirement on firmware{vendor-id}
10:19:12:0150 Fu  Called GetRemotes()
10:19:12:0152 Fu  Called Install(/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1,7)
10:19:12:0152 Fu  got option reason
10:19:12:0152 Fu  got option filename
10:19:12:0152 Fu  Emitting PropertyChanged('Status'='decompressing')
10:19:12:0152 Fu  Emitting PropertyChanged('Status'='decompressing')
10:19:12:0153 As  found file 0	RQR24.05_B0029.hex
10:19:12:0153 As  found file 1	com.logitech.Unifying.RQR24.metainfo.xml
10:19:12:0154 As  found file 2	RQR24.05_B0029.hex.asc
10:19:12:0167 Fu  Using keyring at /var/lib/fwupd/gnupg
10:19:12:0170 Fu  Adding public key /etc/pki/fwupd/GPG-KEY-Hughski-Limited
10:19:12:0175 Fu  importing key 7E5439F64986F7A9E973809BAD8A528FEC44881E [0] Success
10:19:12:0175 Fu  Adding public key /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
10:19:12:0183 Fu  importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
10:19:12:0201 Fu  returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
10:19:12:0201 Fu  marking payload as trusted
10:19:12:0203 Fu  requirement 0.8.2 ge 0.9.5 on org.freedesktop.fwupd passed
10:19:12:0203 Fu  no requirement on firmware{(null)}
10:19:12:0203 Fu  requirement BOT03.0[0-1]_* regex BOT03.01_B0008 on bootloader passed
10:19:12:0203 Fu  no requirement on firmware{vendor-id}
10:19:12:0203 Fu  performing update_prepare() on upower
10:19:12:0203 Fu  FuPending: trying to open database '/var/lib/fwupd/pending.db'
10:19:12:0203 Fu  FuPending: get res
10:19:12:0204 Unifying doing detach in idle
10:19:12:0204 Unifying detaching
10:19:12:0204 Unifying detaching device
10:19:12:0204 Unifying [HID] host->device:10 ff 80 f0 49 43 50 
10:19:12:0210 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1 removed
10:19:12:0234 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/0003:046D:C52B.002C/0003:046D:101B.002D/hidraw/hidraw1 removed
10:19:12:0290 Unifying device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/0003:046D:C52B.002C/0003:046D:2011.002E/hidraw/hidraw2 removed
[ 6.293625] [00005433] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 1
[ 6.293644] [00005433] libusb: debug [udev_hotplug_event] udev hotplug event. action: remove.
[... 179 lines cut ...]
10:19:15:0154 Fu  no requirement on id{org.freedesktop.fwupd}
10:19:15:0154 Fu  no requirement on firmware{(null)}
10:19:15:0154 Fu  no paramater given for firmware{bootloader}
10:19:15:0154 Fu  no requirement on firmware{vendor-id}
10:19:15:0154 Fu  ignoring usb:00:01 [USB BootLoader] as not updatable live or offline
10:19:18:0055 Fu  Called GetUpdates()
10:19:18:0055 Fu  no requirement on id{org.freedesktop.fwupd}
10:19:18:0055 Fu  no requirement on firmware{(null)}
10:19:18:0055 Fu  no requirement on firmware{bootloader}
10:19:18:0055 Fu  no requirement on firmware{vendor-id}
10:19:18:0056 Fu  no requirement on id{org.freedesktop.fwupd}
10:19:18:0056 Fu  no requirement on firmware{(null)}
10:19:18:0056 Fu  no paramater given for firmware{bootloader}
10:19:18:0056 Fu  no requirement on firmware{vendor-id}
10:19:18:0056 Fu  ignoring usb:00:01 [USB BootLoader] as not updatable live or offline
10:19:18:0169 Fu  Called GetUpdates()
10:19:18:0169 Fu  no requirement on id{org.freedesktop.fwupd}
10:19:18:0169 Fu  no requirement on firmware{(null)}
10:19:18:0169 Fu  no requirement on firmware{bootloader}
10:19:18:0169 Fu  no requirement on firmware{vendor-id}
10:19:18:0169 Fu  no requirement on id{org.freedesktop.fwupd}
10:19:18:0169 Fu  no requirement on firmware{(null)}
10:19:18:0169 Fu  no paramater given for firmware{bootloader}
10:19:18:0169 Fu  no requirement on firmware{vendor-id}
10:19:18:0169 Fu  ignoring usb:00:01 [USB BootLoader] as not updatable live or offline

hughsie, 2017-07-27:
@hkskoglund did you compile fwupd form master against the new libgusb?

hkskoglund, 2017-07-27:
No, I just tried libgusb fix with latest fwupd 0.9.5.1.fc26. I'll just wait for the next release of fwupd before doing further testing.

hkskoglund, 2017-07-27:
OK, I managed to build master of fwupd and it worked!

[henning@satellite ~]$ sudo fwupdmgr update
Downloading RQR24.05_B0029 for Unifying Receiver...
Updating RQR24.05_B0029 on Unifying Receiver...
Decompressing…         
Restarting device…     [****************************************]

thibautd, 2017-08-02:
Hi,

I'm sorry but I installed latest version of libgusb and fwupdmgr from Fedora updates-testing and I still have the issue. Maybe I missed something, but it seems that I'm using the correct versions:

libgusb-0.2.11.1.fc26
fwupd-0.9.5.1.fc26

Thanks for your help.

hughsie, 2017-08-02:
> it seems that I'm using the correct versions
> fwupd-0.9.5.1.fc26

This needs to be the unreleased 0.9.6...

hughsie, 2017-08-03:
New build on https://bodhi.fedoraproject.org/updates/FEDORA-2017-148837efa0 -- karma appreciated.

ZeSmoke, 2017-08-03:
Hi, the new build 0.9.6-1 did the job like a charm on my Fedora 26 WorkStation x86_64.

Great job ! Thx

roshanshariff, 2017-08-03:
I tried to leave this comment on Bodhi, but couldn't:

After this update, the fwupd service doesn't start when a Logitech Extreme 3D Pro joystick is connected. The device in question is:
Bus 003 Device 022: ID 046d:c215 Logitech, Inc. Extreme 3D Pro

With 0.9.5 the daemon starts but occasionally prints the following message
fwupd[7002]: report 0x02 unknown length

With the new version 0.9.6, the daemon starts up correctly when the joystick is not plugged in, but prints the output below when it is (the last few lines repeat forever).

However, the new version does successfully update a Logitech Unifying receiver, which 0.9.5 didn't (as long as I unplug the joystick)

$ sudo /usr/libexec/fwupd/fwupd --verbose
21:25:17:0638 Fu  Verbose debugging enabled (on console 1)
21:25:17:0639 Fu  loading config values from /etc/fwupd.conf
21:25:17:0639 Fu  using config path of /etc/fwupd
21:25:17:0639 Fu  loading from /etc/fwupd/remotes.d/fwupd.conf
21:25:17:0639 Fu  loading from /etc/fwupd/remotes.d/lvfs-testing.conf
21:25:17:0640 Fu  loading from /etc/fwupd/remotes.d/lvfs.conf
21:25:17:0640 Fu  loading from /etc/fwupd/remotes.d/vendor.conf
21:25:17:0640 Fu  ordering lvfs-testing=lvfs+1
21:25:17:0640 Fu  ordering lvfs=fwupd+1
21:25:17:0640 Fu  ordering lvfs-testing=lvfs+1
21:25:17:0640 Fu  remote lvfs-testing not enabled, so skipping
21:25:17:0640 As  run AsStore:store-from-file{/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz}
21:25:17:0642 As  run AsStore:store-from-root
21:25:17:0642 As  using icon path /var/lib/fwupd/remotes.d/lvfs/icons
21:25:17:0642 As  run AsStore:match-addons
21:25:17:0642 As  Emitting ::changed() [from-root]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  run AsStore:store-from-file{/usr/share/fwupd/remotes.d/fwupd/metadata.xml}
21:25:17:0643 As  run AsStore:store-from-root
21:25:17:0643 As  using icon path /usr/share/fwupd/remotes.d/fwupd/icons
21:25:17:0643 As  run AsStore:match-addons
21:25:17:0643 As  Emitting ::changed() [from-root]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 As  Emitting ::changed() [add-app]
21:25:17:0643 Fu  remote vendor not enabled, so skipping
21:25:17:0643 Fu  devices now in store:
21:25:17:0643 Fu  1	com.dell.uefi5b7ab884.firmware	Precision E7X10 System Update
21:25:17:0643 Fu  2	com.8bitdo.fc30.firmware	FC30
21:25:17:0643 Fu  3	com.dell.uefi9cb573d5.firmware	Latitude 3480 System Update
21:25:17:0643 Fu  4	com.dell.uefi6180aaaa.firmware	Latitude E5X60 System Update
21:25:17:0643 Fu  5	com.dell.uefia5d83d78.firmware	Precision 5720 AIO System Update
[... 140 lines cut ...]
21:25:17:0691 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0699 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0699 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0707 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0707 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0715 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0715 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0723 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0723 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0731 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0731 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0739 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0739 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0747 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0747 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0755 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0755 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0763 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0763 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0771 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0771 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0779 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0779 Unifying HID++1.0 report 0x00 has unknown length, ignoring
21:25:17:0787 Unifying [HID] device->host:00 f2 87 80 00 00 00 
21:25:17:0787 Unifying HID++1.0 report 0x00 has unknown length, ignoring

hughsie, 2017-08-04:
The joystick seems to be spamming the daemon with "00 f2 87 80 00 00 00" -- I'll try to work this out over the weekend.

thibautd, 2017-08-05:
Fixed the issue for me also, added a +1 Karma. Thanks for quick fix !

Goddard, 2019-12-07:
I don't know if this is a regression, or just similar error, but on Ubuntu 19.10 I am getting this exact error.

failed to get device after detach: failed to wait for detach replug: device 3414db784a57169c10b1f4ec7253baa6a637e80d did not come back

The device is disabled when trying to install the update.  I have to manually unplug and replug it back in for my keyboard to start working again.

Maybe not related, but I found this post on a search.

FFY00, 2019-12-07:
You need to replug the receiver during the update.

Goddard, 2019-12-07:
I tried, but ever time I replug the device it just gives the a not supported.

Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc

Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

No upgrades for Unifying Receiver, current is RQR12.10_B0032: RQR12.05_B0028=older, RQR12.10_B0032=same, RQR12.08_B0030=older, RQR12.07_B0029=older
Downloading RQR12.10_B0032 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Unifying Receiver…
Idle…                    [***************************************]
not supported

Goddard, 2019-12-07:
multiple different ways

Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc

Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

ignoring Unifying Receiver [71413822e87deae707da948bd4b7e537af463c19] as not updatable

Goddard, 2019-12-07:
Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc

Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Downloading RQR12.10_B0032 for Unifying Receiver...
(fwupdmgr:40236): FuCommon-DEBUG: 12:05:40.698: creating path /root/.cache/fwupd
(fwupdmgr:40236): FuMain-DEBUG: 12:05:40.699: skpping download as file already exists
(fwupdmgr:40236): Fwupd-DEBUG: 12:05:40.699: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:40.700: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:40236): Fwupd-DEBUG: 12:05:40.733: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:40.735: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
Updating Unifying Receiver…
Idle…                    [***************************************]
(fwupdmgr:40236): Fwupd-DEBUG: 12:05:42.192: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
(fwupdmgr:40236): Fwupd-DEBUG: 12:05:42.194: Emitting ::status-changed() [device-erase]
Erasing…                 [  -                                    ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:42.195: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Erasing…                 [                               /       ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.392: Emitting ::status-changed() [device-write]
Writing…                 [                               /       ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.393: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [                                       ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.413: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [                                       ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.445: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*                                      ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.477: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*                                      ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.513: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*                                      ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.545: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [**                                     ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.577: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [**                                     ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.613: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [***                                    ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.645: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [***                                    ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.677: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [***                                    ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.709: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [****                                   ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.745: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [****                                   ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.777: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*****                                  ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.809: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*****                                  ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.845: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*****                                  ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.877: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [******                                 ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.909: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [******                                 ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.941: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*******                                ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:43.977: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
Writing…                 [*******                                ](fwupdmgr:40236): Fwupd-DEBUG: 12:05:44.009: Emitting ::device-changed(71413822e87deae707da948bd4b7e537af463c19)
[... cut ...]
Writing…                 [***************************************](fwupdmgr:40236): Fwupd-DEBUG: 12:05:49.402: Emitting ::device-changed(3414db784a57169c10b1f4ec7253baa6a637e80d)
(fwupdmgr:40236): Fwupd-DEBUG: 12:05:49.402: Emitting ::status-changed() [idle]
Writing…                 [***************************************]
(fwupdmgr:40236): Fwupd-DEBUG: 12:05:49.403: Emitting ::changed()
No upgrades for Unifying Receiver, current is RQR12.10_B0032: RQR12.05_B0028=older, RQR12.10_B0032=same, RQR12.08_B0030=older, RQR12.07_B0029=older

superm1, 2019-12-07:
Can you please run the daemon in verbose mode and provide the output from the daemon during this attempt?

Goddard, 2019-12-07:
I'd be happy to.  How?  Past this

sudo fwupdmgr refresh && sudo fwupdmgr -vvv update

I have no idea how to do anything else with your software.

superm1, 2019-12-09:
@Goddard As mentioend in https://github.com/fwupd/fwupd/wiki/Getting-involved you can use:

```
# fwupdmgr modify-config VerboseDomains '*'
```

superm1, 2019-12-09:
And then the system journal for the daemon (`fwupd.service` for non-snap and `snap.fwupd.fwupd.service` for snap)

a-andre, 2020-04-21:
Here is the log for the same error as reported by @Goddard.
```
failed to get device after detach: failed to wait for detach replug: device 75f2cedf3387d4d43fc6cd30d11c6e53f47d1a7 did not come back
```
journalctl output:
```
fwupd[11437]: 18:06:29:0494 FuMain               Called GetRemotes()
fwupd[11437]: 18:06:29:0500 FuMain               Called Install(e75f2cedf3387d4d43fc6cd30d11c6e53f47d1a7,7)
fwupd[11437]: 18:06:29:0500 FuMain               got option reason
fwupd[11437]: 18:06:29:0500 FuMain               got option filename
fwupd[11437]: 18:06:29:0500 FuEngine             Emitting PropertyChanged('Status'='decompressing')
fwupd[11437]: 18:06:29:0500 FuMain               Emitting PropertyChanged('Status'='decompressing')
fwupd[11437]: 18:06:29:0502 FuCabinet            processing folder: 1/1
fwupd[11437]: 18:06:29:0502 FuCabinet            failed to verify com.logitech.Unifying.RQR24.metainfo.xml: no JcatItem
fwupd[11437]: 18:06:29:0502 FuCabinet            processing file: com.logitech.Unifying.RQR24.metainfo.xml
fwupd[11437]: 18:06:29:0502 XbSilo               compiling 463b210def7303f7d40e5a0be9aaefd651f453ed:prefix=components…
fwupd[11437]: 18:06:29:0502 XbSilo               0,28ms      compile 463b210def7303f7d40e5a0be9aaefd651f453ed:prefix=components
fwupd[11437]: 18:06:29:0502 FuCabinet            invalid container checksum (null), fixing up to dcb67c5c079971ee5fc1662c42dea4bee8e1165d
fwupd[11437]: 18:06:29:0502 XbSilo               0,47ms      get size nodetab
fwupd[11437]: 18:06:29:0503 XbSilo               0,04ms      adding strtab element
fwupd[11437]: 18:06:29:0503 XbSilo               0,02ms      adding strtab attr name
fwupd[11437]: 18:06:29:0503 XbSilo               0,03ms      adding strtab attr value
fwupd[11437]: 18:06:29:0503 XbSilo               0,05ms      adding strtab text
fwupd[11437]: 18:06:29:0503 XbSilo               0,02ms      writing nodetab
fwupd[11437]: 18:06:29:0503 XbSilo               0,02ms      fixing ->parent and ->next
fwupd[11437]: 18:06:29:0503 XbSilo               0,00ms      appending strtab
fwupd[11437]: 18:06:29:0503 XbSilo               0,02ms      parse blob
fwupd[11437]: 18:06:29:0503 XbSilo               0,02ms      query on / with `components/component` limit=0 -> 1 results
fwupd[11437]: 18:06:29:0503 XbSilo               0,01ms      query on component with `releases/release` limit=0 -> 1 results
fwupd[11437]: 18:06:29:0503 FuCabinet            processing release: RQR24.11_B0036
fwupd[11437]: 18:06:29:0503 XbSilo               0,06ms      query on release with `checksum[@target='content']` limit=1 -> 1 results
fwupd[11437]: 18:06:29:0518 Jcat                 returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
fwupd[11437]: 18:06:29:0518 FuCabinet            verified payload RQR24.11_B0036.shex using detached
fwupd[11437]: 18:06:29:0518 FuEngine             Emitting PropertyChanged('Status'='idle')
fwupd[11437]: 18:06:29:0518 FuMain               Emitting PropertyChanged('Status'='idle')
fwupd[11437]: 18:06:29:0518 XbSilo               0,03ms      query on / with `components/component` limit=0 -> 1 results
fwupd[11437]: 18:06:29:0518 XbSilo               0,06ms      query on component with `provides/firmware[@type='flashed']` limit=0 -> 4 results
fwupd[11437]: 18:06:29:0518 XbSilo               0,04ms      query on component with `custom/value[@key='LVFS::UpdateProtocol']` limit=1 -> 1 results
fwupd[11437]: 18:06:29:0518 XbSilo               0,00ms      query on component with `releases/release` limit=1 -> 1 results
fwupd[11437]: 18:06:29:0518 XbSilo               0,01ms      query on component with `requires/*` limit=0 -> 7 results
fwupd[11437]: 18:06:29:0518 FuEngine             requirement 1.3.1 ge 1.4.0 on org.freedesktop.fwupd passed
fwupd[11437]: 18:06:29:0518 XbSilo               0,02ms      query on component with `provides/firmware[@type='flashed']` limit=0 -> 4 results
fwupd[11437]: 18:06:29:0518 XbSilo               0,01ms      query on component with `custom/value[@key='LVFS::UpdateProtocol']` limit=1 -> 1 results
fwupd[11437]: 18:06:29:0518 XbSilo               0,00ms      query on component with `releases/release` limit=1 -> 1 results
fwupd[11437]: 18:06:29:0518 XbSilo               0,01ms      query on component with `custom/value[@key='LVFS::VersionFormat']` limit=0 -> 0 results
fwupd[11437]: 18:06:29:0519 XbSilo               0,00ms      query on component with `requires/*` limit=0 -> 7 results
fwupd[11437]: 18:06:29:0519 FuEngine             requirement 1.3.1 ge 1.4.0 on org.freedesktop.fwupd passed
fwupd[11437]: 18:06:29:0519 XbSilo               0,02ms      query on component with `custom/value[@key='LVFS::UpdateMessage']` limit=1 -> 0 results
fwupd[11437]: 18:06:29:0519 FuMain               Emitting PropertyChanged('Status'='waiting-for-auth')
fwupd[11437]: 18:06:29:0525 FuMain               Emitting PropertyChanged('Status'='idle')
fwupd[11437]: 18:06:29:0525 FuIdle               inhibiting: performing update
fwupd[11437]: 18:06:29:0526 FuPlugin             performing composite_prepare() on dell_dock
fwupd[11437]: 18:06:29:0526 XbSilo               0,02ms      query on component with `releases/release` limit=1 -> 1 results
fwupd[11437]: 18:06:29:0526 Fwupd                looking for os-release at /etc/os-release
fwupd[11437]: 18:06:29:0526 XbSilo               0,08ms      query on component with `releases/release/checksum[@target='container']` limit=1 -> 1 results
[... cut ...]
fwupd[11437]: 18:06:39:0563 FuPlugin             performing composite_cleanup() on dell_dock
fwupd[11437]: 18:06:39:0563 FuIdle               uninhibiting: performing update
fwupd[11437]: 18:06:39:0564 FuDeviceList         doing delayed removal
fwupd[11437]: 18:06:39:0564 FuDeviceList         ::removed e75f2cedf3387d4d43fc6cd30d11c6e53f47d1a7
fwupd[11437]: 18:06:39:0564 FuPlugin             performing device_removed() on dell_dock
```

Replug didn't work and just reported `not supported`.

* fwupd 1.4.0
* libgusb 0.3.4

superm1, 2020-04-21:
@Goddard can you please open a new issue with all of your circumstances and logs?  The root cause of this one is definitely fixed, but there

superm1, 2020-04-21:
What state was the device actually in after that?  Was your peripheral working?

a-andre, 2020-04-21:
Before: `Bus 002 Device 043: ID 046d:c52b Logitech, Inc. Unifying Receiver`
After:    `Bus 002 Device 044: ID 046d:aae5 Logitech, Inc. USB BootLoader`

Device stopped working. I had to replug the receiver for device to work again.

closed 2020-05-28 by stale[bot]

Slater91, 2021-05-11:
The issue is still present as of today. This is the output I can see in journalctl when trying to update the unifying receiver that came with my Logitech M570. I am using KDE neon, based on Ubuntu 20.04.

```
May 11 11:43:36 Sleipnir dbus-daemon[1027]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.644' (uid=1000 pid=155920 comm="/usr/bin/plasma-discover ")
May 11 11:43:36 Sleipnir systemd[1]: Starting Firmware update daemon...
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0812 FuRemoteList         loading /usr/share/fwupd/metainfo
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0813 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0813 XbSilo               file: a45a29b0-39ba-2c30-f770-e8baeb751e6c, current:a45a29b0-39ba-2c30-f770-e8baeb751e6c, cached: (null)
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0813 XbSilo               loading silo with file contents
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0814 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd.conf
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0814 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0815 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0815 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0815 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0816 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0816 FuRemoteList         ordering lvfs=fwupd+1
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0816 FuRemoteList         ordering lvfs-testing=lvfs+1
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0816 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0816 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 loaded 1 certificates
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0817 Jcat                 loaded 1 certificates
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0818 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0818 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0823 FuEngine             client certificate exists and working
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHistory            got schema version of 6
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuIdle               setting timeout to 7200s
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuEngine             Failed to load SMBIOS: structure larger than available data
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property Manufacturer=HP
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property EnclosureKind=a
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property Family=103C_5336AN HP EliteBook
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property ProductName=HP EliteBook 745 G5
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property ProductSku=3UP64EA#ABZ
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property BiosVendor=HP
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property BiosVersion=Q81 Ver. 01.15.00
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property BiosMajorRelease=0f
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property BiosMinorRelease=00
May 11 11:43:36 Sleipnir fwupd[165305]: 10:43:36:0824 FuHwids              smbios property BaseboardManufacturer=HP
[... cut ...]
May 11 11:43:39 Sleipnir fwupd[165305]: 10:43:39:0370 FuPlugin             074997a4-cfbd-5d2b-b3a6-39c3941b4d49,c4d2c6c2-9f84-5fb8-95d6-cea26f9d38a2,f84d902d-6dd5-5830-b636-02c13e08ae08 has no updates, so ignoring device
May 11 11:43:47 Sleipnir fwupd[165305]: 10:43:47:0753 FuHistory            modifying device Unifying Receiver [01fedfb8ce801a925bd52c8ff794a1e3795f7a55]
May 11 11:43:47 Sleipnir fwupd[165305]: 10:43:47:0761 FuPlugin             composite_cleanup(dell_dock)
May 11 11:43:47 Sleipnir fwupd[165305]: 10:43:47:0761 FuIdle               uninhibiting: update
```
I am using fwupd 1.5.7 and libgusb2 0.3.6.

Slater91, 2021-05-11:
> Does fwupd 1.5.9 fix this?

I tried updating using Snap, where version 1.6.0 is available, and indeed it does fix this.

rayfoss, 2021-07-23:
This bug crashes plasma-discover, which I use for Snapd free deb+flatpak management with plasma-discover-flatpak-backend.

mYnDstrEAm, 2022-02-04:
Just found this issue: what is the solution to "failed to get device after detach: failed to wait for detach replug: device did not come back"? I asked about it [here](https://unix.stackexchange.com/questions/689284/how-to-solve-failed-to-get-device-after-detach-failed-to-wait-for-detach-replu).

[3 more replies]