fwupd/fwupd #156: fwupd crashed with SIGSEGV in fu_plugin_coldplug()

cagriar, 2017-07-13:
Hi, I've just upgraded to Ubuntu 17.04 and I'm facing "fwupd crashed with SIGSEGV in fu_plugin_coldplug()" error on every startup. The version is fwupd 0.8.1-3.

SegvAnalysis:
 Segfault happened at: 0x7f128c122741 <fu_plugin_coldplug+673>:	mov    0x8(%rax),%rcx
 PC (0x7f128c122741) ok
 source "0x8(%rax)" (0x00000008) not located in a known VMA region (needed readable region)!
 destination "%rcx" ok

Thanks,

hughsie, 2017-07-13:
Can you get a backtrace please. Thanks.

cagriar, 2017-07-13:
#0  0x00007f128c122741 in fu_plugin_coldplug ()
   from /usr/lib/x86_64-linux-gnu/fwupd-plugins-2/libfu_plugin_synapticsmst.so
No symbol table info available.
#1  0x00005577159f7eee in fu_plugin_runner_coldplug ()
No symbol table info available.
#2  0x00005577159f9649 in ?? ()
No symbol table info available.
#3  0x00005577159f414d in main ()
No symbol table info available.

superm1, 2017-07-13:
Can you please install ddebs:
https://github.com/hughsie/fwupd/wiki/Ubuntu-backtracing

cagriar, 2017-07-13:
Unfortunately, the ddebs did not work.

superm1, 2017-07-13:
Did you install the right ones for 17.10 and your architecture?  Those links on the wiki are a bit confusing since they're for 16.10, amd64.

superm1, 2017-07-13:
That said - my best guess is you're running into this:
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1671570

Can you install the proposed package and see if it fixes your bug?  It's waiting for feedback from someone affected for a while now.

cagriar, 2017-07-13:
Which ddebs should I download from http://ddebs.ubuntu.com/ubuntu/pool/main/f/fwupd/ 
I'm sorry I'm a bit confused.

superm1, 2017-07-13:
Well I suspect you might have that same bug I linked above which already has a proposed fix but is waiting for someone to confirm that fix works for them.  Rather than install ddebs and backtrace can you follow https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1671570/comments/27 to test that fix?

cagriar, 2017-07-13:
I'm following https://wiki.ubuntu.com/Testing/EnableProposed and I can run sudo aptitude -t xenial-proposed successfully. How can I install that proposed fix?

cagriar, 2017-07-13:
I think I made it through `sudo apt-get install fwupd/zesty-proposed`

superm1, 2017-07-13:
Yeah that should do it.  You'll need to `sudo systemctl restart fwupd.service `after that or reboot.

cagriar, 2017-07-13:
I've completed installation and did several reboots. fwupd/zesty-proposed update seems to work fine up to now. Thanks.

closed 2017-07-13 by cagriar

superm1, 2017-07-13:
Great thanks.  Please add a comment to https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1671570 so this can move along and get promoted to -updates.