fwupd/fwupd #157: fwupd 0.9.5-1.fc26 crashes

Herbert-Hotz, 2017-07-15:
Hi,
I want just inform you that running Fedora 26 the executing the following command as root crashes:
**# fwupdmgr refresh**

Note:
- Hardware UDOO X26 Plus Advanced
- Installed Fedora 26 from [https://download.fedoraproject.org/pub/fedora-secondary/releases/26/Workstation/i386/iso/Fedora-Workstation-Live-i386-26-1.5.iso](https://download.fedoraproject.org/pub/fedora-secondary/releases/26/Workstation/i386/iso/Fedora-Workstation-Live-i386-26-1.5.iso) - 
- Installed all available updates with `dnf update`

`journalctl -f
Jul 15 12:37:24 localhost.localdomain dbus-daemon[709]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.114' (uid=0 pid=3475 comm="fwupdmgr refresh " label="unconfined_u:unconfined_r:unconfined_t:s0-s0:c0.c1023")
Jul 15 12:37:24 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 15 12:37:25 localhost.localdomain fwupd[3478]: disabling plugin because: failed to startup amt: ME refused connection
Jul 15 12:37:25 localhost.localdomain fwupd[3478]: disabling plugin because: failed to startup dell: Firmware updating not supported
Jul 15 12:37:30 localhost.localdomain fwupd[3478]: g_object_new_valist: object class 'LuDeviceRuntime' has no property named '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3'
Jul 15 12:37:30 localhost.localdomain kernel: fwupd[3478]: segfault at 0 ip b759abba sp bfc94e08 error 4 in libglib-2.0.so.0.5200.3[b7561000+119000]
Jul 15 12:37:30 localhost.localdomain audit[3478]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 subj=system_u:system_r:fwupd_t:s0 pid=3478 comm="fwupd" exe="/usr/libexec/fwupd/fwupd" sig=11 res=1
Jul 15 12:37:30 localhost.localdomain fwupd[3478]: value "13047682544382771521" of type 'guint64' is invalid or out of range for property 'flags' of type 'guint64'
Jul 15 12:37:30 localhost.localdomain systemd[1]: Started Process Core Dump (PID 3483/UID 0).
Jul 15 12:37:30 localhost.localdomain audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@4-3483-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 15 12:37:30 localhost.localdomain audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Jul 15 12:37:30 localhost.localdomain systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jul 15 12:37:30 localhost.localdomain systemd[1]: Failed to start Firmware update daemon.
Jul 15 12:37:30 localhost.localdomain systemd[1]: fwupd.service: Unit entered failed state.
Jul 15 12:37:30 localhost.localdomain systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jul 15 12:37:30 localhost.localdomain systemd-coredump[3485]: Process 3478 (fwupd) of user 0 dumped core.
                                                              
                                                              Stack trace of thread 3478:
                                                              #0  0x00000000b759abba g_str_hash (libglib-2.0.so.0)
                                                              #1  0x00000000b7599699 g_hash_table_insert_internal (libglib-2.0.so.0)
                                                              #2  0x00000000b512144b lu_context_add_udev_device (libfu_plugin_unifying.so)
                                                              #3  0x00000000b5121c80 lu_context_coldplug (libfu_plugin_unifying.so)
                                                              #4  0x00000000b511f8fe fu_plugin_startup (libfu_plugin_unifying.so)
                                                              #5  0x00000000800f216f fu_plugin_runner_startup (fwupd)
                                                              #6  0x00000000800ea294 fu_engine_load (fwupd)
                                                              #7  0x00000000800e1bd0 main (fwupd)
                                                              #8  0x00000000b70372f3 __libc_start_main (libc.so.6)
                                                              #9  0x00000000800e1ed6 _start (fwupd)
Jul 15 12:37:30 localhost.localdomain audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@4-3483-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 15 12:37:31 localhost.localdomain abrt-dump-journal-core[749]: Failed to obtain all required information from journald
`

hughsie, 2017-07-15:
Hi!

Can you get a proper backtrace using gdb or using abrt please? I need
to see the values of the variables passed to g_hash_table_insert.
Thanks!

Richard

Herbert-Hotz, 2017-07-15:
Hi Richard,

Thank's for responding so quick!

Can you please provide instructions how to produce things yo request- cause I don't know how to.
Would eventually help to send yo the crash dump file?

Kind regards,
Herbert

hughsie, 2017-07-17:
This should be fixed by e8b5db66f4f65e9145be5c72e3af1d320defa836 -- I could reproduce the crash and it no longer happens on my 32bit VM.

closed 2017-07-17 by hughsie

Herbert-Hotz, 2017-07-17:
Hi Richard,

Thank you for fixing this bug. Do you have any hint how I could install this update on my machine, or how long does it take until this fix appears in the official Fedora 26 repository?

Cheers Herbert

superm1, 2017-07-17:
@Herbert-Hotz at least until it's uploaded in the next packaged release, you can generate local RPM packages using docker. (Some information available here: https://github.com/hughsie/fwupd/tree/master/contrib)

hughsie, 2017-07-17:
fwiw, I think we should do another release in a couple of weeks, there are a few nice fixes in git now.