fwupd/fwupd #158: The dell self tests leak memory

hughsie, 2017-07-18:
Using `-Db_sanitize=address` :

    --- stdout ---
    /fwupd/plugin{dell:tpm}: OK
    /fwupd/plugin{dell:dock}: OK
    --- stderr ---

    =================================================================
    ==23855==ERROR: LeakSanitizer: detected memory leaks

    Direct leak of 1420 byte(s) in 40 object(s) allocated from:
        #0 0x7f8870b08850 in malloc (/lib64/libasan.so.4+0xde850)
        #1 0x7f886ea7a23f in __GI___vasprintf_chk (/lib64/libc.so.6+0x12323f)

    Direct leak of 1184 byte(s) in 32 object(s) allocated from:
        #0 0x7f8870b08850 in malloc (/lib64/libasan.so.4+0xde850)
        #1 0x7f8870079a38 in g_malloc (/lib64/libglib-2.0.so.0+0x4fa38)

    Direct leak of 1024 byte(s) in 1 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f886ef50c48  (/lib64/libsmbios_c.so.2+0x9c48)

    Direct leak of 48 byte(s) in 3 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x419519 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:375
        #3 0x408773 in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:307
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    Direct leak of 32 byte(s) in 2 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x419519 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:375
        #3 0x4093f1 in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:379
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    Direct leak of 32 byte(s) in 2 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x419519 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:375
        #3 0x408e04 in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:344
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    Direct leak of 16 byte(s) in 1 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x4199a0 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:391
        #3 0x4093f1 in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:379
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    Direct leak of 16 byte(s) in 1 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x4199a0 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:391
        #3 0x4099de in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:414
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    Direct leak of 16 byte(s) in 1 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x419519 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:375
        #3 0x4099de in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:414
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    Direct leak of 16 byte(s) in 1 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x4199a0 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:391
        #3 0x408e04 in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:344
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    Direct leak of 16 byte(s) in 1 object(s) allocated from:
        #0 0x7f8870b08a38 in __interceptor_calloc (/lib64/libasan.so.4+0xdea38)
        #1 0x7f8870079a90 in g_malloc0 (/lib64/libglib-2.0.so.0+0x4fa90)
        #2 0x4199a0 in fu_plugin_dell_device_added_cb ../plugins/dell/fu-plugin-dell.c:391
        #3 0x408773 in fu_plugin_dell_dock_func ../plugins/dell/fu-self-test.c:307
        #4 0x7f887009a197  (/lib64/libglib-2.0.so.0+0x70197)

    SUMMARY: AddressSanitizer: 3820 byte(s) leaked in 85 allocation(s).

superm1, 2017-07-18:
Can you clarify how exactly you were able to compile with `-Db_sanitize=address`?  I would like to reproduce to dig in more closely and see if fixes help.  I tried adding that to a meson compile flag and installing `libasan `in the same fedora 25 docker container in contrib/ but ran into some compilation problems.  Similar stuff happened in debian experimental.

```
[1/4] Generating Fwupd-1.0.gir with a custom command.                                                                                                                                                          
FAILED: libfwupd/Fwupd-1.0.gir                                                                                                                                                                                 
/usr/bin/g-ir-scanner ../libfwupd/fwupd-client.c ../libfwupd/fwupd-client.h ../libfwupd/fwupd-device.c ../libfwupd/fwupd-device.h ../libfwupd/fwupd-enums.c ../libfwupd/fwupd-enums.h ../libfwupd/fwupd-error.c ../libfwupd/fwupd-error.h ../libfwupd/fwupd-release.c ../libfwupd/fwupd-release.h ../libfwupd/fwupd-remote.c ../libfwupd/fwupd-remote.h ../libfwupd/fwupd-result.c ../libfwupd/fwupd-result.h -pthread -I/usr/include/gobject-introspection-1.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include --no-libtool --namespace=Fwupd --nsversion=1.0 --warn-all --output libfwupd/Fwupd-1.0.gir -I/build/libfwupd -I/build/build/libfwupd -I./libfwupd/.. -I../libfwupd/.. --include=Gio-2.0 --include=GObject-2.0 --include=Soup-2.4 --symbol-prefix=fwupd --identifier-prefix=Fwupd --pkg-export=fwupd --cflags-begin -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wempty-body -Wextra -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wwrite-strings -I.. -pthread -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 --cflags-end -L/build/build/libfwupd -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lsoup-2.4 -L/build/build/libfwupd --library fwupd
/usr/lib64/libasan.so.3: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
/usr/lib64/libasan.so.3: warning: the use of `tempnam' is dangerous, better use `mkstemp'
/usr/lib64/libasan.so.3: warning: the use of `tmpnam_r' is dangerous, better use `mkstemp'                                                                                                                     
==1136==ASan runtime does not come first in initial library list; you should either link runtime to your application or manually preload it with LD_PRELOAD.                                                   
Command '[u'/build/build/tmp-introspectzOEI44/Fwupd-1.0', u'--introspect-dump=/build/build/tmp-introspectzOEI44/functions.txt,/build/build/tmp-introspectzOEI44/dump.xml']' returned non-zero exit status 1    
[2/4] Generating Dfu-1.0.gir with a custom command.                                                                                                                                                            
FAILED: libdfu/Dfu-1.0.gir                                                                                                                                                                                     
/usr/bin/g-ir-scanner ../libdfu/dfu-common.c ../libdfu/dfu-common.h ../libdfu/dfu-context.c ../libdfu/dfu-context.h ../libdfu/dfu-device.c ../libdfu/dfu-device.h ../libdfu/dfu-element.c ../libdfu/dfu-element.h ../libdfu/dfu-error.c ../libdfu/dfu-error.h ../libdfu/dfu-firmware.c ../libdfu/dfu-firmware.h ../libdfu/dfu-image.c ../libdfu/dfu-image.h ../libdfu/dfu-sector.c ../libdfu/dfu-sector.h ../libdfu/dfu-target.c ../libdfu/dfu-target.h -pthread -I/usr/include/gobject-introspection-1.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include --no-libtool --namespace=Dfu --nsversion=1.0 --warn-all --output libdfu/Dfu-1.0.gir -I/build/libdfu -I/build/build/libdfu -I./libdfu/.. -I../libdfu/.. --include=AppStreamGlib-1.0 --include=Gio-2.0 --include=GObject-2.0 --include=GUsb-1.0 --symbol-prefix=dfu --identifier-prefix=Dfu --pkg-export=dfu --cflags-begin -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wempty-body -Wextra -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wunde [... cut ...]
==1142==ASan runtime does not come first in initial library list; you should either link runtime to your application or manually preload it with LD_PRELOAD.                                                   
Command '[u'/build/build/tmp-introspectUnigS5/Dfu-1.0', u'--introspect-dump=/build/build/tmp-introspectUnigS5/functions.txt,/build/build/tmp-introspectUnigS5/dump.xml']' returned non-zero exit status 1      
ninja: build stopped: subcommand failed.                                                                                                                                                                       
```

hughsie, 2017-07-18:
Yes, sorry, this is a meson bug that's being worked on. The easiest way of seeing the output is just to remove the calls to generating the gir in your local tree for libfwupd and libdfu.

superm1, 2017-07-18:
The ones in libsmbios seem to be fixed with current libsmbios master.  Between that commit I just made and libsmbios master here's what I end up with left (which seems right to me).

```
./dell-self-test
/fwupd/plugin{dell:tpm}: OK
/fwupd/plugin{dell:dock}: OK

=================================================================
==8212==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 2056 byte(s) in 2 object(s) allocated from:
    #0 0x7f4fc59c7ea0 in calloc (/lib/x86_64-linux-gnu/libasan.so.3+0xc1ea0)
    #1 0x7f4fc4f5b080 in g_malloc0 (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x50080)

Direct leak of 1428 byte(s) in 40 object(s) allocated from:
    #0 0x7f4fc59c7cf8 in malloc (/lib/x86_64-linux-gnu/libasan.so.3+0xc1cf8)
    #1 0x7f4fc38f7bff in __vasprintf_chk (/lib/x86_64-linux-gnu/libc.so.6+0xf8bff)

Direct leak of 1184 byte(s) in 32 object(s) allocated from:
    #0 0x7f4fc59c7cf8 in malloc (/lib/x86_64-linux-gnu/libasan.so.3+0xc1cf8)
    #1 0x7f4fc4f5b028 in g_malloc (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x50028)

Indirect leak of 30 byte(s) in 2 object(s) allocated from:
    #0 0x7f4fc59c7cf8 in malloc (/lib/x86_64-linux-gnu/libasan.so.3+0xc1cf8)
    #1 0x7f4fc4f5b028 in g_malloc (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x50028)

SUMMARY: AddressSanitizer: 4698 byte(s) leaked in 76 allocation(s).
```

hughsie, 2017-07-18:
Thanks for that, but I still see `valgrind --leak-check=full ./plugins/dell/dell-self-test` gives `definitely lost: 4,140 bytes` and it also explains where they were allocated.

superm1, 2017-07-18:
OK I think that should be in better shape now, let me know if you see anything else standing out.

hughsie, 2017-07-19:
Looks good, thanks. In a related note, do the `Conditional jump or move depends on uninitialised value` messages worry you or nothing to worry about?

hughsie, 2017-07-19:
(I think this comes all the way from dell_smi_factory)

superm1, 2017-07-21:
Were you seeing that with libsmbios master?  Or a packaged version, and if so which one.  I'll need to double check.

commit d817c536 (2017-07-18): Correct a memory leak in Dell plugin (Fixes #158)
closed 2017-07-24 by hughsie