fwupd/fwupd #162: push updates instead of pulling

pohly, 2017-07-26:
The current model of pushing firmware to LVFS and then having clients running fwupd pull from it (or some other server) has some drawbacks when it comes to embedded devices:
- devices might not support http(s)
- using GnuPG for validating the update meta data brings in GPLv3-licensed code, which for some
  device manufacturers is not acceptable
- devices need to poll regularly
- the normal, shared database is larger than strictly necessary for each device

Most embedded devices will need a system update mechanism that is owned and controlled by the device manufacturer. The manufacturer knows which firmware can be updated on the device. Therefore it would be useful to deliver new firmware via the system update mechanism and just call fwupdmgr when files are already available on the local filesystem.

In https://bugzilla.yoctoproject.org/show_bug.cgi?id=11468#c2 I've outline how fwupd could support this without too many intrusive changes. If you agree that supporting such a mode would be useful, we can also discuss it here and then see who will have the time to implement it.

I understand that this goes a bit against the desire to get more firmware published via LVFS. But I'm not sure whether firmware that might get distributed by embedded device manufacturers really belongs into the "general purpose" LVFS.

hughsie, 2017-07-26:
Okay, a lot to think about here. I appreciate you opening the issue to work out an upstream solution rather than just patching things locally. Lets go through things one at a time:

> devices might not support https

In this case we'll still have libsoup, but the TLS stuff will be missing, right? No code changes.

> using GnuPG for validating the update metadata

Note, we don't have to use GnuPG; I was originally working with Microsoft using PKCS but for one reason or another that didn't happen. I'd happily support both, although I'm only familiar with GnuPG.

> devices need to poll regularly

Right, or we could just support other things than just `/var/lib/fwupd/remotes.d/` -- e.g. I think `/usr/share/fwupd/remotes.d/` would make perfect sense here. We'd probably need something like `Readonly=true` in the .conf file so that it gets skipped by `fwupdmgr refresh` and the code in gnome-software. I don't think we need anything like an `Unsigned=true` as fwupd trusts that if it's on the filesystem in a directory it owns then it was the one that put it there, and it must have been signature checked before writing.

> and just call fwupdmgr when files are already available on the local filesystem

I'd much rather fwupd just set up a inotify watch on the directory rather than trying to "refresh" from a local remote, as there's really no need to copy the file from /usr to /var

pohly, 2017-07-26:
On Wed, 2017-07-26 at 12:41 +0000, Richard Hughes wrote:
> > devices might not support https
> In this case we'll still have libsoup, but the TLS stuff will be
> missing, right? No code changes.

They might not even have IP connectivity, or not without restrictions.

> Note, we don't have to use GnuPG; I was originally working with
> Microsoft using PKCS but for one reason or another that didn't
> happen. I'd happily support both, although I'm only familiar with
> GnuPG.

That would work. But we ran into the same situation with OSTree and
currently work on solving it there, so if we can use the system update
mechanism as the secure file delivery mechanism also for fwupd, then
nothing needs to be done in fwupd.

The OSTree discussion is here: https://mail.gnome.org/archives/ostree-list/2017-June/msg00002.html

> Right, or we could just support other things than just
> /var/lib/fwupd/remotes.d/ -- e.g. I think /usr/share/fwupd/remotes.d/
> would make perfect sense here. We'd probably need something like [...]

Sounds good to me.

superm1, 2017-07-26:
@pohly Could you provide some insight into what you want fwupd to do if it sets up an inotify and detects something new was dropped on the filesystem?  Should it emit a signal on the bus that you pick up with some other piece of software?  Or are you still expecting a user/process to manually call `# fwupdmgr update`?

hughsie, 2017-07-26:
I think something that's not the daemon needs to call into the daemon to trigger the update, but `fwupdmgr update` feels like a bit of a hack. Couldn't we use the existing systemd service to run this?

hughsie, 2017-07-26:
> Sounds good to me.

I've spent a few minutes on a patch, I'm just adding tests and then I'll open a PR for comments.

hughsie, 2017-07-26:
Something like https://github.com/hughsie/fwupd/pull/163 ? Not completely tested yet, but feedback on general concept welcome.

closed 2017-07-28 by hughsie