fwupd/fwupd #164: UEFI firmware update fails

DCMiner, 2017-07-27:
Hi all,

So, I know the Dell xps 9360 has seen similar problems before [https://github.com/hughsie/fwupd/issues/75](url), but this seems to be ever so slightly different, and, unfortunately, I lack the technical knowhow to account for the differences (and as well it seems like the older problem was fixed in the AUR)... I'm running Antergos (Arch) Linux with the 4.11.9-1-ARCH kernel version. I ran FWUPD, restarted, and nothing happened. When I run, now, similar to the older problem in the link, I get:

`
$ fwupdmgr get-devices
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  DeviceID:             /dev/mei
  DisplayName:          Intel AMT (unprovisioned)
  Plugin:               amt
  Flags:                internal
  Version:              11.6.29
  VersionBootloader:    11.6.29
  Created:              2017-07-27

  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  DeviceID:             UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0
  DisplayName:          XPS 13 9360
  Description:          <p>Updating the system firmware improves performance.</p>
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.3.5
  VersionLowest:        0.1.3.5
  Created:              2017-07-27

  Guid:                 924ab06f-1e93-5314-8e00-01d446fe2e45
  Guid:                 2a93248e-2955-53e1-8a93-8b21473f9c6c
  DeviceID:             usb:00:01:04
  DisplayName:          Mass Storage Device
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x058F
  Version:              1.0
  Created:              2017-07-27

  Guid:                 a2721130-d6d5-51aa-a710-0323525ca469
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Plugin:               udev
  Flags:                internal
  DeviceVendor:         Intel Corporation
  DeviceVendorId:       PCI:0x8086
  Created:              2017-07-27`

`
$ fwupdmgr update
Downloading 0.1.3.7 for XPS 13 9360...
Fetching firmware https://secure-lvfs.rhcloud.com/downloads/8fc8f1e3af81c00490a797def69ea5b8186d69b4-firmware_XPS_9360_1_3_7.cab
Downloading…           [****************************************]
Updating 0.1.3.7 on XPS 13 9360...
Decompressing…         Retrying as an offline update...
Scheduling…            UEFI firmware update failed: Unknown error -1`

So... Following the questions in the earlier problem, I have efivar 31-1, fwupd 0.9.5-3, and fwupdate 9-1. No secure boot, and, as said earlier, 4.11.9-1-ARCH kernel. With debug flag, fwupd produces the following output:

`
$ sudo /usr/lib/fwupd/fwupd -v
[sudo] password for XXX:
09:17:13:0146 Fu  Verbose debugging enabled (on console 1)
09:17:13:0147 Fu  loading config values from /etc/fwupd.conf
09:17:13:0147 Fu  using config path of /etc/fwupd
09:17:13:0147 Fu  loading from /etc/fwupd/remotes.d/lvfs-testing.conf
09:17:13:0147 Fu  loading from /etc/fwupd/remotes.d/lvfs.conf
09:17:13:0147 Fu  using config path of /usr/lib/fwupd
09:17:13:0147 Fu  ordering lvfs-testing=lvfs+1
09:17:13:0147 Fu  remote lvfs-testing not enabled, so skipping
09:17:13:0148 As  run AsStore:store-from-file{/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz}
09:17:13:0150 As  run AsStore:store-from-root
09:17:13:0150 As  using icon path /var/lib/fwupd/remotes.d/lvfs/icons
09:17:13:0151 As  run AsStore:match-addons
09:17:13:0151 As  Emitting ::changed() [from-root]
[... 352 lines cut ...]
So, this is a bit funny, given the "ubuntu" in the directory, since I'm running Antergos / Arch.... Now, I did install a firmware upgrade successfully a couple months ago while running Ubuntu - I wonder if this is leftover from that? Or is it something new, and just misplaced? How would I tell?

efivar has an entry too, though, again, for me, hard to tell what it means...

`
$ efivar -l | grep fwupdate
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0`

And, nothing shows up in the boot directory:

`
$ tree /boot/efi
/boot/efi
└── EFI
    ├── antergos_grub
    │   └── grubx64.efi
    ├── BOOT
    │   └── BOOTX64.efi
    │       └── grubx64.efi
    └── Microsoft
        └── Boot
            └── bootmgfw.efi
                └── grubx64.efi`

So, that's where I'm sitting with everything right now, just trying to figure out what's going on, how to fix it, and whether it's my fault or something in fwupd. Any help would be deeply appreciated. Thanks!

superm1, 2017-07-27:
(I updated your comment to use the triple tick markdown.  In future - please use that to improve readability).

For your old entries, that tells me that you previously had Ubuntu 16.04 or later installed and for some reason the fwupdate arch package didn't clean up on installation.  This script and oneshot systemd service should have run.
https://github.com/rhboot/fwupdate/blob/master/linux/cleanup.in
https://github.com/rhboot/fwupdate/blob/master/linux/cleanup.service.in
I would recommend checking with @ArchangeGabriel to make sure that those are getting run otherwise problems get harder to debug.

To start out, **please first switch to master** and reproduce on master.  There was a bunch of code added to fwupd master that will make debugging this much much easier.  See https://github.com/hughsie/fwupd/wiki/Common-Problems

As we concluded in https://github.com/hughsie/fwupd/issues/100, arch does not automatically install the fwupdate boot EFI application.  **You'll need to manually do this.**  That's probably the biggest problem you're hitting.  See specifically https://github.com/hughsie/fwupd/issues/100#issuecomment-314693835

DCMiner, 2017-07-28:
OK, I was able to manually run the cleanup script and get rid of the leftover entries. I'm having some trouble getting the master branch to build on my machine, but I'll mess around with that on my own for a bit and hopefully report back soon...

superm1, 2017-07-28:
Give this a try:
https://github.com/hughsie/fwupd/blob/master/contrib/README.md

DCMiner, 2017-07-28:
OK. So, I got things set up, copied the files to the directories as described in [#100](https://github.com/hughsie/fwupd/issues/100), booted from the resulting UEFI entry manually, and, everything just worked. As best as I can guess, this was just some weird interference between problems of the type described in [#100](https://github.com/hughsie/fwupd/issues/100) and [#75](https://github.com/hughsie/fwupd/issues/75), and the leftover entries that hadn't been properly cleaned up. Thank you for the pointers, and for helping to keep up this great little piece of software!

closed 2017-07-28 by superm1

ArchangeGabriel, 2017-09-03:
@superm1 Are you saying that the fwupdate package should run the cleanup script on install? Or something else?

ArchangeGabriel, 2017-09-03:
Hum, according to the spec file I would guess so… But this script suppose /boot/efi/EFI (or now a given compile time path).

superm1, 2017-09-05:
@ArchangeGabriel the cleanup script needs to be run if someone is switching distros.  It doesn't matter if it's at install time or if it's at runtime before first use, but problems will occur if it doesn't run.

I'm personally thinking you should require people to symlink that path to the right place if it's not where they use their ESP.  You can probably query where the ESP is problematically from `bootctl `, mount it, and set up the symlink yourself at post install too can't you?

ArchangeGabriel, 2017-10-18:
Forgot I had to reply here too.

Regarding detecting the ESP/mounting it, same issue as in the other discussion. Regarding symlinking, even worse since a lot of people will have /boot as their ESP, and the FS of the ESP doesn’t support symlinking, so we cannot have a pointer from /boot/efi/EFI/arch to /boot.

So the best I can do here is to tell people they should run some instructions if having switched from another distro and have a section in the wiki regarding the proper commands to run.