fwupd/fwupd #165: 8bitdo gamepad no longer working with v.0.9.4-1.fc25

ernesto1, 2017-08-01:
After upgrading the fwupdmgr to version 0.9.4-1.fc25 (fix for issue #121 ) the 8bitdo SFC30 gamedap no longer works.  I connect the gamepad using the USB cable.

journalctl -f yields the following when connecting the device:

`Jul 31 23:02:34 strike kernel: usb 4-2: new full-speed USB device number 4 using uhci_hcd
Jul 31 23:02:34 strike kernel: usb 4-2: New USB device found, idVendor=2dc8, idProduct=ab21
Jul 31 23:02:34 strike kernel: usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jul 31 23:02:34 strike kernel: usb 4-2: Product: SFC30 Joystick
Jul 31 23:02:34 strike kernel: usb 4-2: Manufacturer: SFC30             
Jul 31 23:02:34 strike kernel: input: SFC30              SFC30 Joystick as /devices/pci0000:00/0000:00:1a.1/usb4/4-2/4-2:1.0/0003:2DC8:AB21.0005/input/input22
Jul 31 23:02:34 strike kernel: hid-generic 0003:2DC8:AB21.0005: input,hidraw3: USB HID v1.10 Joystick [SFC30              SFC30 Joystick] on usb-0000:00:1a.1-2/input0
Jul 31 23:02:34 strike mtp-probe[5681]: checking bus 4, device 4: "/sys/devices/pci0000:00/0000:00:1a.1/usb4/4-2"
Jul 31 23:02:34 strike mtp-probe[5681]: bus: 4, device: 4 was not an MTP device
Jul 31 23:02:39 strike fwupd[2188]: Failed to add 8Bitdo device: failed to retrieve from device on ep 0x82: transfer timed out
`

I experienced the same behaviour on two machines where the controller was working albeit fwupdmgr would crash as noted in #121.  On both systems the controller stopped working after the upgrade.

Fedora 26 experiences the same issue.

hughsie, 2017-08-01:
Can you downgrade to the 2.71 release does that bring the controller back to life?

ernesto1, 2017-08-01:
Where can I get that package?

In fedora 26 I can only downgrade to 0.9.4 from the current 0.9.5

```
[ernesto@strike ~]$ sudo dnf --showduplicates list fwupd
Last metadata expiration check: 3:03:16 ago on Tue 01 Aug 2017 06:28:35 AM EDT.
Installed Packages
fwupd.x86_64                                                                                          0.9.5-1.fc26                                                                                           @updates
Available Packages
fwupd.i686                                                                                            0.9.4-1.fc26                                                                                           fedora  
fwupd.x86_64                                                                                          0.9.4-1.fc26                                                                                           fedora  
fwupd.i686                                                                                            0.9.5-1.fc26                                                                                           updates 
fwupd.x86_64                                                                                          0.9.5-1.fc26                                                                                           @updates
fwupd.x86_64                                                                                          0.9.5-1.fc26
```

hughsie, 2017-08-03:
No, sorry. `fwupdmgr downgrade" -- as in downgrade the firmware, not the software.

ernesto1, 2017-08-03:
It says 'No supported devices'

Even though the controller is listed under get-devices

```
[ernesto@strike Downloads]$ fwupdmgr get-devices
  Guid:                 a7fcfbaf-e9e8-59f4-920d-7691dc6c8699
  DeviceID:             usb:00:02
  DisplayName:          SFC30 Gamepad
  Plugin:               ebitdo
  Flags:                allow-online|needs-bootloader
  DeviceVendor:         8bitdo
  DeviceVendorId:       USB:0x2DC8
  Version:              2.71
  Created:              2017-08-03
```

KingDuckZ, 2017-10-04:
I'm trying fwupd 0.9.6 from gentoo, when I run fwupdmgr get-devices my 8bitdo doesn't appear in the list at all. I only see "Lenovo EasyCamera" and "USB2.0-CRW". The daemon status says "using plugins: altos, dfu, ebitdo, steelseries, udev, unifying, upower, usb", but if I run journalctl -f I don't see anything at all relating to my 8bitdo, even if I plug and unplug it. The exact model is 8bitdo zero FC30.

I can't tell if it used to work since this is the first time I try. I might very well be doing something wrong, just wondering where the problem is.

hughsie, 2017-10-05:
@KingDuckZ What does the --verbose daemon log say when you unplug and replug?

ernesto1, 2018-01-20:
Just to close the loop on this thread I started.

On fedora 27 here is how the gamepad behaves:
If you plug the gamepad with a USB cable it will not be detected by the system.  If you however reboot the machine then the joystick will be detected.  It will show up as /dev/input/js0 (assuming no other gamepads are plugged in).

If you use the gamepad under bluetooth, it operates just fine.

You could say the support is close to perfect at this point.  Closing this item.

closed 2018-01-20 by ernesto1