fwupd/fwupd #167: Firmware update of unifying receiver from logitech fails

ggoubert, 2017-08-03:
I am on Arch, I have fwupd 0.9.5-3.
The update of the firmware for the Unifying receiver from Logitech fails with:
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened

Here is the full output of "fwupdmgr -v refresh && fwupdmgr update":
(fwupdmgr:2485): Fu-DEBUG: downloading https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz to /root/.cache/fwupdmgr/lvfs-firmware.xml.gz
Fetching metadata https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
(fwupdmgr:2485): Fu-DEBUG: progress: 27%
Downloading…           [**********                             ](fwupdmgr:2485): Fu-DEBUG: progress: 55%
Downloading…           [**********************                 ](fwupdmgr:2485): Fu-DEBUG: progress: 57%
Downloading…           [**********************                 ](fwupdmgr:2485): Fu-DEBUG: progress: 85%
Downloading…           [**********************************     ](fwupdmgr:2485): Fu-DEBUG: progress: 100%
Downloading…           [****************************************]
(fwupdmgr:2485): Fu-DEBUG: downloading https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc to /root/.cache/fwupdmgr/lvfs-firmware.xml.gz.asc
Fetching signature https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc
(fwupdmgr:2485): Fu-DEBUG: progress: 100%
Downloading…           [****************************************]
(fwupdmgr:2485): Fwupd-DEBUG: Emitting ::device-changed((null))
Downloading RQR12.07_B0029 for Unifying Receiver...
Updating RQR12.07_B0029 on Unifying Receiver...
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened

I restarted fwupd and rebooted the computer. I also tried the dongle on two different usb ports. Always the same result.

bipson, 2017-08-03:
This seems to be issue #155 - a fix will come with yet unreleased version 0.9.6 apparently.

The arch package versioned [0.9.5-3](https://www.archlinux.org/packages/community/x86_64/fwupd/) was last updated 2017-07-05 09:56 UTC, thus cannot contain the fixes (and should not, according to the version number).

You can build from source in the meantime if urgent.

hughsie, 2017-08-03:
New build on https://bodhi.fedoraproject.org/updates/FEDORA-2017-148837efa0 -- karma appreciated.

closed 2017-08-03 by hughsie