fwupd/fwupd #168: Error when trying to update Logitech Unifying receiver

AsciiWolf, 2017-08-04:
I get this error when I try to update my Logitech Unifying receiver:

```
$ sudo fwupdmgr update
Downloading RQR12.07_B0029 for Unifying Receiver...
Updating RQR12.07_B0029 on Unifying Receiver...
Writing…               failed to erase fw @0x00: failed to send data: Device 046d:aaaa has not been opened
```

Then my mouse stops working and the receiver needs to be replugged to make it work again.

OS: Fedora 26 amd64

AsciiWolf, 2017-08-04:
There is also this in the journal:
```
srp 04 16:27:37 desktop.lan fwupd[2761]: unknown SubID 81
srp 04 16:27:42 desktop.lan fwupd[2761]: report 0x20 unknown length
srp 04 16:28:56 desktop.lan fwupd[2761]: report 0x02 unknown length
```

hughsie, 2017-08-04:
Do you have  0.9.6 in updates-testing? If yes, this is a dupe of https://github.com/hughsie/fwupd/issues/167

AsciiWolf, 2017-08-04:
Nope, 0.9.5. I don't have the testing repo enabled.

hughsie, 2017-08-04:
Right, please re-open if this still happens with the new libgusb and fwupd builds.

closed 2017-08-04 by hughsie