fwupd/fwupd #169: Time-out when trying to update Logitech Unifying receiver

piratenpanda, 2017-08-06:
This is latest 0.9.6 on arch linux:

fwupdmgr update --verbose
Downloading RQR24.05_B0029 for Unifying Receiver...
(fwupdmgr:2207): Fu-DEBUG: skpping download as file already exists
Updating RQR24.05_B0029 on Unifying Receiver...
(fwupdmgr:2207): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:2207): Fwupd-DEBUG: Emitting ::status-changed() [idle]

(fwupdmgr:2207): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:2207): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:2207): Fwupd-DEBUG: Emitting ::device-added((null))
(fwupdmgr:2207): Fwupd-DEBUG: Emitting ::changed()
request timed out

Afterwards I have to replug the usb stick to make the mouse work again. How can I provide more information to debug this?

hughsie, 2017-08-07:
What does `fwupd --verbose` say when you do `fwupdmgr update` in another terminal tab? Thanks.

piratenpanda, 2017-08-07:
I only have fwupdate and fwupdmgr in my path it seems although fwupd is installed. How would i run the first command?

hughsie, 2017-08-07:
It's in /usr/libexec/fwupd

piratenpanda, 2017-08-07:
on arch it seems to be /usr/lib/fwupd/fwupd

my output of fwupd --verbose can be found here: https://hastebin.com/eqanuyulas.php

correaa, 2017-08-09:
Same problem here. More details: https://ask.fedoraproject.org/en/question/109038/logitech-unifying-receiver-security-update/

hughsie, 2017-08-09:
Can you try with another computer, or with/without a USB hub? It could be the device is taking too long to enumerate, which might happen if there's a slow hub between the device and the controller. Also try disabling/enabling USB3 if this is what you are using.

piratenpanda, 2017-08-09:
I already tried with or without hub, no difference. I'll try to find another computer when I get home today

piratenpanda, 2017-08-09:
Other laptop, usb2 port, also the same time-out

zuzzurro, 2017-08-09:
Fedora 26 as of toda. Never worked for me, always fails like this:
```
me@localhost 2362> fwupdmgr --verbose update
Downloading RQR24.03_B0027 for Unifying Receiver...
(fwupdmgr:32292): Fu-DEBUG: skpping download as file already exists
Updating RQR24.03_B0027 on Unifying Receiver...
(fwupdmgr:32292): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:32292): Fwupd-DEBUG: Emitting ::status-changed() [idle]

(fwupdmgr:32292): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:32292): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:32292): Fwupd-DEBUG: Emitting ::device-added((null))
(fwupdmgr:32292): Fwupd-DEBUG: Emitting ::changed()
request timed out
```
After the failure the mouse becomes irresponsive and I need to unplug the USB key in order to make it work again.

arigit, 2017-08-13:
I'm having the same problem, with Fedora 26. 
I also had this same issue when I attempted the firmware update with Fedora 25  - I could never get the receiver firmware updated.

`fwupdmgr --verbose update
Downloading RQR24.05_B0029 for Unifying Receiver...
(fwupdmgr:4656): Fu-DEBUG: skpping download as file already exists
Updating RQR24.05_B0029 on Unifying Receiver...
(fwupdmgr:4656): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:4656): Fwupd-DEBUG: Emitting ::status-changed() [idle]

(fwupdmgr:4656): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:4656): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:4656): Fwupd-DEBUG: Emitting ::device-added((null))
(fwupdmgr:4656): Fwupd-DEBUG: Emitting ::changed()
request timed out
`

`fwupdmgr --verbose get-devices
Unifying Receiver
  DeviceID:             /sys/devices/pci0000:00/0000:00:14.0/usb1/1-11
  Guid:                 (blank)
  Guid:                 (blank)
  Description:          <p>A Unifying receiver allows you to connect multiple compatible keyboards and mice to a laptop or desktop computer with a single USB receiver. Updating the firmware on your Unifying receiver improves performance, adds new features and fixes security issues.</p>
  Plugin:               unifying
  Flags:                allow-online|supported
  DeviceVendor:         Logitech
  DeviceVendorId:       USB:0x046D
  Version:              RQR24.01_B0023
  VersionBootloader:    BOT03.01_B0008`

arigit, 2017-08-13:
I tested another, older reciever, and that one updated successfully
FWIW, the receiver that fails to upgrade is a "nano receiver", the low-profile one that comes with the small portable "MX Anywhere" mouse . I had two of these receivers, both fail in the exact same way.

piratenpanda, 2017-08-13:
Yep, Mx Anywhere here too

hughsie, 2017-08-13:
What's the U***** number laser engraved on the USB plug please?

arigit, 2017-08-13:
@hughsie  I think you are referring to this:
M/N: C-U0012
IC: 4418A-CU0012
Both pico receivers that fail to update have the same.

I contrast, the Nano receiver that got upgraded successfully:
M/N: C-U0007
IC: 4418A-CU0007

hughsie, 2017-08-13:
Right; we don't support U0012 -- only U0007 and U0008 -- it's a mistake that fwupd recognized that hardware as updatable. Can you get me the lsusb -vvv for the nano device please.

arigit, 2017-08-14:
Details below
Question - the vulnerability does not apply to this type of HW?

```
Bus 001 Device 023: ID 046d:c52b Logitech, Inc. Unifying Receiver
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc52b Unifying Receiver
  bcdDevice           24.01
  iManufacturer           1 Logitech
  iProduct                2 USB Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 RQR24.01_B0023
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      59
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     148
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      98
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

hughsie, 2017-08-14:
> Question - the vulnerability does not apply to this type of HW?

I've asked Logitech for some help here, I'll update this issue when I hear back. Thanks for the lsusb; I'm a bit confused but hopefully I'll be enlightened soon.

hughsie, 2017-08-14:
So, I heard back. The Pico devices have a different bootloader PID and need some changes in the code and in the metadata. Logitech have been nice enough to send me samples of both types and when I've tested them both I'll push new files to the LVFS.

hughsie, 2017-08-14:
See also https://github.com/Logitech/fw_updates/pull/7 which should fix the metadata-side of the problem too.

closed 2017-08-14 by hughsie

piratenpanda, 2017-09-05:
Works fine now, thanks!

correaa, 2017-09-05:
@piratenpanda , can you tell in what version this works fine? For example, I have this one (and doesn't work)

```
$ fwupdmgr --version
client version:	0.9.6
daemon version:	0.9.6
compile-time dependency versions
	appstream-glib:	0.7.0
	gusb:	0.2.11
```

piratenpanda, 2017-09-05:
On arch i run

> fwupdmgr --version
client version:	0.9.7
daemon version:	0.9.7
compile-time dependency versions
	appstream-glib:	0.6.13
	gusb:	0.2.11

sebastiaanfranken, 2018-01-08:
On Fedora 27 this is fixed, it's working as intended. `# fwupdmgr --version` gives:

> client version:	1.0.2
daemon version:	1.0.2
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11