fwupd/fwupd #171: Fedora 26: can't install latest Dell XPS 9350 BIOS update - fwupdmgr looking at wrong path?

dvelitchkov, 2017-08-13:
Hi guys,

I own an XPS 9350 (Developer Edition) that came with Ubuntu preinstalled. I had a successful run of the fwupdmgr on that Ubuntu install to update my BIOS to 1.4.17. I've since wiped the laptop and installed Fedora 26 on it. 
 
Running `get-updates` shows me a BIOS update being available:

```sh
$fwupdmgr get-updates
XPS 13 9350 System Firmware has firmware updates:
ID:                      com.dell.uefi33773727.firmware
GUID:                    33773727-8ee7-4d81-9fa0-57e8d889e1fa
Update Version:          0.1.4.18
Update Remote ID:        lvfs
Update Checksum:         SHA1(4bd36e439f854416f4716f3a5ab75166d8c408f1)
Update Location:         https://secure-lvfs.rhcloud.com/downloads/01be31ce4749ea73bcd3bf6405b117f7da4d77f6-firmware_XPS_9350_Bios_1.4.18.cab
```

But attempting to update:

```sh
$ sudo fwupdmgr update
Downloading 0.1.4.18 for XPS 13 9350 System Firmware...
Updating 0.1.4.18 on XPS 13 9350 System Firmware...
Decompressing…         
Retrying as an offline update...
Decompressing…         
Scheduling…            UEFI firmware update failed:
{error #0} libfwup.c:1094 get_fd_and_media_path(): open of /boot/efi/EFI/ubuntu/fw/fwupdate-uCVnR8.cap failed: No such file or directory 
```

The `/boot/efi/EFI/ubuntu` path does not exist on my system. I'm assuming this info is being passed down by the previous successful update perhaps? 

Here's what my dir tree looks like:

```sh
# tree /boot/efi
/boot/efi
├── 66677d48997b41adae1c8fa70f078b2e
│   ├── 4.11.10-300.fc26.x86_64
│   └── 4.11.11-300.fc26.x86_64
├── EFI
│   ├── BOOT
│   │   ├── BOOTX64.EFI
│   │   └── fallback.efi
│   ├── Dell
│   │   └── logs
│   │       └── diags_current.xml
│   └── fedora
│       ├── BOOT.CSV
│       ├── fonts
│       │   └── unicode.pf2
│       ├── fw
│       ├── fwupx64.efi
│       ├── gcdx64.efi
│       ├── grub.cfg
│       ├── grubenv
│       ├── grubx64.efi
│       ├── MokManager.efi
│       ├── shim.efi
│       └── shim-fedora.efi
├── mach_kernel
└── System
    └── Library
        └── CoreServices
            └── SystemVersion.plist

13 directories, 15 files
```

Is there a way to clear this state/cache and force fwupdmgr to not look in the wrong place? 

Some more useful info, hopefully:

```sh
$ fwupdmgr get-devices
XPS 13 9350 System Firmware
  DeviceID:             UEFI-33773727-8ee7-4d81-9fa0-57e8d889e1fa-dev0
  Guid:                 33773727-8ee7-4d81-9fa0-57e8d889e1fa
  Description:          <p>Updating the system firmware improves performance.</p>
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.4.17
  VersionLowest:        0.1.4.17
  Created:              2017-08-09

Integrated Webcam HD
  DeviceID:             usb:00:05
  Guid:                 4ec19583-c132-5c27-a9cf-061419e838dc
  Guid:                 6b31e6e6-ba7c-5251-8f4c-7f2571612251
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x0C45
  Version:              86.38
  Created:              2017-08-09

HD Graphics 520
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 a3361e1f-f523-5cfc-ad68-e20d0f792003
  Plugin:               udev
  Flags:                internal
  DeviceVendor:         Intel Corporation
  DeviceVendorId:       PCI:0x8086
  Created:              2017-08-09
```

```sh
$ fwupdmgr get-results UEFI-33773727-8ee7-4d81-9fa0-57e8d889e1fa-dev0
XPS 13 9350 System Firmware
  Guid:                 33773727-8ee7-4d81-9fa0-57e8d889e1fa
  Description:          <p>Updating the system firmware improves performance.</p>
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.4.17
  VersionLowest:        0.1.4.17
  Created:              2017-08-09
  AppstreamId:          com.dell.uefi33773727.firmware
  RemoteID:             lvfs
  Summary:              Firmware for the Dell XPS 13 9350
  UpdateVersion:        0
  FilenameCab:          firmware.bin
  UpdateHash:           SHA1(4bd36e439f854416f4716f3a5ab75166d8c408f1)
  License:              proprietary
  Size:                 11.0 MB
  UpdateUri:            https://secure-lvfs.rhcloud.com/downloads/01be31ce4749ea73bcd3bf6405b117f7da4d77f6-firmware_XPS_9350_Bios_1.4.18.cab
  UrlHomepage:          http://support.dell.com/
  Vendor:               Dell Inc.
  Trusted:              none
  UniqueID:             */*/lvfs/firmware/com.dell.uefi33773727.firmware/*
  PendingState:         success```

superm1, 2017-08-13:
Possibly a duplicate of https://github.com/rhboot/fwupdate/issues/69

Can you check if the one shot service has run? Try running it yourself and see if that cleans up for you.

dvelitchkov, 2017-08-13:
`systemctl start fwupdate-cleanup` did the trick! Thank you!

hughsie, 2017-08-14:
So, what can we do to make this not happen? Should fwupd somehow depend on fwupdate-cleanup?

superm1, 2017-08-14:
Being a oneshot service shouldn't fwupdate-cleanup have gotten called automatically at bootup without needing dependencies?  I'm a bit perplexed why it didn't.  You can tell if it's getting called successfully by the presence of `"@@DATADIR@@/fwupdate/done"`

I believe the above linked issue was specific to RHEL because of a missing kernel commit that makes sysfs attributes immutable.  It's not the exact same issue as was reported in Fedora 26.

I think what should be done is on a clean Fedora 26 install double check for the existence of that file.  If it doesn't exist dig deeper into why it's not working right.

hughsie, 2017-08-14:
`DATADIR`? That's going to be immutable in Fedora Atomic...

superm1, 2017-08-14:
Yep.. https://github.com/rhboot/fwupdate/blob/master/linux/cleanup.in
datadir is really a bad selection, it should be ending up in /var/cache instead.

Guess that needs to be sorted in fwupdate though.

hughsie, 2017-08-14:
Well, it's not a cache; it's instance data. Can you do a PR switching it to something like `/var/lib/fwupdate/cleaned-up` please? I'm rammed this afternoon and battling a headcold. Thanks!

superm1, 2017-08-14:
Sure, I've added a PR^.  @vathpela is really behind on PR's though. <poke>

closed 2017-08-14 by superm1