fwupd/fwupd #172: Bump appstream-glib dependency

gicmo, 2017-08-14:
Configures fine, but then I get build warnings, which seesm to indicate my appstream-glib version is not recent enough.

```
./../../../Code/gnome/fwupd/src/fu-hwids.c: In function ‘fu_hwids_get_guid_for_str’:
../../../../Code/gnome/fwupd/src/fu-hwids.c:88:9: warning: implicit declaration of function ‘as_utils_guid_from_data’; did you mean ‘as_utils_guid_from_string’? [-Wimplicit-function-declaration]
  return as_utils_guid_from_data (namespace_id,
         ^~~~~~~~~~~~~~~~~~~~~~~
         as_utils_guid_from_string
../../../../Code/gnome/fwupd/src/fu-hwids.c:88:9: warning: nested extern declaration of ‘as_utils_guid_from_data’ [-Wnested-externs]
../../../../Code/gnome/fwupd/src/fu-hwids.c:88:9: warning: return makes pointer from integer without a cast [-Wint-conversion]
  return as_utils_guid_from_data (namespace_id,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      (guint8*) data,
      ~~~~~~~~~~~~~~~
      items_written * 2,
      ~~~~~~~~~~~~~~~~~~
      error);
      ~~~~~~
[62/155] Compiling C object 'src/fwupd@exe/fu-hwids.c.o'.
../../../../Code/gnome/fwupd/src/fu-hwids.c: In function ‘fu_hwids_get_guid_for_str’:
../../../../Code/gnome/fwupd/src/fu-hwids.c:88:9: warning: implicit declaration of function ‘as_utils_guid_from_data’; did you mean ‘as_utils_guid_from_string’? [-Wimplicit-function-declaration]
  return as_utils_guid_from_data (namespace_id,
         ^~~~~~~~~~~~~~~~~~~~~~~
         as_utils_guid_from_string
../../../../Code/gnome/fwupd/src/fu-hwids.c:88:9: warning: nested extern declaration of ‘as_utils_guid_from_data’ [-Wnested-externs]
../../../../Code/gnome/fwupd/src/fu-hwids.c:88:9: warning: return makes pointer from integer without a cast [-Wint-conversion]
  return as_utils_guid_from_data (namespace_id,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      (guint8*) data,
      ~~~~~~~~~~~~~~~
      items_written * 2,
      ~~~~~~~~~~~~~~~~~~
      error);
      ~~~~~~
```

hughsie, 2017-08-14:
Hmm, I see `as_utils_guid_from_data, Since: 0.6.13` -- let me check that.

hughsie, 2017-08-14:
Yup, it's certainly in 0.6.13 -- are you perhaps linking against an older version?

closed 2017-08-21 by hughsie