fwupd/fwupd #176: overriding systemdunitdir and udevrulesdir

yorickvP, 2017-08-16:
Hi! I'm trying to update the `fwupd` nix package. It uses `--with-systemdunitdir` and `--with-udevrulesdir` to specify the location the systemd and udev rules should go. Is there a way to override these with the new meson thing?

superm1, 2017-08-16:
It pulls that information from pkgconfig:
`systemd.get_pkgconfig_variable('systemdsystemunitdir')` and `udev.get_pkgconfig_variable('udevdir')`

https://github.com/hughsie/fwupd/blob/master/data/meson.build#L23
https://github.com/hughsie/fwupd/blob/master/meson.build#L270

Are those not accurate for you on nix?

yorickvP, 2017-08-16:
They are accurate, but I don't exactly want to install to there when I'm making a package.

hughsie, 2017-08-16:
Can't you use DESTDIR for a package?

yorickvP, 2017-08-16:
I want to install to $out, this is the location it will be on everyone else's systems, so it's set as a `--prefix`.
The udev files, however, should go to $out/lib/udev/rules (and get symlinked into the rest of the udev rules later). Something similar for the systemd files.

It currently detects the udev rules location as `/nix/store/dgc46a75znkibbhq21vqkvacasr4cqph-systemd-234/lib/udev/`, which is not where I want to put them at all.

superm1, 2017-08-16:
Yeah take a look how arch packages are built for a simple example of how they do it with meson:
https://github.com/hughsie/fwupd/blob/master/contrib/PKGBUILD

yorickvP, 2017-08-16:
I'm currently just patching the meson.build files with [this diff](http://sprunge.us/DgcT?diff).
I got it to build and install (after disabling the docs and man pages, docbook doesn't like them at all). Now I find out that [meson strips out the rpath during install](https://github.com/mesonbuild/meson/blob/master/mesonbuild/scripts/meson_install.py#L351). First thing I can think of is patch that out too. Not a fan.

superm1, 2017-08-16:
What exactly is happening with `DESTDIR `set to `$out`?  It's still picking up `/nix/store/dgc46a75znkibbhq21vqkvacasr4cqph-systemd-234/lib/udev/`?

Can you share the systemd pkg-config?

yorickvP, 2017-08-16:
I patched meson to remove the offending rpath stripping, now the
executables run (with `prefix` set).

With destdir on $out, it installs everything in $out/usr/local (but can't
find libfwupd.so.1). So it doesn't pick up the weird udev path, but also
doesn't include the correct rpath to the library. Then, it tries to load
/usr/local/lib/fwupd-plugins-2, which should be
`/nix/store/m63zqrry493qj91xpfygz7awhn4aq07y-fwupd-0.9.6/usr/local/lib/fwupd-plugins-2`.
It seems `prefix` should be the way to go.

superm1, 2017-08-16:
What you'll want is the combination of prefix and DESTDIR both.  Look more closely at how the arch example above did it.

yorickvP, 2017-08-16:
Oh, sorry, I was still running with that patch. Setting destdir will cause it to install into `$DESTDIR/nix/store/dgc46a75znkibbhq21vqkvacasr4cqph-systemd-234/lib/udev/rules.d/` which is not what I'm aiming for.

superm1, 2017-08-16:
Something seem off on your distro that pkg-config is giving that funky path though.  Is that really what it shows up in pkg-config for systemd and udev?

yorickvP, 2017-08-16:
Yes, it does say that. My distro has no single directory where systemd
files should go, they should go into the package at pkg/lib/systemd, but
there is no global /lib/systemd directory. That's why I don't want it to
use that pkg-config.

closed 2017-08-16 by superm1

superm1, 2017-08-16:
Ah, thanks for explaining.  Hopefully that helps you^

commit 38d7f139 (2017-08-16): Allow configuring systemd and udev directories (Fixes: #176)