fwupd/fwupd #181: RAM usage is high

kontsevoy, 2017-08-20:
fwupd eats about 44MB on my system, adjusted for shared binaries. This is not a lot in our crazy times of multi-GB laptops, but for something used 0.00001% of the time it's on a high side. Much bigger daemons that are used much more frequently, but consume less RAM:

* libvirtd: 15MB
* Bittorrent Sync (Dropbox equivalent): 30MB
* compiz: 37MB
* nautilus: 15MB
*

hughsie, 2017-08-20:
What version of fwupd are you looking at? I see just over 1Mb with master:

        MB
    1.295^                                                                    #   
         |                  @                                                 #:::
         |                  @                                                :#:::
         |                  @        :::@:::::@:::::::::::::::::::@::::@::::@:#:::
         |                  @     ::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |                  @ ::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |                  @:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |                  @:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |                ::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |                ::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |               :::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |               :::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |              @:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |             @@:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |             @@:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |             @@:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |             @@:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |            :@@:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |            :@@:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
         |            :@@:::@:::@:::::::@:::::@::::::::::::::: :::@::::@::::@:#:::
       0 +----------------------------------------------------------------------->Mi
         0                                                                   204.1

hughsie, 2017-08-25:
Feel free to re-open if you have more data showing the problem. Thanks.

closed 2017-08-25 by hughsie

kontsevoy, 2017-08-28:
@hughsie Thanks for the prompt response. I have re-imaged this machine to a different distro for now, but I will keep an eye and next time will do a much better job at reporting a problem! Sorry about this.