fwupd/fwupd #186: Should be able to mirror remotes

hughsie, 2017-08-21:
In https://bugzilla.redhat.com/show_bug.cgi?id=1380834 we discussed a bit using Pulp to mirror a LVFS instance onto a local CDN. This works for the metadata, but not the firmware as the firmware is currently embedded as an absolute URI.

Things we could do:

 * Support a relative URI in the metadata, although that doesn't help much if the firmware download location isn't the same as the metadata location.

* Support a baseURI element in the XML and use relative URIs -- whilst cleanest this is going to break older versions of GNOME Software and `fwupdmgr`

* Support a per-remote RewriteURI in the remote `.conf` file, something like `RewriteURI=s/s3.amazonaws.com/lvfsbucket/cdn.redhat.com/` -- which is pretty icky

* Support a per-system RewriteURIs in the `fwupd.conf` file.

* Support a `ForceRelativeURI=true` parameter in the `.conf` file -- which forces fwupd clients to ignore the absolute URI and instead use a URI constructed from the metadata URI and the firmware basename -- this would enable the vendor (Red Hat) to just ship a `redhat.conf` remote and just disable `lvfs.conf` remote.

I'm erring on the last idea, but all ideas welcome.

superm1, 2017-08-21:
>Support a baseURI element in the XML and use relative >URIs -- whilst cleanest this is going to break older versions >of GNOME Software and fwupdmgr

You could just generate two sets of metadata for a while, point newer clients that support this at the newer metadata. At some point stop making the old metadata (prob when access numbers approach zero and all major distros patch in support for the new one in versions they care about)

Does this really solve the problem though? If you have base URI in metadata you still have to change it when mirrored by pulp which also means resigning.

hughsie, 2017-08-21:
>If you have base URI in metadata you still have to change it when mirrored by pulp which also means resigning.

Urgh, correct indeed. We'd need the baseURI to be set in the `.conf` file if it's not the same as the metadata.

superm1, 2017-08-21:
I think you want what's done for be remote specific rather than client specific. Someone could have a local mirror they put firmware on that's not in lvfs and if it's daemon wide you break them.

Ok so how about this:
Split up base URI and firmware filename into separate entries in the metadata. Keep the existing (combined) entry in there for compatibility with older versions and deprecated etc.

Then in the remote allow overriding base URI. There doesn't need to be ugly sed magic, or relying on it being a relative URI (which might not always be true and you back yourself into a potential corner).

To support relative URI recognize base URI of "." Or "/" or something similar.

closed 2017-08-23 by hughsie