fwupd/fwupd #190: thunderbolt: safe mode handling missing

gicmo, 2017-08-22:
The "old" thunderbolt plugin was able to deal with a host controller being in "safe mode", i.e. a mode where it loses its identity information and the only possible thing one can do with it is to re-flash a new firmware image. Currently the (new) thunderbolt plugin ignores devices that don't have a `unique_id` or valid `vendor_id`/`device_id` attributes. We can still detect that a device is the host controller (i.e. `fu_plugin_thunderbolt_is_host` returning `TRUE`) and then for known hardware (e.g. Dell), we could fill in the right `vendor_id`/`device_id`s.

superm1, 2017-08-22:
Technically does falcon ridge without dram fall into this same category too? If so we should continue the discussion from the pull request on how to handle this here as well.

superm1, 2017-08-23:
@ybernat is there an easy way to instrument a device into safe mode?  I don't even know what all the attributes will look like in that mode, will there be a `unique_id`?

ybernat, 2017-08-23:
FR - not exactly. With FR, even without a DROM to report the vendor/model IDs, we know the NVM version, which isn't the case for safe-mode. But I agree this is close enough and if there is a consistent way to identify the controller from the platform to handle safe-mode, we can hope for it to work for FR too.

For adding controller to safe-mode - nothing that I know about. In lab we do it by flashing an unsigned FW with an external tool but this is usually not possible for production controller.

superm1, 2017-08-23:
Could you share the sysfs structure for a device in safe mode and comparative structure in normal mode?
I'll fill in the stuff for the Dell handling based on what actually shows up in that context.

ybernat, 2017-08-23:
Yeah, forgot to mention: I'm not sure if the unique_id is available on safe-mode, will have to test it. Reading nvm_version returns an error (I don't remember off the top of my head which error it is). nvm_active isn't available.

And just a reminder, safe-mode is only for host controllers.

closed 2017-08-25 by hughsie