fwupd/fwupd #192: cannot downgrade uefi firmware of XPS 15 9560

hferreiro, 2017-08-23:
I am trying to downgrade the firmware of a XPS 15 laptop. These were my attempts:

```
$ fwupdmgr -v --allow-older install fdab6b6740a254c640ef6c109ff7692e1d1d0531-XPS9560v1.2.4.cab 
(fwupdmgr:8389): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:8389): Fwupd-DEBUG: Emitting ::status-changed() [idle]

Retrying as an offline update...
(fwupdmgr:8389): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:8389): Fwupd-DEBUG: Emitting ::status-changed() [idle]

Specified firmware is older than the minimum required version '0.1.3.4 < 0.1.2.4
```
$ fwupdmgr -v downgrade
(fwupdmgr:8612): Fu-DEBUG: ignoring 0.1.3.4 as older than 0.1.3.4
(fwupdmgr:8612): Fu-DEBUG: ignoring 0.1.3.3 as older than lowest 0.1.3.4
(fwupdmgr:8612): Fu-DEBUG: ignoring 0.1.2.4 as older than lowest 0.1.3.4
(fwupdmgr:8612): Fu-DEBUG: ignoring 0.1.1.3 as older than lowest 0.1.3.4
(fwupdmgr:8612): Fu-DEBUG: ignoring 0.1.0.3 as older than lowest 0.1.3.4
No supported releases
```

hughsie, 2017-08-23:
When you do a UEFI update the "minimum version" can be specified to prevent a downgrade -- if for instance the non volatile store is upgraded to a new format which is incompatible with the old firmware version. It's designed so you can't brick your computer by applying old firmware versions. I don't know why Dell increased the minimum version for the latest upgrade, but I assume there was a good reason. @superm1 might know more details.

hferreiro, 2017-08-23:
I'm having lots of problems with the firmware. In particular, ethernet-over-usb not working and wifi disconnections. I've read reports on reddit of people downgrading the firmware, so I assumed I could do it with fwupd, though I'd also like to know the reasons.

superm1, 2017-08-23:
For security reasons programmatic downgrades via UEFI capsule are disallowed.  You may be able to downgrade via the F12 POST menu (and placing the EXE file on a FAT32 on the ESP) and firmware will prevent downgrades in cases of NVRAM format changes here.

**Regarding your issues;**
If it's a Thunderbolt Ethernet device:
* make sure that you've enabled Thunderbolt security in the kernel (kernel 4.13+) or turned off Thunderbolt security in the BIOS (< kernel 4.13).
* Don't boot with it plugged in.  There are known issues that will be fixed in kernel 4.14.

I would encourage you to report your issues to support though so they can be appropriately prioritized and fixed in the future.

closed 2017-08-23 by superm1

hughsie, 2017-08-23:
> For security reasons programmatic downgrades via UEFI capsule are disallowed

Is this a Dell-specific thing or an upstream UEFI best practice thing? If the former, are there any details I can share? I've been asked by 3 people now about downgrading Dell BIOSes.

superm1, 2017-08-23:
I don't have anything else I can share on it. 

In general security best practice (not just UEFI) is to prevent letting people programmatically downgrade to take advantage of a vulnerability in an older version though.

ybernat, 2017-08-23:
Maybe it's relevant to note that ESRT definition includes "Lowest Supported Firmware Version" field.
https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/esrt-table-definition

hughsie, 2017-08-23:
> In general security best practice is to prevent letting people programmatically downgrade

Agreed; but was this a security update? I didn't see anything in the LVFS release notes.

superm1, 2017-08-23:
Security updates land in almost every dell maintenance bios release. They're often (intentionally) not documented in a public changelog.