fwupd/fwupd #207: synaptics: pointer aliasing issues

ybernat, 2017-08-30:
Compiling with clang, the following warnings appear for synaptics files:
'''
../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-common.c:153:66: warning: cast from 'const guint8 *' (aka 'const unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
                        rc = synapticsmst_common_write_dpcd (connection, REG_RC_DATA, (gint *)buf, cur_length);
                                                                                      ^~~~~~~~~~~
../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-common.c:280:12: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
                                                            (gint *)buf,
                                                            ^~~~~~~~~~~
../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-common.c:314:13: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
                                                             (gint *)cmd_data,
                                                             ^~~~~~~~~~~~~~~~
../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-common.c:363:11: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
                                                    (gint *)buf, length);
                                                    ^~~~~~~~~~~

../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-device.c:270:62: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
        rc = synapticsmst_common_read_dpcd (connection, REG_RC_CAP, (gint *)byte, 1);
                                                                    ^~~~~~~~~~~~
../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-device.c:273:62: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
                        synapticsmst_common_read_dpcd (connection, REG_VENDOR_ID, (gint *)byte, 3);
                                                                                  ^~~~~~~~~~~~
../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-device.c:785:65: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
        if (synapticsmst_common_aux_node_read (connection, REG_RC_CAP, (gint *)byte, 1) == DPCD_SUCCESS) {
                                                                       ^~~~~~~~~~~~
../thunderbolt_sw_2-fwupd/plugins/synapticsmst/synapticsmst-device.c:789:11: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'gint *' (aka 'int *') increases required alignment from 1 to 4 [-Wcast-align]
                                                           (gint *)byte, 3);
                                                           ^~~~~~~~~~~~
'''

hughsie, 2017-08-30:
Yup, this is going to explode on PPC and ARM I guess, although I don't think thunderbolt MST docks are going to be much common there yet. Certainly needs fixing tho.

superm1, 2017-08-31:
I have some fixes for this locally but I need to do more testing. The box with cascaded MST hubs is having (unrelated) fwupd crashes in libgusb so I need to sort that first.

hughsie, 2017-08-31:
> fwupd crashes in libgusb

That worries me a little, I *was* going to spin a new tarball tomorrow...

superm1, 2017-08-31:
The whole thing is a little perplexing to me.    Latest libgusb.  I spent a fair amount of time this morning trying to step through with a debugger.  Maybe you have some idea.

- I can't reproduce it when run under valgrind.
- If I make no source changes from master, it crashes at https://github.com/hughsie/fwupd/blob/master/plugins/dell/fu-plugin-dell.c#L911
```
test@test-Latitude-5414:~/fwupd/build$ sudo gdb /usr/local/libexec/fwupd/fwupd 
GNU gdb (Ubuntu 8.0-0ubuntu4) 8.0
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/local/libexec/fwupd/fwupd...done.
(gdb) run -v
Starting program: /usr/local/libexec/fwupd/fwupd -v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
17:33:27:0564 Fu  Verbose debugging enabled (on console 1)
17:33:27:0565 Fu  loading config values from /etc/fwupd.conf
[New Thread 0x7f96e581a700 (LWP 4417)]
17:33:27:0575 Fu  using config path of /etc/fwupd
17:33:27:0575 Fu  loading from /etc/fwupd/remotes.d/vendor.conf
17:33:27:0575 Fu  loading from /etc/fwupd/remotes.d/lvfs-testing.conf
17:33:27:0575 Fu  loading from /etc/fwupd/remotes.d/fwupd.conf
17:33:27:0576 Fu  loading from /etc/fwupd/remotes.d/lvfs.conf
17:33:27:0576 Fu  ordering lvfs-testing=lvfs+1
17:33:27:0576 Fu  ordering lvfs=fwupd+1
17:33:27:0576 Fu  ordering lvfs-testing=lvfs+1
17:33:27:0576 Fu  remote lvfs-testing not enabled, so skipping
17:33:27:0576 Fu  no /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz, so skipping
17:33:27:0576 As  run AsStore:store-from-file{/usr/local/share/fwupd/remotes.d/fwupd/metadata.xml}
17:33:27:0577 As  run AsStore:store-from-root
17:33:27:0577 As  using icon path /usr/local/share/fwupd/remotes.d/fwupd/icons
17:33:27:0577 As  run AsStore:match-addons
17:33:27:0577 As  Emitting ::changed() [from-root]
17:33:27:0577 As  Emitting ::changed() [add-app]
17:33:27:0577 As  Emitting ::changed() [add-app]
17:33:27:0577 Fu  remote vendor not enabled, so skipping
17:33:27:0577 Fu  devices now in store:
17:33:27:0577 Fu  1	UEFI-dummy-dev0	UEFI Updates
17:33:27:0577 Fu  2	com.via.VL811.firmware	VL811 Firmware
17:33:27:0577 Fu  3	com.via.VL811+.firmware	VL811+ Firmware
17:33:27:0577 Fu  4	com.via.VL812.firmware	VL812 Firmware
17:33:27:0577 Fu  5	com.via.VL812_B2.firmware	VL812 B2 Firmware
[New Thread 0x7f96e5019700 (LWP 4418)]
[New Thread 0x7f96e4818700 (LWP 4419)]
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/sys_vendor=Dell Inc.
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/chassis_type=9
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/product_family=
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/product_name=Latitude 5414
17:33:27:0581 Fu  no /sys/class/dmi/id/product_sku so ignoring
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/bios_vendor=Dell Inc.
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/bios_version=1.1.0
17:33:27:0581 Fu  no /sys/class/dmi/id/bios_major_release so ignoring
17:33:27:0581 Fu  no /sys/class/dmi/id/bios_minor_release so ignoring
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/board_vendor=Dell Inc.
17:33:27:0581 Fu  smbios property /sys/class/dmi/id/board_name=0FY1VN
17:33:27:0581 Fu  HardwareID-0 is not available, not available as 'ProductSku' unknown
17:33:27:0581 Fu  HardwareID-1 is not available, not available as 'BiosMajorRelease' unknown
17:33:27:0581 Fu  HardwareID-2 is not available, not available as 'BiosMajorRelease' unknown
17:33:27:0581 Fu  HardwareID-3 is not available, not available as 'ProductSku' unknown
17:33:27:0581 Fu  HardwareID-4 is not available, not available as 'ProductSku' unknown
[... 379 lines cut ...]
#2  0x00007fa5dcea560e in _int_memalign (av=av@entry=0x7fa5dd1dfb00 <main_arena>, alignment=alignment@entry=1024, bytes=bytes@entry=1008) at malloc.c:4459
#3  0x00007fa5dcea9a8d in _mid_memalign (address=<optimized out>, bytes=1008, alignment=1024) at malloc.c:3158
#4  0x00007fa5dcea9a8d in __posix_memalign (memptr=0x7fffffffda20, alignment=<optimized out>, size=1008) at malloc.c:5071
#5  0x00007fa5de3ea7a7 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007fa5de3eb453 in g_slice_alloc () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x00007fa5de3bc1fe in g_hash_table_new_full () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x000000010001cce8 in fu_device_init (device=0x1002beaa0) at ../src/fu-device.c:241
#9  0x00007fa5de6cb7c5 in g_type_create_instance () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#10 0x00007fa5de6ac358 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007fa5de6ade05 in g_object_new_with_properties () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007fa5de6ae881 in g_object_new () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x000000010001cd92 in fu_device_new () at ../src/fu-device.c:262
[... cut ...]
#21 0x00007fa5ddbdc588 in g_usb_context_add_device (context=context@entry=0x10027ce50, dev=<optimized out>) at gusb-context.c:352
#22 0x00007fa5ddbdc708 in g_usb_context_rescan (context=context@entry=0x10027ce50) at gusb-context.c:453
#23 0x00007fa5ddbdc7c8 in g_usb_context_enumerate (context=0x10027ce50) at gusb-context.c:530
#24 0x0000000100017b16 in fu_engine_load (self=0x10026d000, error=0x7fffffffe4b0) at ../src/fu-engine.c:2592
#25 0x000000010001aa2b in main (argc=1, argv=0x7fffffffe658) at ../src/fu-main.c:915
```

superm1, 2017-08-31:
(And it only happens on this box for me; can't reproduce it elsewhere)

ybernat, 2017-08-31:
I'd compile it with address sanitizer and try again.

ybernat, 2017-08-31:
(Give `-Db_sanitize=address` flag to meson)

superm1, 2017-08-31:
That fails compiling on some of the introspection stuff.  I'll see if I can work around it.

hughsie, 2017-08-31:
I assume that's the interesting thread? e.g. `thread apply all bt` doesn't show anything more plausible?

ybernat, 2017-08-31:
@superm1 I found this:
https://github.com/google/sanitizers/issues/679#issuecomment-292872704

This is the case in my F26 that I use for fwupd development.

superm1, 2017-08-31:
@hughsie I guess potentially interesting is that there is another thread  already running for handling `g_usb_context_event_thread_cb`.  This could be a race condition perhaps?

@ybernat The gir stuff should have been fixed by 
 https://github.com/mesonbuild/meson/commit/ccb253189ad970354d46f0a8e3bf381774c404af (which I have meson 0.42.0) but it isn't, just re-running ninja seems to get past it.  I'm still hitting `Shadow memory range interleaves with an existing memory mapping. ASan cannot proceed correctly` even with LD_PRELOADing though too.

superm1, 2017-08-31:
So it's something really wonky going on with the libsmbios error path causing problems in fwupd.  This particular system has some problems with libsmbios and the existing error handling code was doing the right thing.  For now i'm going to revert the thing that made this run more widely.  You can go ahead with a release tomorrow with this revert in place and i'll dig more into this for the future.

hughsie, 2018-01-17:
What do we think of this issue now?

superm1, 2018-01-17:
The aliasing issues are still real and need to be fixed. It requires re-plumbing and testing though, so I haven't found the time for it yet.

commit 22139175 (2018-08-11): synapticsmst: Fix alignment issues (Closes: #207)
commit d4e443e8 (2018-08-11): synapticsmst: Fix alignment issues (Closes: #207)
closed 2018-08-11 by hughsie