fwupd/fwupd #208: libdfu: pointer aliasing issues

ybernat, 2017-08-30:
Compiling with clang, the following warnings appear for libdfu:
```
../thunderbolt_sw_2-fwupd/libdfu/dfu-cipher-xtea.c:102:17: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'guint32 *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
        guint32 *tmp = (guint32 *) data;
                       ^~~~~~~~~~~~~~~~
../thunderbolt_sw_2-fwupd/libdfu/dfu-cipher-xtea.c:144:17: warning: cast from 'guint8 *' (aka 'unsigned char *') to 'guint32 *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
        guint32 *tmp = (guint32 *) data;
                       ^~~~~~~~~~~~~~~~
```

hughsie, 2017-08-30:
Do you know how to fix this without copying the data buf to a new array? I guess the proper thing to do would be to *allocate* a `guint32` array big enough to hold `data` and then to memcpy the data over, although that seems mighty inefficient.

ybernat, 2017-08-30:
1. Is it expected that the `data` argument will be aligned in a way suitable for `uint32`? If so, you can just validate it by testing the address (but I'm not sure how to [get the alignment of a type](http://en.cppreference.com/w/c/keyword/_Alignof) before C11 unless it's available as GNU extension in gcc.
1. Seems like `may_alias` attribute supported by gcc is relevant (I assume gcc generates a safe code in this case if needed, e.g. when compiling for ARM).
https://gcc.gnu.org/onlinedocs/gcc-7.2.0/gcc/Common-Type-Attributes.html#Common-Type-Attributes

closed 2017-08-30 by hughsie