fwupd/fwupd #217: gnome-software gets confused when there are multiple unifying receivers in different USB ports

agoode, 2017-09-02:
I have a few unifying receivers. I had 1 inserted and opened gnome-software and saw there was the firmware update. Great! That worked fine.

Then I inserted another one I had without removing the other. gnome-software refreshed and showed me there was another update. I clicked the button and nothing happened.

Then I unplugged all the receivers, and plugged only one in. No luck:

```
Sep 01 20:43:51 lincoln5 gnome-software-service.desktop[25519]: 00:43:51:0773 Gs  tried overwriting com.logitech.Unifying.RQR12.firmware key fwupd::DeviceID from /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14.4/3-14.4.1/3-14.4.1.1 to /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14.4/3-14.4.4
Sep 01 20:43:51 lincoln5 gnome-software-service.desktop[25519]: 00:43:51:0775 Gs  tried overwriting com.logitech.Unifying.RQR12.firmware key fwupd::DeviceID from /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14.4/3-14.4.1/3-14.4.1.1 to /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14.4/3-14.4.4
Sep 01 20:43:51 lincoln5 PackageKit[13120]: get-updates transaction /1167_abeddceb from uid 1000 finished with success after 85ms
Sep 01 20:43:58 lincoln5 gnome-software-service.desktop[25519]: 00:43:58:0358 Gs  failed to install com.logitech.Unifying.RQR12.firmware: Device /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14.4/3-14.4.1/3-14.4.1.1 was not found
```

This persisted until I moved the new device back into the USB port the old device was in. Then the update worked.

hughsie, 2017-09-04:
I guess I can see the logic for installing multiple Unifying receivers from a "updating them" point of view. In theory the code should work now with multiple devices, but it's a bit fragile when we switch the device from runtime mode to bootloader mode as we have to use some heuristics to match up the devices in their different modes.

agoode, 2017-09-04:
Outside of the updating scenario, I've often had multiple receivers in a single machine when I've had multiple devices (rather than pairing them all to a single receiver).

Given the error message (`tried overwriting com.logitech.Unifying.RQR12.firmware key fwupd::DeviceID from /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14.4/3-14.4.1/3-14.4.1.1 to /sys/devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14.4/3-14.4.4`), it looks like things might get messed up if you moved the receiver from one port to another (even with one device).

It seems like you do want to legitimately overwrite this value if the device physically moves between USB ports.

hughsie, 2018-09-12:
@agoode  -- Can you try with fwupd 1.1.2 please -- we rewrote and improved the unifying plugin and it now seems to work a lot better for more people. I've just tested updating with 4 different unifying dongles plugged in!

superm1, 2018-10-15:
I'm closing older bugs with `needsinfo` tags such as this one.  Can you please retry with a current release (1.1.3) and if the issues continue to exist, please re-open.

Even if the current release is not yet packaged for your distro, you can install it via snap like this:
```
# snap install --classic fwupd
```

closed 2018-10-15 by superm1