fwupd/fwupd #219: Arch package out of date

Noah-Huppert, 2017-09-03:
It appears that the [fwupd](https://www.archlinux.org/packages/community/x86_64/fwupd/) package is out of date.  

When installed it shows:

```
$ fwupdmgr --version
client version: 0.9.6
daemon version: 0.9.6
compile-time dependency versions
              appstream-glib: 0.6.13
              gusb:      0.2.11
```

I believe the 0.9.7 release will fix an issue I am having where my Dell XPS 13 fails to `fwupdmgr update` with the error `No such file or directory` for quite a few efivars. This is a brand new arch install and the location of my boot directory mount point has changed. Which I believe #171 might fix.

ArchangeGabriel, 2017-09-03:
This is not the place to report that Arch package are outdated. There is the OOD feature of Arch for this. ;)

So I’m already aware that an update is available, I’m looking at this currently.

Noah-Huppert, 2017-09-03:
In that case I'll close this, sorry for nagging ya.

closed 2017-09-03 by Noah-Huppert