fwupd/fwupd #220: powertop proposes fwupd is using excessive power

cmurf, 2017-09-04:
Powertop consistently claims fwupd is a top power consumer. If the laptop is at idle, it's the 2nd power consumer, the first being the backlight. If the laptop is being used, it's a top 10 power consumer, which just doesn't make sense. And strace isn't revealing why, so I'm not sure if this is a powertop bug or how to get more info.

Attaching powertop and strace.

[powertopfwupd.txt](https://github.com/hughsie/fwupd/files/1275309/powertopfwupd.txt)
[stracefwupd.txt](https://github.com/hughsie/fwupd/files/1275310/stracefwupd.txt)

hughsie, 2017-09-04:
Hey! fwupd should be doing basically nothing most of the time. Can you get the output of `sudo /usr/libexec/fwupd/fwupd --verbose` please, hopefully when it's displaying the 100% CPU issue. Thanks!

cmurf, 2017-09-04:
ps aux and top show 0.1 CPU usage, while powertop shows it's a major battery drag; which sounds contradictory.

  463 mW     21.4 µs/s       0.4        Process        [PID 13033] /usr/libexec/fwupd/fwupd --verbose

That's even higher than without --verbose, and about the same as firefox, quite a bit more than the wifi radio.

[fwupd-verbose.txt](https://github.com/hughsie/fwupd/files/1275489/fwupd-verbose.txt)

hughsie, 2017-09-05:
I'd say the algorithm they're using for power consumption is probably broken. It could be the large number of USB devices we have open (which is a valid complaint) which is triggering some heuristic. Does the power consumption of your system suddenly go up when you start fwupd? 0.2W (!) should be quite measurable if powertop is to be believed.

superm1, 2017-09-05:
As an experiment to confirm this hypothesis I think it'd be useful to move the USB plugin out of the plugin directory, restart fwupd and check power again.

hughsie, 2017-09-05:
That's not going to work; the `G_USB_CONTEXT_FLAGS_AUTO_OPEN_DEVICES` is going to auto-open everything on hotplug and coldplug. I guess I was wrong to add that flag if:

a) we really do prevent runtime PM of those devices
b) powertop makes us look bad

I'm playing with a new way to fix this.

superm1, 2017-09-05:
That particular flag shouldn't be used in fwupd 0.9.6 though (as tested by OP) should it though?

hughsie, 2017-09-06:
1ad45caeecd81ac6b646f336dc49d9ff2db253cd is included in 0.9.6.

superm1, 2017-09-06:
Oh yeah it is.  My mistake.

closed 2017-09-07 by hughsie