fwupd/fwupd #246: Dell XPS 15 9560 EFI scheduled firmware update never occurs

Ramblurr, 2017-09-12:
This issue is similar to #54 but it is for the XPS 15 95**60** but: it is 2017, I am on Fedora 26, and secure boot is enabled.

* an EFI update poped up via the gnome software manager
  * clicking install doesn't really do anything except make the entry disappear for a few seconds, then it re-appears
* rebooting doesn't seem to do anything either
* running `fwupdmgr  update` manually doesn't cause anything to happen on reboot either

```bash
$ sudo fwupdmgr refresh
Fetching metadata https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
Downloading…           [****************************************]
Fetching signature https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc
Downloading…           [****************************************]
$ sudo fwupdmgr get-updates
XPS 15 9560 System Firmware has firmware updates:
ID:                      com.dell.uefi34578c72.firmware
GUID:                    34578c72-11dc-4378-bc7f-b643866f598c
Update Version:          0.1.4.0
Update Remote ID:        lvfs
Update Checksum:         SHA1(9113a98356abde59cf9bf454ab1313acc570c1ee)
Update Location:         https://fwupd.org/downloads/5b8e9cbc79dd4f37df0f792196f246d9b7da1e0d-firmware_XPS_9560_1_4_0.wu.cab
$ sudo fwupdmgr  update
Downloading 0.1.4.0 for XPS 15 9560 System Firmware...
Updating 0.1.4.0 on XPS 15 9560 System Firmware...
Decompressing…         
Retrying as an offline update...
Decompressing…         
Scheduling…
```

### Versions
```bash
$ rpm -qa fwupdate-efi fwupd-labels fwupdate-libs fwupd  efivar-libs efibootmgr
fwupd-0.9.6-1.fc26.x86_64
efivar-libs-31-1.fc26.x86_64
fwupdate-efi-8-4.fc26.x86_64
efibootmgr-15-1.fc26.x86_64
fwupd-labels-0.9.6-1.fc26.x86_64
fwupdate-libs-8-4.fc26.x86_64

$ sudo efibootmgr -v
BootNext: 0002
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000* Windows Boot Manager	HD(2,GPT,b0e00793-d7c2-477f-af04-ef32549393ee,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Fedora	HD(2,GPT,b0e00793-d7c2-477f-af04-ef32549393ee,0xe1800,0x32000)/File(\EFI\fedora\shim.efi)
Boot0002* Linux-Firmware-Updater \fwupx64.efi	HD(2,GPT,b0e00793-d7c2-477f-af04-ef32549393ee,0xe1800,0x32000)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...
```

### Questions

> Are you using an NVMe disk in the machine?

Yes

> Do you have secure boot turned on?

Yes (note, this is different than #54)

> If so - can you confirm if there is a signed version of fwupx64.efi on your EFI system partition?

The file exists, but appears to be empty.

```$ sudo ls -l /boot/efi/EFI/fedora/fwupx64.efi
-rwx------. 1 root root 0 Jul  8 22:24 /boot/efi/EFI/fedora/fwupx64.efi
```

> Is the capsule staged in the ESP in that directory I listed above?

It seems so..
```bash 
$ sudo ls -l /boot/efi/EFI/fedora/fw/fwupdate-SIF2Rr.cap
-rwx------. 1 root root 10331978 Sep 12 09:13 /boot/efi/EFI/fedora/fw/fwupdate-SIF2Rr.cap
```

> Is the metadata created in /sys/firmware/efi/efivars? You can check by looking for fwupdate-* in there.

Yes.

```bash
$ sudo ls /sys/firmware/efi/efivars | grep -i fwup
fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
```

superm1, 2017-09-12:
Do you have the signed version of fwupx64.efi installed?  Forgive me, i'm not sure what package that's distributed in for Fedora.  It's also possible that this isn't available in signed form for Fedora, but I don't know.  I know Ubuntu distributes it in http://launchpad.net/ubuntu/+source/fwupdate-signed.

If it's not available, then I would recommend turning off secure boot and trying again.

Ramblurr, 2017-09-17:
> Do you have the signed version of fwupx64.efi installed?

How do I determine that exactly? The file `/boot/efi/EFI/fedora/fwupx64.efi` is 0 bytes, so it's definitely not signed.

superm1, 2017-09-18:
@Ramblurr you sure?  You might need to reinstall the fwupdate package that provides that binary if that file is 0 bytes.  That should be closer to 60-65k.

Ramblurr, 2017-09-19:
Hm, yea it seems something went wrong at some point. I reinstalled the pacakge with

```
sudo dnf reinstall fwupdate-efi
``

then updated with
```
sudo fwupdmgr update
```

then rebooted, and it flashed the update successfully.

I think something went wrong with the GUI update tool in Fedora, gnome-software i think.

closed 2017-09-19 by superm1