fwupd/fwupd #252: Installation fails in po/make-images.sh

rffontenelle, 2017-09-13:
Building with JHBuild and default modulesets for GNOME 3.26, installation fails with the following error output:
```
Running custom install script '/home/rffontenelle/jhbuild/checkout/fwupd/po/make-images.sh'
Traceback (most recent call last):
  File "/home/rffontenelle/jhbuild/checkout/fwupd/po/make-images", line 193, in <module>
    linguas=sys.argv[3])
  File "/home/rffontenelle/jhbuild/checkout/fwupd/po/make-images", line 59, in __init__
    self.fontmap = PangoCairo.font_map_get_default()
  File "/home/rffontenelle/jhbuild/install/lib/python3.6/site-packages/gi/module.py", line 137, in __getattr__
    self.__name__, name))
AttributeError: 'gi.repository.PangoCairo' object has no attribute 'font_map_get_default'
```

**Environment:**
* fwupd: latest master commit
* JHBuild: latest version in GNOME's Git repository
* moduleset: gnome-apps-3.26
* distro: Arch Linux 64-bit

**Steps to Reproduce**
* command `jhbuild build fwupd`
* tried also adding `-af` flags in order to re-run all configuration steps and force

**Full build and install output**:
* [build-install-output.txt](https://github.com/hughsie/fwupd/files/1300700/build-install-output.txt)

superm1, 2017-09-13:
What version of pangocairo is on your system?  As far as I can tell this should be available since pangocairo 1.10.

https://developer.gnome.org/pango/stable/api-index-1-10.html#api-index-1.10?

??

rffontenelle, 2017-09-14:
1.40.12, I suppose.

Here is the content of file `~/jhbuild/install/lib/pkgconfig/pangocairo.pc`:
``` 
prefix=/home/rffontenelle/jhbuild/install
exec_prefix=/home/rffontenelle/jhbuild/install
libdir=/home/rffontenelle/jhbuild/install/lib
includedir=/home/rffontenelle/jhbuild/install/include

Name: Pango Cairo
Description: Cairo rendering support for Pango
Version: 1.40.12
Requires: pango cairo
Requires.private: pangoft2
Libs: -L${libdir} -lpangocairo-1.0
Cflags: -I${includedir}/pango-1.0
```

Also, looks like I found `font_map_get_default`:
```
$  grep -R font_map_get_default /home/rffontenelle/jhbuild/install/include/pango-1.0/
/home/rffontenelle/jhbuild/install/include/pango-1.0/pango/pangocairo.h:PangoFontMap *pango_cairo_font_map_get_default       (void);
```

hughsie, 2017-09-18:
I'm unable to reproduce this using jhbuild, and even the Arch CI container is passing. I think this is an issue with your system somewhere, sorry to be unhelpful.

closed 2017-09-18 by hughsie