fwupd/fwupd #253: Cannot update Logitech Unifying Receiver (version of org.freedesktop.fwupd incorrect: failed predicate [0.9.2 ge 0.8.1])

Prinzhorn, 2017-09-14:
No sure if there is something wrong with my system or if that's a bug in fwupd. I hope somebody can help me install the update.

```
$ sudo fwupdmgr update
Downloading RQR12.07_B0029 for Unifying [runtime]...
Updating RQR12.07_B0029 on Unifying [runtime]...
Decompressing…         
version of org.freedesktop.fwupd incorrect: failed predicate [0.9.2 ge 0.8.1]
```

[image]

hughsie, 2017-09-14:
You need fwupd at least version 0.9.2 to apply the update.

closed 2017-09-14 by hughsie

Prinzhorn, 2017-09-14:
Thank you very much for the quick answer. That means I either need to find a ppa or compile it myself, right?

Piskvor, 2017-09-15:
@Prinzhorn: Found no PPAs, compiling is a yak-shaving chain of dependencies. What worked for me was this: get a bootable USB disk with a distro containing a current version (I went with [Xubuntu 17.10 b1](http://cdimage.ubuntu.com/xubuntu/releases/artful/beta-1/), which has 0.9.7. Note that you probably want a 64-bit version of the OS - 32-bit fwupdate segfaults if device is plugged in.

- Boot the live distro
- plug in the receiver
- `apt update`
- `apt install fwupd`
- edit /etc/fwupd/remotes.d/lvfs-testing.conf - change `Enabled=false` to `Enabled=true`
- `fwupdmgr refresh && fwupdmgr update`

superm1, 2017-09-15:
@Piskvor 32 bit fwupd is segfaulting when a device is plugged in for you?  I think we should fix that.  Would you mind to file a separate issue with the details for this?

Prinzhorn, 2017-09-29:
@Piskvor thank you very much. I just did the upgrade as you've described, even though I already updated it on AskUbuntu some time ago https://askubuntu.com/questions/955177/cannot-update-logitech-unifying-receiver-version-of-org-freedesktop-fwupd-incor