fwupd/fwupd #255: Stuck in weird state on Dell XPS 9360 ("already scheduled to be updated" but not updating)

FredDeschenes, 2017-09-14:
Hi!

I seem to have messed up fwupdmgr/fwupdate switching between the Arch linux packages and building directly from Git. I'm currently unable to install the 0.2.1.0 firmware because fwupdmgr seems to think that the update is already scheduled to be applied.

This seems similar to #246, feel free to close this/merge the issues if that's the case.

Command outputs : 
```shell
$ fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceID:             /dev/mei
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Plugin:               amt
  Flags:                internal|registered
  Version:              11.6.29
  VersionBootloader:    11.6.29
  Created:              2017-09-14

XPS 13 9360 System Firmware
  DeviceID:             UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  Description:          <p>Updating the system firmware improves performance.</p>
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.3.5
  VersionLowest:        0.1.3.5
  Created:              2017-09-14

Ethernet Adapter
  DeviceID:             usb:00:02
  Guid:                 8e78c79b-daca-5d43-8443-d5067f76eead
  Guid:                 f87cd651-df02-535b-a01e-6a0eba8e5bc9
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x045E
  Version:              48.0
  Created:              2017-09-14

Integrated Webcam HD
  DeviceID:             usb:00:05
  Guid:                 4ec19583-c132-5c27-a9cf-061419e838dc
  Guid:                 6b31e6e6-ba7c-5251-8f4c-7f2571612251
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x0C45
  Version:              86.38
  Created:              2017-09-14

Touchscreen
  DeviceID:             usb:00:04
  Guid:                 31cc46f6-9bdc-5b1e-b48b-1c791781dcdc
  Guid:                 9fd88db6-a96d-505b-bbea-92aee9d07455
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x04F3
  Version:              17.18
  Created:              2017-09-14

HD Graphics 620
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  Plugin:               udev
  Flags:                internal|registered
  DeviceVendor:         Intel Corporation
  DeviceVendorId:       PCI:0x8086
  Created:              2017-09-14
```
```shell
$ sudo fwupdmgr update     
Downloading 0.2.1.0 for XPS 13 9360 System Firmware...
Updating 0.2.1.0 on XPS 13 9360 System Firmware...
Decompressing…         [  -                                     ]
Scheduling…            [    \                                   ]Retrying as an offline update...
Decompressing…         [      \                                 ]
UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0 is already scheduled to be updated
```
```shell
$ efibootmgr -v
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002
Boot0002* Linux Boot Manager	HD(1,GPT,bcd77ca3-9d02-42b2-bce5-9c07a001c4dd,0x800,0x100000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0003* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0005* USB NIC	BBS(128,Realtek PXE B00 D14,0x0)..BO
Boot0006* M.2 PCIe SSD	BBS(HD,THNSN5512GPUK NVMe TOSHIBA 512G,0x0)..BO
Boot0007* UEFI: THNSN5512GPUK NVMe TOSHIBA 512GB, Partition 1	HD(1,GPT,ec5d7f16-bc18-46e4-bbc2-7420b4f08ab9,0x800,0xfa000)/File(EFI\boot\bootx64.efi)..BO
```
```shell
$ tree /boot
/boot
├── EFI
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Dell
│   │   └── Bios
│   │       └── Recovery
│   │           ├── BIOS_CUR.RCV
│   │           └── BIOS_PRE.rcv
│   ├── EFI
│   │   └── arch
│   │       ├── fw
│   │       └── fwupx64.efi
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   ├── arch-fallback.conf
│   │   ├── arch-lts.conf
│   │   └── arch-lts-fallback.conf
│   └── loader.conf
├── vmlinuz-linux
└── vmlinuz-linux-lts
```
`efivar -l | grep fw` and `ls /sys/firmware/efi/efivars | grep -i fwu` don't output anything.

I've tried running the cleanup script (`/usr/lib/fwupdate/cleanup`) with no success.

Other relevant info from similar threads :
- OS : Arch Linux, all up to date
- Using packages from `community` repo
- I am using an NVME disk
- SecureBoot is disabled
- The update capsule is not found anywhere in /boot (see `tree` output above)

superm1, 2017-09-14:
Thanks for the very detailed debugging information.  I think after we have your particular issue solved you should write a model page on our [wiki ](https://github.com/hughsie/fwupd/wiki )of all the stuff to provide that you found from the various issues.
The only things you didn't add that I think would be useful to debug this:

- have you restarted the fwupd daemon since switching packages?  just to rule out some inconsistency there
- Are you running git right now?  Or something older?  If you can shared `fwupdgmr --version `that would be useful.  It looks like it's at least 0.9.7 or later.

That error means this particular update is already in fwupd's database to be updated.
It comes from here: https://github.com/hughsie/fwupd/blob/master/src/fu-plugin.c#L861

I'm wondering if this was a bug introduced in https://github.com/hughsie/fwupd/commit/a785a1c2f68ec19acf465e129967f5713ae453c5 ?
Admittedly I haven't tested that update codepath myself in a while.  Essentially it comes down to https://github.com/hughsie/fwupd/commit/a785a1c2f68ec19acf465e129967f5713ae453c5#diff-f48ff14382e44f5696a38aae81585a37L946 is no longer calling the function for update.  It's stuck in a perpetual loop that it won't apply the update.  It's not clear to me if the fallback to offline codepath is actually still supposed to be run or not after that commit.

To figure this out, can you please kill fwupd and provide `/usr/lib/fwupd/fwupd -v `output before running `fwupdmgr update`. 

@vathpela was actually just pointing out gnome-software acting really funny on fedora 26 this week trying to apply an update.  I didn't check which version of fwupd he was on, but given the work on ux-capsule he's doing i'd guess he's on `wip/hughsie/uefi-bgrt` which should be based on 0.9.7 and possibly have this same problem.

hughsie, 2017-09-15:
> Retrying as an offline update

You should not see this anymore; UEFI updates don't need to be retried since a785a1c. I'll remove this fallback from the client code as it could be that something failed (efivars?) when scheduling the update and we just retried like we used to do for offline -- basically we can make it a whole lot simpler now.

To recover from the UEFI update stuck in the pending database you can do (with git master, sorry...) `fwupdmgr clear-offline` and then I'll add a commit that removes the now-obsolete fallback code so you can try again. I suspect you'll get an error when you try to update, but at least this time you'll see what it is.

FredDeschenes, 2017-09-15:
@superm1 : Yeah I always try to be as detailed as possible in my bug reports, I hate getting incomplete ones, just makes sense not to fill incomplete ones myself ;)

I can write a model in the wiki, but wouldn't you prefer an [issue template](https://github.com/blog/2111-issue-and-pull-request-templates)?

> have you restarted the fwupd daemon since switching packages? just to rule out some inconsistency there

I haven't restarted it explicitly, but I have rebooted multiple times with no success.

> Are you running git right now?

Nope, I'm running the Arch package.

```shell
$ fwupdmgr --version
client version:	0.9.7
daemon version:	0.9.7
compile-time dependency versions
	appstream-glib:	0.6.13
	gusb:	0.2.11
```

> To figure this out, can you please kill fwupd and provide /usr/lib/fwupd/fwupd -v

Output's here, the process hangs at that point and I had to ctrl-c it:
[fwupd_dash_v_output.txt](https://github.com/hughsie/fwupd/files/1306417/fwupd_dash_v_output.txt)
Running `fwupdmgr update` after this gave the same result as before.

@hughsie : I'll try this out and update you, thanks for the quick fix!

FredDeschenes, 2017-09-15:
@hughsie : `fwupdmgr clear-offline` from git master seems to have went ok (no output and no error exit code). I'll try the update again once I get your ok to pull, thanks again!

hughsie, 2017-09-15:
Sure, please try again and tell us how it goes.

FredDeschenes, 2017-09-15:
As you suspected earlier, I'm still getting an error when updating, but at least we now know what's going on : 

```shell
$ sudo fwupdmgr update     
Downloading 0.2.1.0 for System Firmware...
Updating 0.2.1.0 on System Firmware...
Decompressing…         [  -                                     ]
Scheduling…            [    \                                   ]UEFI firmware update failed:
{error #0} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory 
{error #1} lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory 
{error #2} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory 
{error #3} lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory 
{error #4} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory 
{error #5} lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory 
{error #6} libfwup.c:1207 get_fd_and_media_path(): mkostemps(/boot/efi/EFI//boot/efi/EFI/arch/fw/fw/fwupdate-3rN0qf.cap) failed: No such file or directory
```

I'm guessing the last one (`/boot/efi/EFI//boot/efi/EFI/arch/fw/fw/fwupdate-3rN0qf.cap`) is just a compiler flag I forgot to set, no idea about the efivars though.

superm1, 2017-09-15:
To me it looks like you have the wrong prefix complied into fwupdate/libfwup. The other errors are red herrings I believe.

FredDeschenes, 2017-09-15:
I know absolutely nothing about meson/ninja (or C development in general to be honest :P), I just used the same flags used in [Arch's PKGBUILD](https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/fwupd) : 

```shell
$ meson --prefix=/usr \
        --sysconfdir=/etc \
        --localstatedir=/var \
        --libexecdir=/usr/lib \
        --buildtype=release \
        ../build
```
and
```
$ ninja -v -C ../build
```

How would I set this prefix?

superm1, 2017-09-15:
It's not actually a fwupd issue it's fwupdate (which is s dependency for fwupd)

hughsie, 2017-09-15:
> It's not actually a fwupd issue it's fwupdate

Agree. Are you using some random git checkout of libfwupdate? I think Peter fixed the fw/fw issue last week.

FredDeschenes, 2017-09-15:
No I'm using the Arch package : https://www.archlinux.org/packages/community/x86_64/fwupdate/

I'll try building it myself with the correct path, although from what I can see it should already be ok (I remember having this issue some time ago and it was indeed fixed).

FredDeschenes, 2017-09-15:
It worked!

Looks like the issue is with the ArchLinux package itself, as I just compiled/installed it myself without any modifications (`makepkg -s` and `pacman -U fwupdate-whatever.pkg.tar.gz`), rebooted (could have probably just restarted the service), ran `fwupdmgr update` and got the update installed. I'm guessing @ArchangeGabriel had something set to override the default path when the package was built (the path seems overridable from a git config) and it's using this path for everyone, I'll put in a bug on the ArchLinux bugbase for this.

Thanks all for your help!

closed 2017-09-15 by superm1