fwupd/fwupd #257: fwupd fails to start when /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz is corrupt

pohly, 2017-09-15:
File system corruption after a hard power off of a virtual machine left the file corrupt:
```
 # ls -l /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz
-rw-r--r--    1 root     root         32703 Sep 15 14:42 /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz
# zcat /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz >/dev/null
zcat: no gzip/bzip2/xz magic
```

That prevents fwupd from starting:
```
14:45:33:0986 As  run AsStore:store-from-file{/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz}
Failed to load engine: Failed to load AppStream data: Failed to parse /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz file: Invalid compressed data
```

And fwupmgr is unhappy like this:
```
Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

I guess fwupd should recover from this somehow and continue to start. Perhaps just remove the broken file(s)?

superm1, 2017-09-15:
I agree it probably should probably recover.  Can you see if this is enough?  I think it should be and metadata will regenerate on the next refresh, but you can confirm.

```
diff --git a/src/fu-engine.c b/src/fu-engine.c
index e2caaed..5c09c58 100644
--- a/src/fu-engine.c
+++ b/src/fu-engine.c
@@ -2781,6 +2781,7 @@ fu_engine_cleanup_state (GError **error)
 gboolean
 fu_engine_load (FuEngine *self, GError **error)
 {
+       g_autoptr(GError) error_appstream = NULL;
        g_autoptr(GError) error_hwids = NULL;
        g_autoptr(GError) error_smbios = NULL;
 
@@ -2796,8 +2797,7 @@ fu_engine_load (FuEngine *self, GError **error)
        /* load AppStream metadata */
        as_store_add_filter (self->store, AS_APP_KIND_FIRMWARE);
        if (!fu_engine_load_metadata_store (self, error)) {
-               g_prefix_error (error, "Failed to load AppStream data: ");
-               return FALSE;
+               g_warning ("Failed to load AppStream Data: %s", error_appstream->message);
        }
 
        /* set shared USB context */
```

hughsie, 2017-09-16:
I think this needs to one layer deeper, i.e. if `a`,`b`,`c`,`d` are configured remotes, and `b` fails with an error we still want to try and load `c` and `d`. I'm also trying to make a test harness so we can test stuff like this and the AsStore merging code.

hughsie, 2017-09-16:
Fixed in 068d34307a6d779492c4b0898e76e8490f702e62

closed 2017-09-16 by hughsie