fwupd/fwupd #258: Fwupd kills my keyboard :-)

ernstp, 2017-09-15:
sep 15 19:39:45 mammut fwupd[2591]: failed to open: failed to ping G15 Gaming Keyboard: failed to receive: timeout
sep 15 19:39:45 mammut fwupd[2591]: failed to open: failed to ping G15 Gaming Keyboard: failed to send: failed to write: wrote -1 of 7
r

After this my keyboard no longer works. fwupd 0.9.7-2 in Ubuntu 17.10.

hughsie, 2017-09-15:
Does running git master help? We merged in some close-on-error fixes recently.

superm1, 2017-09-15:
The ubuntu docker container is broken due to a bug in systemd postinstall in ubuntu, but you should be able to compile packages for git easily like this on 17.10:

```
apt build-dep fwupd
git clone https://github.com/hughsie/fwupd/
cd fwupd
./contrib/ci/build_and_install_pkgs.sh
rm *fwupd-tests*
sudo dpkg -i *.deb
```

ernstp, 2017-09-18:
Running from git master did not help...

hughsie, 2017-09-18:
Can you test with master again please, and if it still fails please reopen this bug. Thanks!

closed 2017-09-18 by hughsie