fwupd/fwupd #259: fwupd segfaults at start

bigon, 2017-09-17:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdgmr --version
client version:	0.9.7
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out
```

```shell
$ fwupdmgr get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

The system is not using EFI

Please answer the following questions:

- Operating system and version: Debian unstable
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no

fwupd segfaults at start with the following trace:
```
#0  0x0000556f337e7dd2 in FU_IS_DEVICE (ptr=0x556f34bdc170) at ../src/fu-device.h:31
#1  0x0000556f337e7dd2 in fu_plugin_device_add (plugin=0x556f34bab840 [FuPlugin], device=0x556f34bdc170) at ../src/fu-plugin.c:317
#2  0x0000556f337e7eed in fu_plugin_device_add_delay_cb (user_data=0x556f34bdedf0, user_data@entry=<error reading variable: value has been optimized out>) at ../src/fu-plugin.c:374
#3  0x00007fe839841853 in g_timeout_dispatch (source=source@entry=0x556f34bde830, callback=<optimized out>, user_data=<optimized out>) at ../../../../glib/gmain.c:4633
#4  0x00007fe839840dd5 in g_main_dispatch (context=0x556f34b6c800) at ../../../../glib/gmain.c:3148
#5  0x00007fe839840dd5 in g_main_context_dispatch (context=context@entry=0x556f34b6c800) at ../../../../glib/gmain.c:3813
#6  0x00007fe8398411a0 in g_main_context_iterate (context=0x556f34b6c800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../../glib/gmain.c:3886
#7  0x00007fe8398414b2 in g_main_loop_run (loop=0x556f34b4edf0) at ../../../../glib/gmain.c:4082
#8  0x0000556f337d9dca in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
```
I'm pretty sure this was not the case with 0.8.2

If I'm unplugging the Logitech unify dongle, fwupd starts properly

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=874521

bigon, 2017-09-17:
[fwupd.txt](https://github.com/hughsie/fwupd/files/1308896/fwupd.txt)

bigon, 2017-09-17:
These are the logs after unpluging the unifying dongle starting fwupd and then plugging the dongle again

[fwupd2.txt](https://github.com/hughsie/fwupd/files/1308897/fwupd2.txt)

hughsie, 2017-09-17:
So, we're loading the daemon, waiting for the 200ms callback to register the daemon, when we then find out the udev plugin has a pending device-add, then the udev delayed device happens and we explode.

hughsie, 2017-09-17:
Does valgrind on the daemon in the crashing set up show anything crazy? Thanks.

bigon, 2017-09-18:
Seems to complains about `Invalid read of size 8`:

[valgrind output](https://github.com/hughsie/fwupd/files/1309858/vg_fwupd.txt)

hughsie, 2017-09-18:
Fixed in 0dec274e56c23c4afdda75bb42d5b6c9928cb1b6 -- thanks for all the logs!

bigon, 2017-10-08:
Are you sure this is fixed?

I rebuilt 0.9.7 with that patch and I still get:
```
#0  0x000055ea6d0e4c7f in FU_IS_DEVICE (ptr=0x55ea6d48e170) at ../src/fu-device.h:31
#1  0x000055ea6d0e4c7f in fu_plugin_device_add (plugin=0x55ea6d463840 [FuPlugin], device=0x55ea6d48e170) at ../src/fu-plugin.c:317
#2  0x000055ea6d0e4db0 in fu_plugin_device_add_delay_cb (user_data=0x55ea6d490990, user_data@entry=<error reading variable: value has been optimized out>) at ../src/fu-plugin.c:373
#3  0x00007fc16d082853 in g_timeout_dispatch (source=source@entry=0x55ea6d492160, callback=<optimized out>, user_data=<optimized out>) at ../../../../glib/gmain.c:4633
#4  0x00007fc16d081dd5 in g_main_dispatch (context=0x55ea6d41e800) at ../../../../glib/gmain.c:3148
#5  0x00007fc16d081dd5 in g_main_context_dispatch (context=context@entry=0x55ea6d41e800) at ../../../../glib/gmain.c:3813
#6  0x00007fc16d0821a0 in g_main_context_iterate (context=0x55ea6d41e800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../../glib/gmain.c:3886
#7  0x00007fc16d0824b2 in g_main_loop_run (loop=0x55ea6d400df0) at ../../../../glib/gmain.c:4082
#8  0x000055ea6d0d68b2 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
```

superm1, 2017-10-09:
@bigon Well that's too bad to hear.
Since there was a bunch of other large changes (like FuDeviceLocker stuff), would you mind to try with the recently 1.0.0 release?  Keep in mind it's ABI incompatible, so please use the helper scripts for generating packages to use to test.

superm1, 2017-10-18:
I've put 1.0.0 into unstable.  Please test with that and let us know if that's fixed for you.

superm1, 2017-10-18:
Closing per https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=874521#63

closed 2017-10-18 by superm1