fwupd/fwupd #265: Suggestion: select DFU protocol for Logitech products based on meta-data

mviredaz, 2017-09-20:
Currently, "fwupd" is able to recognize Logitech Unifying receivers and devices (mice/keyboards) based on hard-coded values (VIDs, PIDs, etc.).  Then, the "DFU protocol" (that is, the set of commands to be used to perform an update) is also hard-coded based on the previous step.  Ultimately, there are 5 possible protocols.

Logitech has many other non-Unifying products that use one of the same 5 DFU protocols.  These products include most Gaming mice and keyboards, as well as the presenters.  Currently, a code change would be required to update such a product (i.e., add the information required to recognize it).

I suggest to add the DFU protocol in the meta-data, so that the information required to update a Logitech device is fully contained in the meta-data.  This would allow Logitech to deploy DFUs for new products without code changes.

Note, it may be required to add some other information in the meta-data.  For example, the PID is only indirectly available (via a hash).

hughsie, 2017-10-10:
I've been thinking about this for some time. The problem with having this metadata in the update XML is that we need to know about the device ahead-of-time, rather than just when we're about to install firmware on it. We need to know if the device is updatable before we have the LVFS or local .cab file so we can add it to the daemon as a valid device.

Flipping this around, is there any way to query the hardware to find out what update protocol it is going to use?

mviredaz, 2017-10-13:
This is possible with new devices that implement HID++ 2.0.  In the case, the presence of Feature 0x00C1 or 0x00C2 advertises the use of the 0x00D0 feature in boot-loader to perform DFU.

Unfortunately, there is no similar mechanism in old devices that implement HID++ 1.0.  In this case, the best that you can do, is to query the version of the boot-loader.  If you get an error, this means that there is no boot-loader and, therefore, the device is not upgradable.  If you get a valid response, you know that the device is upgradable, since it has a boot-loader.  However, you do not know which DFU protocol it uses.

hughsie, 2018-09-12:
@mviredaz I think we have a way to do this now, with the quirk files we use in fwupd. Recentl we rewrote and improved the unifying plugin and it now it's about half the size and a much easier thing to hack on. Can you give me some more ideas on what the version check can be used for? The unedifying receivers need support in code for the various quirks, and I'm guessing all the peripheral stuff you're likely to update is >= HID++2.0, right?

mviredaz, 2018-09-19:
As stated above, you can detect the DFU protocol of HID++ 2.0 devices (currently, only the peripherals).  That is, if Feature 0x00C1 or 0x00C2 is exposed, then the DFU protocol is Feature 0x00D0).  However, we have no such mechanism for HID++ 1.0.  This is mostly relevant for receivers.  That is why, my proposal was to add this information in the meta-data.  For instance, within the "<requires>" list.

However, last October, you mentioned a chicken-and-egg problem (see above).  Do we still have the constrain that we need to know if a device/receiver is upgradable before getting access to the meta-data ?

FFY00, 2019-09-23:
/unstale

hughsie, 2020-08-18:
I don't think anyone is planning to work on this. Setting the protocol from the features implemented should be easy to add.

closed 2020-08-18 by hughsie