fwupd/fwupd #268: Busylooping around ioctl(..., USBDEVFS_REAPURBNDELAY)

wjt, 2017-09-22:
Once or twice a day, a `fwupd` thread busyloops here:

```
[pid  2859] ioctl(42, USBDEVFS_REAPURBNDELAY, 0x7f4823ffeb70) = -1 ENODEV (No such device)
[pid  2859] poll([{fd=9, events=POLLIN}, {fd=11, events=POLLIN}, {fd=22, events=POLLOUT}, {fd=24, events=POLLOUT}, {fd=29, events=POLLOUT}, {fd=35, events=POLLOUT}, {fd=36, events=POLLOUT}, {fd=42, events=POLLOUT}, {fd=43, events=POLLOUT}, {fd=45, events=POLLOUT}, {fd=50, events=POLLOUT}, {fd=54, events=POLLOUT}, {fd=56, events=POLLOUT}], 13, 60000) = 1 ([{fd=42, revents=POLLHUP}])
```

```
Thread 4 (Thread 0x7f4823fff700 (LWP 2859)):
#0  0x00007f4835f1d5b7 in ioctl () from target:/lib64/libc.so.6
#1  0x00007f4833cab3c3 in reap_for_handle () from target:/lib64/libusb-1.0.so.0
#2  0x00007f4833cad390 in op_handle_events () from target:/lib64/libusb-1.0.so.0
#3  0x00007f4833ca7279 in handle_events () from target:/lib64/libusb-1.0.so.0
#4  0x00007f4833ca827a in libusb_handle_events_timeout_completed () from target:/lib64/libusb-1.0.so.0
#5  0x00007f4833ca836f in libusb_handle_events () from target:/lib64/libusb-1.0.so.0
#6  0x00007f4836f74b91 in g_usb_context_event_thread_cb () from target:/lib64/libgusb.so.2
#7  0x00007f483778e536 in g_thread_proxy () from target:/lib64/libglib-2.0.so.0
#8  0x00007f48361ef36d in start_thread () from target:/lib64/libpthread.so.0
#9  0x00007f4835f27b8f in clone () from target:/lib64/libc.so.6
```

There is nothing interesting in the journal:

```
% journalctl -b0 -u fwupd
-- Logs begin at Sat 2016-10-01 18:30:40 BST, end at Fri 2017-09-22 11:18:04 BST. --
Sep 21 15:38:42 tensionsheet systemd[1]: Starting Firmware update daemon...
Sep 21 15:38:42 tensionsheet fwupd[2834]: disabling plugin because: failed to startup amt: ME refused connection
Sep 21 15:38:42 tensionsheet fwupd[2834]: disabling plugin because: failed to startup dell: Firmware updating not supported
Sep 21 15:38:42 tensionsheet fwupd[2834]: disabling plugin because: failed to startup test: Test plugin is only used for continuous integration
Sep 21 15:38:45 tensionsheet fwupd[2834]: disabling plugin because: failed to coldplug raspberrypi: Raspberry PI firmware updating not supported, no /boot/start.elf
Sep 21 15:38:45 tensionsheet fwupd[2834]: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported by OEM
Sep 21 15:38:45 tensionsheet fwupd[2834]: using plugins: altos, colorhug, dfu, ebitdo, steelseries, udev, uefi, unifying, upower, usb
Sep 21 15:38:45 tensionsheet fwupd[2834]: Daemon ready for requests
Sep 21 15:38:45 tensionsheet systemd[1]: Started Firmware update daemon.
```

Is there anything useful I can determine from `fwupd` while it's in this state (currently interrupted in `gdb`)? Or some extra logging that would be useful next time this recurs?

----

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdgmr --version
client version:	0.9.6
daemon version:	(null)
compile-time dependency versions
	appstream-glib:	0.7.0
	gusb:	0.2.11
```

```shell
$ fwupdmgr get-devices
80MK System Firmware
  DeviceID:             UEFI-0f6b8991-929f-47e1-8541-36dd7cf7cf5b-dev0
  Guid:                 0f6b8991-929f-47e1-8541-36dd7cf7cf5b
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac
  Version:              1.55.0
  VersionLowest:        0.18.0
  Created:              2017-09-21

Lenovo EasyCamera
  DeviceID:             usb:00:06
  Guid:                 48a505fb-1a79-5141-8a2b-b656a9db7432
  Guid:                 ddf800ff-c10c-538b-9873-f8b4825b1ad6
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x13D3
  Version:              5.7
  Created:              2017-09-21

HD Graphics 520
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 a3361e1f-f523-5cfc-ad68-e20d0f792003
  Plugin:               udev
  Flags:                internal
  DeviceVendor:         Intel Corporation
  DeviceVendorId:       PCI:0x8086
  Created:              2017-09-21

Unifying Receiver
  DeviceID:             /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4.3
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Plugin:               unifying
  Flags:                allow-online|supported
  DeviceVendor:         Logitech
  DeviceVendorId:       USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Created:              2017-09-21

USB Receiver
  DeviceID:             usb:00:02:04:03
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 39a2d8d5-9394-5e6a-b7fc-6fcebfe777aa
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x046D
  Version:              18.7
  Created:              2017-09-21

SuperMulti RW   
  DeviceID:             usb:00:02:04:01
  Guid:                 4bfba958-5fa6-5627-aba3-78967394eda0
  Guid:                 55c1f705-48e8-5976-af25-964974884d0d
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x152E
  Version:              1.98
  Created:              2017-09-21

USB Keyboard
  DeviceID:             usb:00:02:04:04
  Guid:                 f612a048-5185-504c-871f-6d10f27c382e
  Guid:                 18e34304-5a5c-5462-bd5f-13bf61230802
  Plugin:               usb
  Flags:                none
  DeviceVendorId:       USB:0x04D9
  Version:              1.1
  Created:              2017-09-21
```

Please answer the following questions:

- Operating system and version: Fedora 26
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? yes; rebooting or restarting `fwupd` makes this issue go away for a while, but it will return!
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no

wjt, 2017-09-22:
Aha! I unplugged all devices one by one – busylooping continued. Then I reattached the hub, followed by the Logitech Unifying Receiver – the busyloop stopped.

wjt, 2017-09-22:
I'm running the daemon now with `G_MESSAGES_DEBUG=all UNIFYING_HW_DEBUG=1` and will report back when this problem recurs.

wjt, 2017-10-02:
Nothing particularly interesting in `journalctl -u fwupd` even with those flags. The problem recurred after a suspend/resume cycle.
[fwupd.txt](https://github.com/hughsie/fwupd/files/1348777/fwupd.txt)

This time around, the mouse would freeze for a second or so every 5-10 seconds, then start responding again. This stopped when I killed fwupd though correlation does not imply causality!

hughsie, 2017-11-08:
Could you try this with fwupd >= 1.0.0 please? We fixed a ton of usb-open/close bugs with FuDeviceLocker.

wjt, 2017-11-22:
Thanks, I've just updated to F27 and  fwupd 1.0.1. I'll reopen if the issue recurs.

closed 2017-11-22 by wjt