fwupd/fwupd #273: We should have a i686 CI target

hughsie, 2017-10-09:
I don't mind if this is Debian or Fedora, although the latter (along with CentOS) seems to have deprecated i686 builds. We had a crasher bug in the past on i386 so I guess it's worth testing.

superm1, 2017-10-09:
Do you think it needs to be a full test suite run?  Or Just a compile test?

superm1, 2017-10-09:
And furthermore full "installed" test suite, or just local test suite?

hughsie, 2017-10-09:
I've prefer the whole installed shebang, the bug last time was a runtime crash, not a compile failure.

superm1, 2017-10-09:
OK.  So then I guess the question is repurpose one of the existing tests to do it's run as 32 bit, or introduce another  test.  Another test will increase the time of every CI run by 5-10 minutes, repurposing an existing one shouldn't change anything.

Ubuntu has also deprecated 32 bit too, so it's probably better to put this in one of the Debian builds.

hughsie, 2017-10-09:
I think another target is fine; we could argue that Debian could just be converted to 32 bit builder, but debian x64 is kinda an important target too. I'm not normally waiting on builds unless it's release time but either would be fine. We could even disable the arch one by default, although I'm not sure that helps much.

closed 2017-10-12 by hughsie