fwupd/fwupd #275: fwupd causing issues with devices sharing chip

TheOnlyJoey, 2017-10-11:
In this case devices with 0483:5750 STMicroelectronics chips will be claimed by fwupd and get stuck in a error state.
fwupd should check the usb descriptor to see if it is actually the right device, since chips can be re-used for multiple devices (this one being use in the Deepoon E2, NOLO CV1 and to name 2) while being claimed for a 8bitdo controller.

More information in the original bugreport: https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1717009

hughsie, 2017-10-12:
Do you know what part of the USB descriptor can be compared? Would you like to have a go at a patch to fix this? I'm happy to help as much as required.

TheOnlyJoey, 2017-10-12:
Well the manufacturer and product string would be sufficient, we do this exact check in OpenHMD as well. 
This example is using a check for exact this same chip: https://github.com/OpenHMD/OpenHMD/blob/master/src/drv_deepoon/deepoon.c#L294 (using hid descriptors but USB has this information as well)

closed 2017-10-12 by hughsie