fwupd/fwupd #278: alpha and hppa 1.0.0 fail to build from source

superm1, 2017-10-17:
neither of these are super common architectures, but 0.9.7 did build successfully, this just regressed in upping to 1.0.0.

Alpha (1.0.0: https://buildd.debian.org/status/fetch.php?pkg=fwupd&arch=alpha&ver=1.0.0-1&stamp=1508244127&raw=0):

```
[10/171] cc  -Ilibfwupd/fwupd@sha -Ilibfwupd -I../libfwupd -Ilibfwupd/.. -I../libfwupd/.. -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/alpha-linux-gnu/glib-2.0/include -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings -Werror -Wno-error=cpp -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -specs=/usr/share/dpkg/pie-compile.specs -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -pthread '-DG_LOG_DOMAIN="Fwupd"' '-DLOCALSTATEDIR="/var"' -MMD -MQ 'libfwupd/fwupd@sha/fwupd-device.c.o' -MF 'libfwupd/fwupd@sha/fwupd-device.c.o.d' -o 'libfwupd/fwupd@sha/fwupd-device.c.o' -c ../libfwupd/fwupd-device.c
FAILED: libfwupd/fwupd@sha/fwupd-device.c.o 
cc  -Ilibfwupd/fwupd@sha -Ilibfwupd -I../libfwupd -Ilibfwupd/.. -I../libfwupd/.. -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/alpha-linux-gnu/glib-2.0/include -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings -Werror -Wno-error=cpp -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -specs=/usr/share/dpkg/pie-compile.specs -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -pthread '-DG_LOG_DOMAIN="Fwupd"' '-DLOCALSTATEDIR="/var"' -MMD -MQ 'libfwupd/fwupd@sha/fwupd-device.c.o' -MF 'libfwupd/fwupd@sha/fwupd-device.c.o.d' -o 'libfwupd/fwupd@sha/fwupd-device.c.o' -c ../libfwupd/fwupd-device.c
cc1: error: -fstack-protector not supported for this target [-Werror]
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
dh_auto_build: cd obj-alpha-linux-gnu && ninja -j1 -v returned exit code 1
debian/rules:17: recipe for target 'build-arch' failed
make: *** [build-arch] Error 1
dpkg-buildpackage: error: debian/rules build-arch gave error exit status 2
```

hppa (1.0.0: https://buildd.debian.org/status/fetch.php?pkg=fwupd&arch=hppa&ver=1.0.0-1&stamp=1508247593&raw=0):
```
[[10/171] cc  -Isrc/fwupdprivate@sta -Isrc -I../src -Isrc/.. -I../src/.. -Isrc/../libfwupd -I../src/../libfwupd -I/usr/include/libappstream-glib -I/usr/include/uuid -I/usr/include/libgcab-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/hppa-linux-gnu/glib-2.0/include -I/usr/include/gio-unix-2.0/ -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/polkit-1 -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -fdiagnostics-color=always -pipe -std=c99 -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings -Werror -Wno-error=cpp -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -Wformat -Werror=format-security -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -pthread '-DG_LOG_DOMAIN="Fu"' '-DLOCALSTATEDIR="/var"' '-DSYSFSFIRMWAREDIR="/sys/firmware"' '-DFU_OFFLINE_DESTDIR=""' -MMD -MQ 'src/fwupdprivate@sta/fu-device.c.o' -MF 'src/fwupdprivate@sta/fu-device.c.o.d' -o 'src/fwupdprivate@sta/fu-device.c.o' -c ../src/fu-device.c
FAILED: src/fwupdprivate@sta/fu-device.c.o 
cc  -Isrc/fwupdprivate@sta -Isrc -I../src -Isrc/.. -I../src/.. -Isrc/../libfwupd -I../src/../libfwupd -I/usr/include/libappstream-glib -I/usr/include/uuid -I/usr/include/libgcab-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/hppa-linux-gnu/glib-2.0/include -I/usr/include/gio-unix-2.0/ -I/usr/include/gudev-1.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/polkit-1 -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -fdiagnostics-color=always -pipe -std=c99 -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings -Werror -Wno-error=cpp -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -Wformat -Werror=format-security -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -pthread '-DG_LOG_DOMAIN="Fu"' '-DLOCALSTATEDIR="/var"' '-DSYSFSFIRMWAREDIR="/sys/firmware"' '-DFU_OFFLINE_DESTDIR=""' -MMD -MQ 'src/fwupdprivate@sta/fu-device.c.o' -MF 'src/fwupdprivate@sta/fu-device.c.o.d' -o 'src/fwupdprivate@sta/fu-device.c.o' -c ../src/fu-device.c
cc1: error: -fstack-protector not supported for this target [-Werror]
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
dh_auto_build: cd obj-hppa-linux-gnu && ninja -j1 -v returned exit code 1
debian/rules:17: recipe for target 'build-arch' failed
make: *** [build-arch] Error 1
dpkg-buildpackage: error: debian/rules build-arch gave error exit status 2
```

[... cut ...] rror=format-security -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -pthread '-DG_LOG_DOMAIN="Dfu"' -MMD -MQ 'libdfu/dfu@sha/dfu-element.c.o' -MF 'libdfu/dfu@sha/dfu-element.c.o.d' -o 'libdfu/dfu@sha/dfu-element.c.o' -c ../libdfu/dfu-element.c
cc1: warning: -fstack-protector not supported for this target
```

So to me it looks like https://github.com/hughsie/fwupd/blob/master/meson_options.txt#L20 is the reason.  That is the default behavior (not just for CI...)

hughsie, 2017-10-18:
So we should just change the meson.options.txt to false? If so, lgtm.

superm1, 2017-10-18:
closed by 
bce4ab6

closed 2017-10-18 by superm1