fwupd/fwupd #279: Failure to build 1.0.0 (undefined reference in thunderbolt plugin)

ArchangeGabriel, 2017-10-18:
Trying to compile the package for ArchLinux leads to this error:

```
FAILED: plugins/thunderbolt/tbtfwucli 
cc  -o plugins/thunderbolt/tbtfwucli 'plugins/thunderbolt/tbtfwucli@exe/fu-thunderbolt-tool.c.o' -flto -Wl,--no-undefined -Wl,--as-needed -Wl,-O1 -Wl,-z,relro -Wl,-z,now -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -Wl,--start-group plugins/thunderbolt/libfu_plugin_thunderbolt.so libfwupd/libfwupd.so.2.0.0 src/libfwupdprivate.a -lappstream-glib -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lglib-2.0 -lgusb -lgobject-2.0 -lglib-2.0 -lusb-1.0 -lsoup-2.4 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -larchive -Wl,--end-group -lgudev-1.0 -lgobject-2.0 -lglib-2.0 -lappstream-glib -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lgudev-1.0 -lgobject-2.0 -lglib-2.0 -lgusb -lgobject-2.0 -lglib-2.0 -lusb-1.0 -lpolkit-gobject-1 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lsoup-2.4 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lsqlite3 -larchive '-Wl,-rpath,$ORIGIN/:$ORIGIN/../../libfwupd' -Wl,-rpath-link,/build/fwupd/src/build/plugins/thunderbolt:/build/fwupd/src/build/libfwupd  
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_set_percentage'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_set_metadata_boolean'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_alloc_data'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_device_remove'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_get_type'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_set_metadata'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_cache_lookup'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_add_guid'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_cache_add'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_get_metadata_boolean'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_new'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_set_name'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_cache_remove'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_set_status'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_get_type'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_get_data'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_device_get_metadata'
plugins/thunderbolt/libfu_plugin_thunderbolt.so: undefined reference to `fu_plugin_device_add'
collect2: error: ld returned 1 exit status
```

Full log at https://gist.github.com/ArchangeGabriel/694da7668285b782f407700d0b0d8682

P.S.: I’m looking at the fwupdate path issue right now.

superm1, 2017-10-18:
That's weird, we do CI with arch and that problem hasn't happened...

ArchangeGabriel, 2017-10-18:
The CI use `build-type=plain` while I set `build-type=release`, and more generally I use `arch-meson` (recently added helper, code here: https://git.archlinux.org/svntogit/packages.git/tree/trunk/arch-meson?h=packages/meson) instead of what is in the CI PKGBUILD.

I’ll try without it.

superm1, 2017-10-18:
OK so after this is sorted probably should change the arch CI in fwupd to use the same helper as you use for releasing.
I'd guess it's the `-D b_lto=true` is more likely candidate.
Do you have gcc-ar or llvm-ar available?  Seems like maybe would want those around when using that option (https://github.com/mesonbuild/meson/commit/bf2af8b7f1e411cf9be5b7a0c1e432debf70e071)

ArchangeGabriel, 2017-10-18:
Yes, they are (already) available at build time.

superm1, 2017-10-18:
Can you override arch-meson by adding a `-D b_lto=false `to see if that helps?

ArchangeGabriel, 2017-10-18:
It works that way. Does that help you determining the origin of this issue?

superm1, 2017-10-18:
I think you'll need to check with the Arch meson maintainer why they added `-D b_lto=true `in the first place.  that's link time optimization, and I suppose it fails with fwupd for whatever reason.  It's not the default behavior in meson to enable it.  It's actually default to disabled for all compilers in meson upstream.
https://github.com/mesonbuild/meson/blob/master/mesonbuild/compilers/compilers.py#L211

Until the Arch meson maintainer changes that default back to the upstream one I would recommend for now that in your PKGBUILD you just keep an override to `-D b_lto `to false.  It's a one line change.
Also, make sure that you submit a PR to update CI to use `arch-meson` so we're watching for stuff like this in the future.

hughsie, 2017-10-18:
Not me. Someone from arch needs to fix this I'm afraid.

superm1, 2017-10-18:
@ArchangeGabriel patches welcome, but until then please use my recommendation and override the behavior set by the arch maintainer.

ArchangeGabriel, 2017-10-18:
That’s what I’ve done: https://www.archlinux.org/packages/community/x86_64/fwupd/ ;)

I’ve asked other people from Arch their opinion about this specific error.

hughsie, 2017-10-23:
I'm still not sure how LTO could possibly work with `dlopen()`ed modules...

hughsie, 2017-10-28:
Okay, closing. Please re-open if you know of a way to fix this with a project that uses dlopen.

closed 2017-10-28 by hughsie