fwupd/fwupd #280: TBT self tests are failing on Ubuntu clang CI

superm1, 2017-10-18:
```
8/10 thunderbolt-self-test                   FAIL     0.18 s
 9/10 dell-self-test                          OK       0.02 s
10/10 synapticsmst-self-test                  OK       0.01 s
OK:         9
FAIL:       1
SKIP:       0
TIMEOUT:    0
The output from the failed tests:
 8/10 thunderbolt-self-test                   FAIL     0.18 s
--- command ---
LD_PRELOAD='libumockdev-preload.so.0' /build/build/obj-x86_64-linux-gnu/plugins/thunderbolt/thunderbolt-self-test
--- stderr ---
(/build/build/obj-x86_64-linux-gnu/plugins/thunderbolt/thunderbolt-self-test:1065): FuPluginThunderbolt-WARNING **: Need to run within umockdev wrapper (umockdev-wrapper thunderbolt-self-test)!
-------
Full log written to /build/build/obj-x86_64-linux-gnu/meson-logs/testlog.txt
FAILED: meson-test 
/usr/bin/python3 -u /usr/share/meson/meson test --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
dh_auto_test: ninja test returned exit code 1
debian/rules:48: recipe for target 'override_dh_auto_test' failed
make[1]: *** [override_dh_auto_test] Error 1
make[1]: Leaving directory '/build/build'
debian/rules:17: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
debuild: fatal error at line 1124:
dpkg-buildpackage -rfakeroot -us -uc failed
```

Ubuntu recently (10/15) moved to to umockdev 0.9.4-1 and now thunderbolt self tests fail.  It's a bit odd because Debian has the same build.

superm1, 2017-10-18:
Upstream did change this in 0.9.4, so I guess this is a regression:
https://github.com/martinpitt/umockdev/commit/49a170190e2b2d56c0615aa42b910b790e135741

superm1, 2017-10-18:
Oh so debian testing hasn't switched to umockdev 0.9.4 yet, only debian unstable has.  That explains things..

gicmo, 2017-10-19:
> Upstream did change this in 0.9.4, [...]
Yeah, looking at the change, instead of checking for the presence of `libumockdev-preload` in `LD_PRELOAD` it is now checking if `/sys` is a "real" filesystem (which indicates it is running inside the mock env).
The problem is that we are checking if we run inside umockdev *before* creating a `UMockdevTestbed` (because we create a new one inside the setup method for each individual test). 
An easy solution would be to change our detection by re-implementing the original umockdev check. What do you guys think?

hughsie, 2017-10-19:
I'm really not the expert here. I'm happy to defer to you guys :)

commit fecfe3b8 (2017-10-19): thunderbolt: move test for mock environment (Closes #280)

umockdev 0.9.4 changed how to test for running in mock and this
causes the assumptions on mock testing at init to fail.

Move the test for this to after the environment has been set up.
commit ba19b180 (2017-10-19): thunderbolt: move test for mock environment (Closes #280)

umockdev 0.9.4 changed how to test for running in mock and this
causes the assumptions on mock testing at init to fail.

Move the test for this to after the environment has been set up.
closed 2017-10-19 by superm1