fwupd/fwupd #284: Refresh fails (key import fail)

gaenserich, 2017-10-22:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdgmr --version
client version: 1.0.0
daemon version: 1.0.0
compile-time dependency versions
        appstream-glib: 0.6.13
        gusb:   0.2.11
```

```shell
$ fwupdmgr get-devices
SFC30 Gamepad
  DeviceId:             usb:00:01
  Guid:                 a7fcfbaf-e9e8-59f4-920d-7691dc6c8699
  Summary:              A resdesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|needs-bootloader|registered
  Vendor:               8bitdo
  VendorId:             USB:0x2DC8
  Version:              2,71
  Icon:                 input-gaming
  Created:              2017-10-22
```

Please answer the following questions:

- Operating system and version: ArchLinux
```shell
$ uname -a
Linux trauerschwan 4.13.7-1-ARCH #1 SMP PREEMPT Tue Oct 17 12:19:41 EDT 2017 x86_64 GNU/Linux
```
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): _pacman_

- Have you tried rebooting? _yes_

- Are you using an NVMe disk? _no_

While attempting to update the firmware on an 8bitdo SFC30, I get the following when trying to refresh the metadata (both as a user and as root):
```shell
$ fwupdmgr -v refresh
(fwupdmgr:2014): Fu-DEBUG: creating path /root/.cache/fwupdmgr
(fwupdmgr:2014): Fu-DEBUG: downloading https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz to /root/.cache/fwupdmgr/lvfs-metadata.xml.gz
Fetching metadata https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
(fwupdmgr:2014): Fu-DEBUG: progress: 24%
Downloading…           [*********                              ](fwupdmgr:2014): Fu-DEBUG: progress: 48%
Downloading…           [******************                     ](fwupdmgr:2014): Fu-DEBUG: progress: 50%
Downloading…           [*******************                    ](fwupdmgr:2014): Fu-DEBUG: progress: 75%
Downloading…           [*****************************          ](fwupdmgr:2014): Fu-DEBUG: progress: 99%
Downloading…           [************************************** ](fwupdmgr:2014): Fu-DEBUG: progress: 100%
Downloading…           [***************************************]
(fwupdmgr:2014): Fu-DEBUG: downloading https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc to /root/.cache/fwupdmgr/lvfs-metadata.xml.gz.asc
Fetching signature https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc
(fwupdmgr:2014): Fu-DEBUG: progress: 100%

Failed to update metadata: key import failed /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service
```
Permissions of key files
```shell
$ cd /etc/pki
$ tree -hp
.
├── [drwxr-xr-x 4.0K]  fwupd
│   ├── [-rw-r--r-- 1.7K]  GPG-KEY-Hughski-Limited
│   ├── [-rw-r--r--  959]  GPG-KEY-Linux-Vendor-Firmware-Service
│   └── [-rw-r--r-- 1.6K]  LVFS-CA.pem
└── [drwxr-xr-x 4.0K]  fwupd-metadata
    ├── [-rw-r--r--  959]  GPG-KEY-Linux-Vendor-Firmware-Service
    └── [-rw-r--r-- 1.6K]  LVFS-CA.pem

2 directories, 5 files
```
sha512sum on file in question:
```shell
sha512sum /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service
c5b8c41792aed6b8fd42311eb6385672e3600b7b7e5da0557f39cc317e0f19972206cd5f2759106686244390c26668310a560c04e28d6cf7e783361736a1c055  /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service
```

gaenserich, 2017-10-22:
Almost forgot to mention: I added `GPG-KEY-Linux-Vendor-Firmware-Service` to the public key ring of both root and the user account, but that does not appear to change anything. Is there a way to add `GPG-KEY-Linux-Vendor-Firmware-Service` to the appropriate keyring?

hughsie, 2017-10-22:
What does fwupd --verbose say when root?

gaenserich, 2017-10-22:
It actually works when I manually started fwupd as root, although the last time I had fwupd running was before reboot. (After reboot, I hadn't started the daemon, I just ran fwupdmgr on its own).  Here's what the daemon threw out:
```shell
$ /usr/lib/fwupd/fwupd --verbose
18:38:06:0423 Fu  Verbose debugging enabled (on console 1)
18:38:06:0424 Fu  loading config values from /etc/fwupd/daemon.conf
18:38:06:0425 Fu  using config path of /etc/fwupd
18:38:06:0425 Fu  loading config from /etc/fwupd/remotes.d/fwupd.conf
18:38:06:0425 Fu  loading config from /etc/fwupd/remotes.d/lvfs.conf
18:38:06:0429 Fu  loading config from /etc/fwupd/remotes.d/vendor.conf
18:38:06:0429 Fu  loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
18:38:06:0429 Fu  using config path of /usr/lib/fwupd
18:38:06:0429 Fu  ordering lvfs=fwupd+1
18:38:06:0429 Fu  ordering lvfs-testing=lvfs+1
18:38:06:0429 Fu  remote lvfs-testing not enabled, so skipping
18:38:06:0430 Fu  no /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz, so skipping
18:38:06:0430 As  run AsStore:store-from-file{/usr/share/fwupd/remotes.d/fwupd/metadata.xml}
18:38:06:0430 As  run AsStore:store-from-root
18:38:06:0430 As  using icon path /usr/share/fwupd/remotes.d/fwupd/icons
18:38:06:0431 As  run AsStore:match-addons
18:38:06:0431 As  Emitting ::changed() [from-root]
18:38:06:0431 As  Emitting ::changed() [add-app]
18:38:06:0431 As  Emitting ::changed() [add-app]
18:38:06:0431 Fu  remote vendor not enabled, so skipping
18:38:06:0431 Fu  devices now in store:
18:38:06:0431 Fu  1     UEFI-dummy-dev0 UEFI Updates [0.0.1]
18:38:06:0431 Fu  2     com.via.VL811.firmware  VL811 Firmware [0.0.983]
18:38:06:0431 Fu  3     com.via.VL811+.firmware VL811+ Firmware [0.0.9095]
18:38:06:0431 Fu  4     com.via.VL812.firmware  VL812 Firmware [0.0.8581]
18:38:06:0431 Fu  5     com.via.VL812_B2.firmware       VL812 B2 Firmware [0.0.9091]
18:38:06:0441 Fu  smbios property Manufacturer=ASUSTeK Computer Inc.
18:38:06:0441 Fu  smbios property EnclosureKind=10
18:38:06:0441 Fu  smbios property Family=
18:38:06:0441 Fu  smbios property ProductName=UL30A
18:38:06:0441 Fu  smbios property ProductSku=
18:38:06:0441 Fu  smbios property BiosVendor=American Megatrends Inc.
18:38:06:0441 Fu  smbios property BiosVersion=212
18:38:06:0441 Fu  smbios property BiosMajorRelease=2
18:38:06:0442 Fu  smbios property BiosMinorRelease=12
18:38:06:0442 Fu  smbios property BaseboardManufacturer=ASUSTeK Computer Inc.
18:38:06:0442 Fu  smbios property BaseboardProduct=UL30A
18:38:06:0442 Fu  {8d75f0f1-5e10-5671-a31a-47f854d59fe8}   <- ASUSTeK Computer Inc.        &
          &UL30A               &                    &American Megatrends Inc.&212    &2&12
18:38:06:0442 Fu  {e7224170-92fe-5bfa-8060-f3e9dc830564}   <- ASUSTeK Computer Inc.        &
          &UL30A               &American Megatrends Inc.&212    &2&12
18:38:06:0442 Fu  {65662677-e387-52bb-af98-dadd0b0850ac}   <- ASUSTeK Computer Inc.        &UL30A
          &American Megatrends Inc.&212    &2&12
18:38:06:0442 Fu  {6306564f-6e22-5c80-9cf8-085dbd2af2e7}   <- ASUSTeK Computer Inc.        &
          &UL30A               &                    &ASUSTeK Computer Inc.        &UL30A
18:38:06:0442 Fu  {ce911992-b072-528b-92f6-ccad88c3448c}   <- ASUSTeK Computer Inc.        & 
          &UL30A               &
18:38:06:0442 Fu  {88d32d15-004a-517e-859a-3001363444fb}   <- ASUSTeK Computer Inc.        &
          &UL30A
18:38:06:0442 Fu  {403d152a-4216-5ace-b5e9-18239f206b79}   <- ASUSTeK Computer Inc.        &
          &ASUSTeK Computer Inc.        &UL30A
18:38:06:0442 Fu  {1a91cd0d-4d7e-5f93-acdc-b0117f446c4e}   <- ASUSTeK Computer Inc.        &

18:38:06:0442 Fu  {2226c0fa-e21a-5dc8-bad6-0d6453857e7f}   <- ASUSTeK Computer Inc.        &UL30A
          &ASUSTeK Computer Inc.        &UL30A
18:38:06:0442 Fu  {f33caab8-256c-50ab-be5b-669f356a5da7}   <- ASUSTeK Computer Inc.        &UL30A

18:38:06:0442 Fu  {403d152a-4216-5ace-b5e9-18239f206b79}   <- ASUSTeK Computer Inc.        &
          &ASUSTeK Computer Inc.        &UL30A
18:38:06:0443 Fu  {1a91cd0d-4d7e-5f93-acdc-b0117f446c4e}   <- ASUSTeK Computer Inc.        &

18:38:06:0443 Fu  {0b058ae3-2f24-5a8e-af13-49dc07dfae77}   <- ASUSTeK Computer Inc.        &10
18:38:06:0443 Fu  {3b9b4928-7389-5e43-8fda-b059974beb45}   <- ASUSTeK Computer Inc.        &ASUSTeK Co
mputer Inc.        &UL30A
18:38:06:0443 Fu  {7cd58f21-34a3-5eb3-84c2-3dfb4c90b6d3}   <- ASUSTeK Computer Inc.
18:38:06:0443 Fu  adding plugin /usr/lib/fwupd-plugins-3/libfu_plugin_ebitdo.so
18:38:06:0444 Fu  adding plugin /usr/lib/fwupd-plugins-3/libfu_plugin_upower.so
[... 288 lines cut ...]
18:38:27:0268 Fu  53    com.dell.uefi1610b70e.firmware  Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming System Update [0.1.0.7,0.1.0.4,0.1.0.3,…]
18:38:27:0269 Fu  54    com.dell.uefi222e3362.firmware  Precision Tower 3420 System Update [0.2.3.5]
18:38:27:0269 Fu  55    com.hughski.ColorHugALS.firmware        ColorHugALS [4.0.3,4.0.2,3.0.2,…]
18:38:27:0269 Fu  56    com.dell.uefia0a3aa54.firmware  Embedded Box PC 5000 System Update [0.1.3.4,0.1.2.0,0.1.1.3,…]
18:38:27:0269 Fu  57    com.logitech.Unifying.RQR12.firmware    Logitech Unifying Receiver [RQR12.07_B0029]
18:38:27:0269 Fu  58    com.dell.uefia81a55fe.firmware  OptiPlex 3240 AIO System Update [0.1.5.19,0.1.5.17,0.1.5.12,…]
18:38:27:0269 Fu  59    com.dell.uefi34578c72.firmware  XPS 15 9560/Precision 5520 System Update [0.1.5.0,0.1.4.0,0.1.3.4,…]
[... cut ...]
18:38:31:0044 Fu  loading config values from /etc/fwupd/daemon.conf
18:38:31:0047 Fu  using config path of /etc/fwupd
18:38:31:0047 Fu  loading config from /etc/fwupd/remotes.d/fwupd.conf
18:38:31:0048 Fu  loading config from /etc/fwupd/remotes.d/lvfs.conf
18:38:31:0049 Fu  loading config from /etc/fwupd/remotes.d/vendor.conf
18:38:31:0049 Fu  loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
18:38:31:0049 Fu  using config path of /usr/lib/fwupd
18:38:31:0049 Fu  ordering lvfs=fwupd+1
18:38:31:0049 Fu  ordering lvfs-testing=lvfs+1
```

hughsie, 2017-10-23:
So, the verbose log looks fine, and the checksums match. Are you able to fun the fwupd daemon using --verbose when it's autostarted? That should dump out more details into the journal. I'm trying to work out if Selinux/apparmour/cgroups are causing problems. Thanks.

hughsie, 2017-11-08:
Okay, please reopen if you have logs.

closed 2017-11-08 by hughsie

vendion, 2019-10-03:
I'm seeing this same error:

```
fwupdmgr --version
client version:	1.3.2
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.2
```

```
fwupdmgr get-devices
20ENCTO1WW
│
├─Walter p50 Thunderbolt Controller:
│     Device ID:           3d5e16884098554c25c40efc379129ec6b23ff42
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     16.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|registered
│     GUID:                9414852e-fa88-51e7-96f5-45268b29b86c ← TBT-01095050
│   
├─System Firmware:
│     Device ID:           2c1302f31806a0e0d57c377d99e18dae56351413
│     Current version:     0.1.52
│     Minimum Version:     0.1.46
│     Vendor:              LENOVO
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUID:                ddc0ee61-e7f0-4e7d-acc5-c070a398838e
│   
├─TOSHIBA THNSFJ256GDNU A:
│     Device ID:           5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:             ATA Drive
│     Current version:     JYLA1102
│     Serial Number:       86NS10KNTJ4V
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               f260d559-ffd1-56ce-88a6-fc38c53d8e9d ← IDE\TOSHIBA_THNSFJ256GDNU_A_________________JYLA1102
│                          9bc9dfb3-2789-5a46-afff-739966fddfd9 ← IDE\0TOSHIBA_THNSFJ256GDNU_A_________________
│                          9d768a22-7625-501d-bfa7-7155287899e0 ← TOSHIBA THNSFJ256GDNU A
│   
├─TOSHIBA THNSFJ256GDNU A:
│     Device ID:           612faa85309fec62ddfdce9f4635cdfef838d74e
│     Summary:             ATA Drive
│     Current version:     JYLA1102
│     Serial Number:       86NS10KYTJ4V
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               f260d559-ffd1-56ce-88a6-fc38c53d8e9d ← IDE\TOSHIBA_THNSFJ256GDNU_A_________________JYLA1102
│                          9bc9dfb3-2789-5a46-afff-739966fddfd9 ← IDE\0TOSHIBA_THNSFJ256GDNU_A_________________
│                          9d768a22-7625-501d-bfa7-7155287899e0 ← TOSHIBA THNSFJ256GDNU A
│   
├─UEFI Device Firmware:
│     Device ID:           1f9ed21ce218ebcd0e1c803b7da08dd4e33d01a3
│     Current version:     176.16.1000
│     Minimum Version:     0.0.1
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUID:                671d19d0-d43c-4852-98d9-1ce16f9967e4
│   
├─UEFI Device Firmware:
│     Device ID:           6c8326bcda55dc53571921c60da36f320d3efa3f
│     Current version:     0.1.16
│     Minimum Version:     0.0.1
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUID:                a9971959-9246-4a5b-b2f2-ba6fdcb19349
│   
└─WDC WD5000LPLX-08ZNTT0:
      Device ID:           948241a24320627284597ec95079cc1341c90518
      Summary:             ATA Drive
      Current version:     03.01A03
      Serial Number:       WD-WX61A96LNKXY
[... 766 lines cut ...]
01:56:30:0251 FuPlugin             performing udev_device_added() on ata
01:56:30:0251 FuPlugin             performing udev_device_added() on nvme
01:56:30:0251 FuPlugin             performing udev_device_added() on synapticsmst
01:56:30:0251 FuPlugin             performing udev_device_added() on udev
01:56:30:0251 FuPlugin             performing udev_device_added() on wacom_raw
01:56:30:0251 FuEngine             Emitting PropertyChanged('Status'='idle')
01:56:30:0251 FuMain               Emitting PropertyChanged('Status'='idle')
01:56:30:0253 FuMain               Started with locale en_US.UTF-8
01:56:30:0253 FuMain               Daemon ready for requests
01:56:30:0253 FuMain               FuMain: acquired name: org.freedesktop.fwupd
01:56:33:0302 FuMain               Called GetRemotes()
01:56:33:0527 FuMain               Called UpdateMetadata(lvfs,10,11)
01:56:33:0570 FuKeyring            Using keyring at /var/lib/fwupd/gnupg
01:56:33:0581 FuKeyring            Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
01:56:33:0590 FuKeyring            imported: 0, unchanged: 0, not_imported: 0
01:58:47:0978 FuMain               Called GetDevices()
01:58:47:0982 FuMain               Called GetHistory()
02:01:07:0562 FuMain               Called GetRemotes()
02:01:07:0749 FuMain               Called UpdateMetadata(lvfs,10,11)
02:01:07:0749 FuKeyring            Using keyring at /var/lib/fwupd/gnupg
02:01:07:0753 FuKeyring            Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
02:01:07:0758 FuKeyring            imported: 0, unchanged: 0, not_imported: 0
```

I had to remove the directory that the gpg keychain is stored (/var/lib/fwupd/gnupg) and restart the daemon to clear the issue.

hughsie, 2019-10-03:
This looks odd:

    02:01:07:0753 FuKeyring            Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
    02:01:07:0758 FuKeyring            imported: 0, unchanged: 0, not_imported: 0

It's supposed to be something like this:

    08:00:20:0016 FuKeyring            Adding GnuPG public key /home/hughsie/.root/etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
    08:00:20:0027 FuKeyring            importing key CAE0420259890141E701156E0A6DCEF6B46F3DBF [0] Success

Unfortunately gnupg is about as easy to debug as a yak with toothache.

superm1, 2019-10-03:
I saw stuff like this when we had problems with the initial snaps.  It ended up being missing packages, but it was only debugged by using `GPGME_DEBUG=9` in the environment and referring to gpgme code for what all the messages actually meant.

vendion, 2019-10-03:
Agreed it did look odd, and would explain why allowing fwupd to just recreate the keychain resolved the issue. I had just ran into this issue yesterday and saw that there wasn't much of a solution so figured post it here for future reference.

dragondaksh, 2020-09-04:
By the way its nothing odd with this issue, the gpg binary file is either not supported, missing or depending library file for gpg binary is missing.

getjack, 2020-10-23:
I just could reproduce this issue after upgrading my Ubuntu 18.04 to 20.04 using the APT version of fwupd. I get almost the exact output as the issue creator.

```
06:00:36:0169 FuMain               updating lvfs
06:00:36:0209 FuKeyring            Using keyring at /var/lib/fwupd/gnupg
06:00:36:0220 FuKeyring            Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
06:00:36:0236 FuKeyring            imported: 0, unchanged: 0, not_imported: 0
key import failed /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
```

I cannot use the SNAP version, which seems to be newer, but AppArmor seems to block it and i don't get why.

superm1, 2020-10-23:
@getjack can you please check for the gpg binary ok your system? If it is there can you please help open a new bug report and we can diagnose you separately.