fwupd/fwupd #288: Is fwupd eating m battery?

godiard, 2017-10-23:
Powertop reports:

```shell
PowerTOP v2.9     Overview   Idle stats   Frequency stats   Device stats   Tunables                                     

The battery reports a discharge rate of 11.7 W
The power consumed was 231 J
The estimated remaining time is 3 hours, 53 minutes

Summary: 752,5 wakeups/second,  0,0 GPU ops/seconds, 0,0 VFS ops/sec and 14,2% CPU use

Power est.              Usage       Events/s    Category       Description
  7.59 W     26,4 µs/s       0,5        Process        [PID 2905] /usr/libexec/fwupd/fwupd
  2.89 W     10,0%                      Device         Display backlight
  1.12 W      0,0 pkts/s                Device         nic:virbr0
  172 mW      5,3 ms/s     225,9        Interrupt      PS/2 Touchpad / Keyboard / Mouse
  102 mW     26,9 ms/s     129,3        Process        [PID 24055] /opt/google/chrome/chrome --type=renderer --field-trial-handle=1578705589192891195,61639737
 76.5 mW    100,0%                      Device         Radio device: ideapad_acpi
 51.5 mW     36,9 ms/s      61,0        Process        [PID 2273] /usr/bin/gnome-shell
 38.2 mW      2,9 pkts/s                Device         Network interface: wlp2s0 (iwlwifi)
 37.7 mW     14,1 ms/s      47,1        Process        [PID 23579] /opt/google/chrome/chrome --type=gpu-process --field-trial-handle=1578705589192891195,61639
 37.6 mW      4,6 ms/s      48,8        Process        [PID 23503] /opt/google/chrome/chrome
 37.4 mW      3,3 ms/s      48,8        Process        [PID 2280] /usr/bin/Xwayland :0 -rootless -noreset -listen 4 -listen 5 -displayfd 6
 29.5 mW      8,7 ms/s      37,3        Process        [PID 23469] /opt/google/chrome/chrome
 15.7 mW    224,2 µs/s      20,7        Process        [PID 8] [rcu_sched]
 11.6 mW    150,3 µs/s      15,3        Process        [PID 844] /sbin/rngd -f
 10.1 mW    610,2 µs/s      13,2        Process        [PID 2298] /usr/libexec/at-spi2-registryd --use-gnome-session
 6.97 mW      3,5 ms/s       8,5        Process        [PID 23748] /opt/google/chrome/chrome --type=renderer --field-trial-handle=1578705589192891195,61639737
 6.41 mW    647,7 µs/s       8,3        Timer          hrtimer_wakeup
 6.25 mW      2,1 ms/s       7,9        Process        [PID 23894] /opt/google/chrome/chrome --type=renderer --field-trial-handle=1578705589192891195,61639737
 4.80 mW      3,5 ms/s       5,7        Interrupt      [7] sched(softirq)
 4.68 mW      3,2 ms/s       5,6        Process        [PID 24682] /opt/google/chrome/chrome --type=renderer --field-trial-handle=1578705589192891195,61639737
 4.22 mW    224,4 µs/s       5,5        Process        [PID 404] [i915/signal:0]
 4.19 mW     10,3 ms/s       3,6        Process        [PID 18503] /usr/libexec/gnome-terminal-server
 3.80 mW      1,2 ms/s       4,8        Timer          tick_sched_timer
 3.01 mW     95,1 µs/s       4,0        kWork          gc_worker
 2.90 mW    116,6 µs/s       3,8        Process        [PID 29] [rcuos/2]
 2.74 mW    117,8 µs/s       3,6        Process        [PID 61] [rcuos/6]
 2.38 mW     52,8 µs/s       3,1        Process        [PID 45] [rcuos/4]
```
I don't know is this is useful but strace reports:

```shell
[gonzalo@localhost ~]$ ps ax | grep fw
 2905 ?        Ssl    0:01 /usr/libexec/fwupd/fwupd

[gonzalo@localhost ~]$ sudo strace -p  2905
strace: Process 2905 attached
restart_syscall(<... resuming interrupted poll ...>) = 0
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
poll([{fd=3, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}], 3, 5000) = 0 (Timeout)
```
forever...

```shell
$ fwupdgmr --version
client version:	0.9.6
daemon version:	0.9.6
compile-time dependency versions
	appstream-glib:	0.7.0
	gusb:	0.2.11
```
[... 144 lines cut ...]

- Operating system and version:
Fedora 26

- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
From Fedora repository
- Have you tried rebooting?
No. 
- Are you using an NVMe disk?
No
- Is secure boot enabled (only for the UEFI plugin)?
I think so:
```shell
[gonzalo@localhost ~]$ dmesg | grep EFI
[    0.000000] efi: EFI v2.31 by INSYDE Corp.
[    0.000000] ACPI: UEFI 0x000000008CFFC000 000236 (v01 LENOVO CB-01    00000001 ACPI 00040000)
[    1.282900] fb0: EFI VGA frame buffer device
[    1.510807] Loaded UEFI:db cert 'Microsoft Windows Production PCA 2011: a92902398e16c49778cd90f99e4f9ae17c55af53' linked to secondary sys keyring
[    1.510823] Loaded UEFI:db cert 'Microsoft Corporation UEFI CA 2011: 13adbf4309bd82709c8cd54f316ed522988a1bd4' linked to secondary sys keyring
[    1.510832] Loaded UEFI:db cert 'Y401Y501: b12ab2c18d792db641973ab09bf0629c' linked to secondary sys keyring
[    1.511602] Loaded UEFI:MokListRT cert 'Fedora Secure Boot CA: fde32599c2d61db1bf5807335d7b20e4cd963b42' linked to secondary sys keyring
[    2.207554] fb: switching to inteldrmfb from EFI VGA

```
I hope this is useful in some way. Will try reboot and check if happen again.

godiard, 2017-10-23:
After reboot, again:
```shell
PowerTOP v2.9     Overview   Idle stats   Frequency stats   Device stats   Tunables                                     

The battery reports a discharge rate of 15.6 W
The power consumed was 0.00 J
The estimated remaining time is 2 hours, 33 minutes

Summary: 655,6 wakeups/second,  0,0 GPU ops/seconds, 0,0 VFS ops/sec and 3,8% CPU use

Power est.              Usage       Events/s    Category       Description
  8.78 W     23,2 µs/s      0,00        Process        [PID 2954] /usr/libexec/fwupd/fwupd
  2.00 W      0,0 pkts/s                Device         nic:virbr0
  1.80 W     10,0%                      Device         Display backlight
  197 mW     20,5 ms/s     385,7        Process        [PID 2761] /usr/libexec/tracker-miner-fs
 32.9 mW      1,0 ms/s      64,8        Interrupt      [4] block(softirq)
 15.9 mW    100,6 µs/s      31,4        Process        [PID 8] [rcu_sched]
 14.4 mW     14,9 µs/s      28,5        kWork          cfq_kick_queue
 10.4 mW     95,7 µs/s      20,6        Process        [PID 865] /sbin/rngd -f
 6.02 mW      1,0 pkts/s                Device         Network interface: wlp2s0 (iwlwifi)
 5.97 mW    157,5 µs/s      11,8        Timer          hrtimer_wakeup
 5.96 mW     31,6 µs/s      11,8        Process        [PID 10] [rcuos/0]
 5.72 mW      7,6 ms/s       9,8        Process        [PID 2270] /usr/bin/gnome-shell
 5.22 mW      2,5 ms/s       9,8        Process        [PID 2946] /usr/libexec/gnome-terminal-server
 4.99 mW    300,6 µs/s       9,8        Timer          tick_sched_timer
 4.47 mW     62,5 µs/s       8,8        kWork          gc_worker
 3.50 mW    228,9 µs/s       6,9        Interrupt      [28] i915
 3.48 mW     32,7 µs/s       6,9        Process        [PID 29] [rcuos/2]
 2.49 mW     95,1 µs/s       4,9        Process        [PID 643] [irq/30-iwlwifi]
 2.04 mW    544,8 µs/s       3,9        Process        [PID 2686] /usr/sbin/cupsd -l
 1.50 mW     85,3 µs/s       2,9        Process        [PID 407] [i915/signal:0]
 1.49 mW     55,4 µs/s       2,9        kWork          i915_gem_idle_work_handler
 1.49 mW      1,7 µs/s       2,9        kWork          blk_delay_work
 1.07 mW      0,8 ms/s       2,0        Interrupt      [7] sched(softirq)
 1.00 mW    125,2 µs/s       2,0        Process        [PID 3484] pool
 999 µW      65,7 µs/s       2,0        Process        [PID 1531] /usr/libexec/gsd-color
 993 µW       2,9 µs/s       2,0        kWork          intel_atomic_helper_free_state_
 992 µW       0,0 µs/s       2,0        kWork          intel_atomic_commit_work
 516 µW     195,5 µs/s       1,0        Process        [PID 3505] powertop
 508 µW     120,1 µs/s       1,0        Process        [PID 429] [jbd2/sda4-8]
 503 µW      64,1 µs/s       1,0        Process        [PID 3486] pool
 502 µW      60,4 µs/s       1,0        Process        [PID 1049] /usr/bin/python -Es /usr/sbin/tuned -l -P
```
Tell me if you need more information

hughsie, 2017-10-24:
Powertop uses a pretty broken heuristic, 8.78W is clearly insane. That said, we did fix a bug to avoid opening lots of devices, so if you could try 1.0.0 and tell me if it's fixed then I'll know I need to backport the FuDeviceLocker patches back to 0.9.*

godiard, 2017-10-25:
I have installed fwupd 1.0.0 from rawhide (I needed uninstall gnome-software) and fwupd does not appear anymore in powertop list. Consume was reduced, but I don't know if the deamon is running right now.

How can I be sure? 

```
PowerTOP v2.9     Overview   Idle stats   Frequency stats   Device stats   Tunables                                     

The battery reports a discharge rate of 8.40 W
The power consumed was 190 J
The estimated remaining time is 4 hours, 27 minutes

Summary: 543,2 wakeups/second,  0,0 GPU ops/seconds, 0,0 VFS ops/sec and 10,5% CPU use

Power est.              Usage       Events/s    Category       Description
  3.66 W      0,0 pkts/s                Device         nic:virbr0
  2.40 W     10,0%                      Device         Display backlight
  1.36 W      2,1 pkts/s                Device         Network interface: wlp2s0 (iwlwifi)
  972 mW    100,0%                      Device         Radio device: ideapad_acpi
  270 mW     14,2 ms/s      99,2        Process        [PID 3596] /opt/google/chrome/chrome
  153 mW     24,1 ms/s      54,3        Process        [PID 3759] /opt/google/chrome/chrome --type=renderer --field-trial-handle=16515333486807966487,55470936
  126 mW      5,1 ms/s      46,7        Process        [PID 3813] /opt/google/chrome/chrome --type=renderer --field-trial-handle=16515333486807966487,55470936
  109 mW      9,0 ms/s      39,7        Process        [PID 3711] /opt/google/chrome/chrome --type=gpu-process --field-trial-handle=16515333486807966487,55470
 95.1 mW      0,0 pkts/s                Device         Network interface: enp1s0 (alx)
 95.0 mW      2,7 ms/s      35,2        Process        [PID 3761] /opt/google/chrome/chrome --type=renderer --field-trial-handle=16515333486807966487,55470936
 91.7 mW    583,3 µs/s      34,2        Interrupt      PS/2 Touchpad / Keyboard / Mouse
 87.9 mW      2,0 ms/s      32,6        Process        [PID 2255] /usr/bin/Xwayland :0 -rootless -noreset -listen 4 -listen 5 -displayfd 6
 68.9 mW    258,4 µs/s      25,8        Process        [PID 9] [rcu_sched]
 62.2 mW     11,3 ms/s      21,8        Process        [PID 2248] /usr/bin/gnome-shell
 36.7 mW      1,0 ms/s      13,6        Timer          hrtimer_wakeup
 35.8 mW      4,7 ms/s      12,8        Process        [PID 4146] /opt/google/chrome/chrome --type=renderer --field-trial-handle=16515333486807966487,55470936
 26.3 mW      4,0 ms/s       9,3        Process        [PID 3952] /opt/google/chrome/chrome --type=renderer --field-trial-handle=16515333486807966487,55470936

```

hughsie, 2017-10-25:
Either `fwupdmgr get-devices` or `ps aux | grep fwupd`

godiard, 2017-10-25:
Ok, running :)

Maybe you shoould backport that changes!

hughsie, 2017-11-08:
Backporting this is a large change. I'd like to get some more testing of 1.0.0 before doing a smaller backport.

closed 2017-11-08 by hughsie

godiard, 2017-11-08:
No problem, let me know if I can help with any data.
Thanks for all your work! (big Fedora fun here)